chipweinberger/flutter_blue_plus #847: [Help]: Initializing BluetoothAdapter throws exception

shuizhuyu1015, 2024-04-11:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.2

### Flutter Version

3.3.10

### What OS?

Android

### OS Version

Android 11

### Bluetooth Module

Nordic nRF5

### What is your problem?

Initializing BluetoothAdapter throws exception, but FBP is still available.
The exception doesn't seem to make a difference. I'm just giving feedback.

### Logs

```shell
D/[FBP-Android](  482): [FBP] onMethodCall: flutterHotRestart
D/[FBP-Android](  482): [FBP] initializing BluetoothAdapter
D/BluetoothAdapter(  482): java.lang.Throwable
D/BluetoothAdapter(  482): 	at android.bluetooth.BluetoothAdapter.<init>(BluetoothAdapter.java:762)
D/BluetoothAdapter(  482): 	at android.bluetooth.BluetoothAdapter.getDefaultAdapter(BluetoothAdapter.java:735)
D/BluetoothAdapter(  482): 	at android.bluetooth.BluetoothManager.<init>(BluetoothManager.java:72)
D/BluetoothAdapter(  482): 	at android.app.SystemServiceRegistry$10.createService(SystemServiceRegistry.java:325)
D/BluetoothAdapter(  482): 	at android.app.SystemServiceRegistry$10.createService(SystemServiceRegistry.java:322)
D/BluetoothAdapter(  482): 	at android.app.SystemServiceRegistry$CachedServiceFetcher.getService(SystemServiceRegistry.java:1722)
D/BluetoothAdapter(  482): 	at android.app.SystemServiceRegistry.getSystemService(SystemServiceRegistry.java:1402)
D/BluetoothAdapter(  482): 	at android.app.ContextImpl.getSystemService(ContextImpl.java:1940)
D/BluetoothAdapter(  482): 	at android.content.ContextWrapper.getSystemService(ContextWrapper.java:803)
D/BluetoothAdapter(  482): 	at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.onMethodCall(FlutterBluePlusPlugin.java:278)
D/BluetoothAdapter(  482): 	at io.flutter.plugin.common.MethodChannel$IncomingMethodCallHandler.onMessage(MethodChannel.java:262)
D/BluetoothAdapter(  482): 	at io.flutter.embedding.engine.dart.DartMessenger.invokeHandler(DartMessenger.java:295)
D/BluetoothAdapter(  482): 	at io.flutter.embedding.engine.dart.DartMessenger.lambda$dispatchMessageToQueue$0$io-flutter-embedding-engine-dart-DartMessenger(DartMessenger.java:319)
D/BluetoothAdapter(  482): 	at io.flutter.embedding.engine.dart.DartMessenger$$ExternalSyntheticLambda0.run(Unknown Source:12)
D/BluetoothAdapter(  482): 	at android.os.Handler.handleCallback(Handler.java:938)
D/BluetoothAdapter(  482): 	at android.os.Handler.dispatchMessage(Handler.java:99)
D/BluetoothAdapter(  482): 	at android.os.Looper.loop(Looper.java:233)
D/BluetoothAdapter(  482): 	at android.app.ActivityThread.main(ActivityThread.java:8030)
D/BluetoothAdapter(  482): 	at java.lang.reflect.Method.invoke(Native Method)
D/BluetoothAdapter(  482): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:631)
D/BluetoothAdapter(  482): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:978)
D/[FBP-Android](  482): [FBP] disconnectAllDevices(flutterHotRestart)
D/[FBP-Android](  482): [FBP] connectedPeripherals: 0
D/[FBP-Android](  482): [FBP] onMethodCall: getAdapterState
```

chipweinberger, 2024-04-11:
never seen this before. strange. thanks.

chipweinberger, 2024-04-11:
I don't think there is anything FBP can do to fix this. but lmk

chipweinberger, 2024-04-17:
same issue: https://github.com/boskokg/flutter_blue_plus/issues/508

theirs went away randomly

chipweinberger, 2024-04-17:
closing because this seems more like an android bug, and not something we can fix.