chipweinberger/flutter_blue_plus #849: [Help]: Can't scan again for devices on Android after turning on and turning off Bluetooth.

kryjomskiDev, 2024-04-11:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.1

### Flutter Version

3.19.2

### What OS?

Android

### OS Version

13

### Bluetooth Module

nRF52833

### What is your problem?

Hi, I've noticed strange behavior when trying to rescan for devices after turning off and then on bluetooth on Android.
When Bluetooth is switched from the off state to the on state on Android and when I try to run the startScan method again, I get this error:
```sh
E/[FBP-Android](12294): [FBP] onScanFailed: SCAN_FAILED_ALREADY_STARTED
I/flutter (12294): 2024-04-11T12:29:50.332665	E	MachineSelectionCubit._listenToFoundedDevicesStream.<ac>:	 MachineSelectionCubit Founded devices stream error
I/flutter (12294): FlutterBluePlusException | scan | android-code: 1 | SCAN_FAILED_ALREADY_STARTED
```
I am getting this error in the FlutterBluePlus.scanResults stream. At first I thought that maybe I don't need to invoke the startScan method again and without calling this method the result is slightly different. I don't get this error, but I also don't scan for devices, so I can't get new results. I even tried wrapping the startScan method in some if checks and using the (FlutterBluePlus.isScanningNow) getter to call startScan if there is no scan running, but the error still occurs. What else I tried was listening to the isScanning stream, and see what result I'll get before running startScan(). I got "false" answer from stream which tells that there is noScanning and while invoking startScan method I still see error which tells me that scan already started.

I've tested this on a number of Android devices, as well as on different versions of the operating system (starting from Android 10 and ending with Android 13) on all devices I encounter the same problem. I also tried to reproduce the problem in an example app and succeeded to replicate the problem. It is worth mentioning that on iOS such a problem does not occur.

To reproduce: 
1. Turn on bluetooth
2. Start scan for device 
3. Turn off bluetooth
4. Turn on bluetooth
5. Try scan for device

Here is my code, you can also try to reproduce on example app: 
```dart

 StreamSubscription<BluetoothAdapterState>? _bluetoothStatusSubscription;
 StreamSubscription<List<ScanResult>>? _foundedDevicesSubscription;
  
 Future<void> init() async {
   await _listenToBluetoothStatusStream();
  }

 Future<void> _listenToBluetoothStatusStream() async {
    await _bluetoothStatusSubscription?.cancel();

   _bluetoothStatusSubscription = FlutterBluePlus.adapterState.listen(_bluetoothStatusListener);
  }

  Future<void> _bluetoothStatusListener(BluetoothAdapterState event) async {
    try {
      if (event == BluetoothAdapterState.on) {
        if (!FlutterBluePlus.isScanningNow) {
          await  FlutterBluePlus.startScan();
        }
        await _listenToFoundedDevicesStream();
      } else if (event == BluetoothAdapterState.off) {
       // handle this state
      } else if (event == BluetoothAdapterState.unauthorized) {
       // handle this state
      }
    } catch (error, st) {
      Fimber.e('$runtimeType Bluetooth status listener error', ex: error, stacktrace: st);
    }
  }

Future<void> _listenToFoundedDevicesStream() async {
    await _foundedDevicesSubscription?.cancel();
    _foundedDevicesSubscription = FlutterBluePlus.scanResults.listen(
      _foundedDevicesListener,
      onError: (error, st) {
       // Here Im getting the error.
        Fimber.e('$runtimeType Founded devices stream error', ex: error, stacktrace: st);
      },
      cancelOnError: true,
    );
  }

  Future<void> _foundedDevicesListener(List<ScanResult> devices) async {
  print(devices);
  }
```
Here are logs from console (as you can see I printed also events from isScanningNow getter  e.g 'isScanningNow: false' which is before startScan) : 
```sh
D/[FBP-Android](12294): [FBP] onMethodCall: flutterHotRestart
D/BluetoothAdapter(12294): isLeEnabled(): ON
D/[FBP-Android](12294): [FBP] disconnectAllDevices(flutterHotRestart)
D/[FBP-Android](12294): [FBP] connectedPeripherals: 0
D/[FBP-Android](12294): [FBP] onMethodCall: startScan
D/BluetoothAdapter(12294): isLeEnabled(): ON
D/BluetoothLeScanner(12294): onScannerRegistered() - status=0 scannerId=6 mScannerId=0
D/[FBP-Android](12294): [FBP] onMethodCall: getAdapterState
I/flutter (12294): BluetoothOn
I/flutter (12294): isScanningNow: true
D/[FBP-Android](12294): [FBP] OnAdapterStateChanged: turningOff
D/[FBP-Android](12294): [FBP] disconnectAllDevices(adapterTurnOff)
D/[FBP-Android](12294): [FBP] OnAdapterStateChanged: off
D/[FBP-Android](12294): [FBP] disconnectAllDevices(adapterTurnOff)
I/flutter (12294): BluetoothOff
D/[FBP-Android](12294): [FBP] OnAdapterStateChanged: turningOn
D/[FBP-Android](12294): [FBP] OnAdapterStateChanged: on
I/flutter (12294): BluetoothOn
I/flutter (12294): isScanningNow: false
D/[FBP-Android](12294): [FBP] onMethodCall: startScan
D/BluetoothAdapter(12294): isLeEnabled(): ON
E/[FBP-Android](12294): [FBP] onScanFailed: SCAN_FAILED_ALREADY_STARTED
I/flutter (12294): 2024-04-11T12:29:50.332665	E	MachineSelectionCubit._listenToFoundedDevicesStream.<ac>:	 MachineSelectionCubit Founded devices stream error
I/flutter (12294): FlutterBluePlusException | scan | android-code: 1 | SCAN_FAILED_ALREADY_STARTED
I/flutter (12294): 	
I/flutter (12294): ----------------FIREBASE CRASHLYTICS----------------
I/flutter (12294): The following exception was thrown MachineSelectionCubit Founded devices stream error:
I/flutter (12294): FlutterBluePlusException | scan | android-code: 1 | SCAN_FAILED_ALREADY_STARTED
I/flutter (12294): 
I/flutter (12294): ----------------------------------------------------
```
Currently the workaround that I found is to just use the FlutterBluePlus.startScan() method 2 times after turning from off to on state: 
```dart
  Future<void> _bluetoothStatusListener(BluetoothAdapterState event) async {
    try {
      if (event == BluetoothAdapterState.on) {
      _wasBluetoothOn = true;
        await _listenToFoundedDevicesStream();
         await  FlutterBluePlus.startScan();
        if (Platform.isAndroid && _applyWorkaround) {
           await  FlutterBluePlus.startScan();
        }
      } else if (event == BluetoothAdapterState.off) {
         if (_wasBluetoothOn) {
          _applyWorkaround = true;
        }
       // handle this state
      } else if (event == BluetoothAdapterState.unauthorized) {
       // handle this state
      }
    } catch (error, st) {
      Fimber.e('$runtimeType Bluetooth status listener error', ex: error, stacktrace: st);
    }
  }
```
This workaround is not perfect since Im still receiving the error but at least the scanning is running again and I can see the results.

### Logs

```shell
D/[FBP-Android](12294): [FBP] onMethodCall: flutterHotRestart
D/BluetoothAdapter(12294): isLeEnabled(): ON
D/[FBP-Android](12294): [FBP] disconnectAllDevices(flutterHotRestart)
D/[FBP-Android](12294): [FBP] connectedPeripherals: 0
D/[FBP-Android](12294): [FBP] onMethodCall: startScan
D/BluetoothAdapter(12294): isLeEnabled(): ON
D/BluetoothLeScanner(12294): onScannerRegistered() - status=0 scannerId=6 mScannerId=0
D/[FBP-Android](12294): [FBP] onMethodCall: getAdapterState
I/flutter (12294): BluetoothOn
I/flutter (12294): isScanningNow: true
D/[FBP-Android](12294): [FBP] OnAdapterStateChanged: turningOff
D/[FBP-Android](12294): [FBP] disconnectAllDevices(adapterTurnOff)
D/[FBP-Android](12294): [FBP] OnAdapterStateChanged: off
D/[FBP-Android](12294): [FBP] disconnectAllDevices(adapterTurnOff)
I/flutter (12294): BluetoothOff
D/[FBP-Android](12294): [FBP] OnAdapterStateChanged: turningOn
D/[FBP-Android](12294): [FBP] OnAdapterStateChanged: on
I/flutter (12294): BluetoothOn
I/flutter (12294): isScanningNow: false
D/[FBP-Android](12294): [FBP] onMethodCall: startScan
D/BluetoothAdapter(12294): isLeEnabled(): ON
E/[FBP-Android](12294): [FBP] onScanFailed: SCAN_FAILED_ALREADY_STARTED
I/flutter (12294): 2024-04-11T12:29:50.332665	E	MachineSelectionCubit._listenToFoundedDevicesStream.<ac>:	 MachineSelectionCubit Founded devices stream error
I/flutter (12294): FlutterBluePlusException | scan | android-code: 1 | SCAN_FAILED_ALREADY_STARTED
I/flutter (12294): 	
I/flutter (12294): ----------------FIREBASE CRASHLYTICS----------------
I/flutter (12294): The following exception was thrown MachineSelectionCubit Founded devices stream error:
I/flutter (12294): FlutterBluePlusException | scan | android-code: 1 | SCAN_FAILED_ALREADY_STARTED
I/flutter (12294): 
I/flutter (12294): ----------------------------------------------------
```

chipweinberger, 2024-04-11:
try rebooting your device

seems like a bug in Android

also update your FBP

kryjomskiDev, 2024-04-12:
After FBP update to 1.32.2 error still occurs.

chipweinberger, 2024-04-12:
try rebooting your device

kryjomskiDev, 2024-04-14:
Unfortunately rebooting didn't help. This happens on all Android devices I've used

chipweinberger, 2024-04-18:
wonder why no one else is having this problem

olo95, 2024-04-18:
+1, same issue. Checking isScanningNow property, returns false and I proceed with the scan. Then the same error occurs.

rodkiewicz, 2024-04-18:
I have the same problem. `  flutter_blue_plus: 1.32.1`

kryjomskiDev, 2024-04-18:
Yes, I was able to reproduce the problem in the example app. I also checked the delays (60s, 30s, 1s, 200ms), but got the same result.

chipweinberger, 2024-04-18:
thanks for reporting.

this error comes from android. Im not aware how to fix it currently.

olo95, 2024-04-19:
Tried stopScan and then waited a second until next scan is launched. Same error.

chipweinberger, 2024-04-19:
did you try stopScan with version 1.32.0?

olo95, 2024-04-19:
Yes, I tried earlier versions and reached to 1.32.1 which threw this error. 1.32.0 and earlier versions doesnt have that issue.

chipweinberger, 2024-04-19:
in 1.32.0 & earlier the error happens but is ignored.

see this commit: https://github.com/boskokg/flutter_blue_plus/commit/d3aa378e92de81652322c9d0d4ed16120cd0bcc0

chipweinberger, 2024-04-19:
because the error is ignored, you should try reverting this commit (https://github.com/boskokg/flutter_blue_plus/commit/69ccb70711180908d5474ab28be1e1019afea134) so that you can call `stopScan` on version 1.32.2

olo95, 2024-04-19:
I don't use scan filters, could you show me which configuration I could test for you?

chipweinberger, 2024-04-19:
just try setting any scan filter, and see if the problem still happens.

kryjomskiDev, 2024-04-19:
I set the "withNames" filter and after turning bluetooth off and on, I didn't get an error, but I also didn't get any scan results in the stream. Before turning off bluetooth I could see my device in the stream. Can confirm that reverting the commit or downgrading the FBP version works.

olo95, 2024-04-19:
For me, yes, it does work

kryjomskiDev, 2024-04-19:
correct

chipweinberger, 2024-04-19:
pushed fix: https://github.com/boskokg/flutter_blue_plus/commit/4aba5cca47fb5cd5835d05d4154b41bd53aab5b8

you'll need to call `stopScan` if you encounter the SCAN_FAILED_ALREADY_STARTED error

chipweinberger, 2024-04-19:
pushed in 1.32.3

chipweinberger, 2024-04-19:
this seems like a better fix: https://stackoverflow.com/questions/29781393/android-ble-scan-scan-failed-already-started-and-getting-hold-of-a-bluetoothdev#comment97662415_29906120

maybe someone can try it?

chipweinberger, 2024-04-19:
the relevant android code is here: https://cs.android.com/android/platform/superproject/main/+/main:packages/modules/Bluetooth/framework/java/android/bluetooth/le/BluetoothLeScanner.java?q=SCAN_FAILED_ALREADY_STARTED

chipweinberger, 2024-04-19:
I've found the problem.

In our dart code we stop scanning when the adapter is turned off, `_stopScan(invokePlatform: false)` but `invokePlatform: false` is false. This means out dart code thinks we stopped scanning, but the platform code thinks we are still scanning.

```
  static Future<dynamic> _methodCallHandler(MethodCall call) async {

.....

    // keep track of adapter states
    if (call.method == "OnAdapterStateChanged") {
      BmBluetoothAdapterState r = BmBluetoothAdapterState.fromMap(call.arguments);
      _adapterStateNow = r.adapterState;
      if (isScanningNow && r.adapterState != BmAdapterStateEnum.on) {
        _stopScan(invokePlatform: false);
      }
      if (r.adapterState == BmAdapterStateEnum.on) {
        for (DeviceIdentifier d in _autoConnect) {
          BluetoothDevice(remoteId: d).connect(autoConnect: true, mtu: null).onError((e, s) {
            if (logLevel != LogLevel.none) {
              print("[FBP] [AutoConnect] connection failed: $e");
            }
          });
        }
      }
    }
```

I'm not sure the best way to resolve the problem.

The easiest solution is to stop scanning in the android java code when BLE is turned off.

olo95, 2024-04-19:
What was the intention behind setting invokePlatform to false?

chipweinberger, 2024-04-19:
It was a bug. I thought it was unnecessary to invoke the platform. Bluetooth is off, why would we need to stop the scan?

I did not realize that scanning would resume when bluetooth was turned back on.

----

I've pushed imo the best fix: https://github.com/boskokg/flutter_blue_plus/commit/99740a73c221c9d79837fe2be804962583e3dcfd

This stops scanning when BLE is turned off. This is most expected IMO.

I use native code instead of `invokePlatform:true`. Using `invokePlatform:true` would cause additional logs & would use the platform channel and be async. It's better to do it in native code imo.

----

please try master branch!

olo95, 2024-04-19:
Thank you for your work :)

tucici, 2024-09-24:
> 应该在1.32.4中固定

The latest version 1.32.11 is not working and still has this issue.

 I use your demo to do the following: 
turn on bluetooth---> scan -- > quickly turn off bluetooth -- > quickly turn on bluetooth -- > scan again, 
and it will appear. 

You should re-assign "ScanCallback" in the android Bluetooth code when "startscan" is called

chipweinberger, 2024-09-24:
@tucici please open a PR

chipweinberger, 2024-11-08:
Even though I did not have a problem, I noticed this code was not doing anything because `scanner` is null.

```
            // stop scanning when bluetooth is turned off
            if (adapterState != BluetoothAdapter.STATE_ON) {
                if (mBluetoothAdapter != null && mIsScanning) {
                    BluetoothLeScanner scanner = mBluetoothAdapter.getBluetoothLeScanner();
                    if (scanner != null) {
                        scanner.stopScan(getScanCallback());
                        mIsScanning = false;
                    }
                }
            }
```

therefore I changed it to this:

```
            // stop scanning when bluetooth is turned back on
            if (adapterState != BluetoothAdapter.STATE_ON) {
                if (mBluetoothAdapter != null && mIsScanning) {
                    BluetoothLeScanner scanner = mBluetoothAdapter.getBluetoothLeScanner();
                    if (scanner != null) {
                        scanner.stopScan(getScanCallback());
                        mIsScanning = false;
                    }
                }
            }
```

let me know if you have problems

open a new issue if you have problems.