chipweinberger/flutter_blue_plus #850: [Help]:

JMFUS, 2024-04-11:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.2

### Flutter Version

3.86.0

### What OS?

iOS

### OS Version

iOS17

### Bluetooth Module

Raspberry Pi Pico WH

### What is your problem?

Hello
Here is my code :
``` _isBluetoothOn() async {
    if (await FlutterBluePlus.isSupported == false) {
      print("Bluetooth not supported by this device");
      return;
    }
// handle bluetooth on & off
// note: for iOS the initial state is typically BluetoothAdapterState.unknown
// note: if you have permissions issues you will get stuck at BluetoothAdapterState.unauthorized
    var subscription =
        FlutterBluePlus.adapterState.listen((BluetoothAdapterState state) {
      print(state);
      if (state == BluetoothAdapterState.on) {
        setState(() {
          BTState = state;
        });
      } else {
        ShowDialogAndSnackbar().showAlertDialog(
            context, "Vous devez activer le Bluetooth".i18n(), "");
      }
    });
    // turn on bluetooth ourself if we can
    // for iOS, the user controls bluetooth enable/disable
    if (Platform.isAndroid) {
      await FlutterBluePlus.turnOn();
    }
    if (Platform.isIOS) {
      var request = await Permission.bluetooth.request();
    }
    // cancel to prevent duplicate listeners
    subscription.cancel();
  }

  Future<void> _scanDevices() async {
    print("======================== SCAN DEVICES ========================");

    var status = await Permission.bluetooth.status;
    if (status.isDenied) {
      print("======================== DENIED ========================");
    } else if (status.isGranted) {
      print("======================== GRANTED ========================");
    } else if (status.isPermanentlyDenied) {
      print(
          "======================== PERMANENTLY DENIED ========================");
    }
    // if (await Permission.bluetooth.isPermanentlyDenied) {
    if (await Permission.bluetooth.isDenied) {
      // If user has permanently denied permission, direct them to app settings
      openAppSettings();
    }
    var scanStatus = await Permission.bluetoothScan.request().isGranted;
    var bluetoothConnectStatus =
        await Permission.bluetoothConnect.request().isGranted;
    print("======================== SCAN STATUS :  $scanStatus");
    print(
        "======================== BT CONNECT STATUS :  $bluetoothConnectStatus");
    // if (!scanStatus || !bluetoothConnectStatus) {
    //   openAppSettings();
    // }
    if (scanStatus && bluetoothConnectStatus) {
      FlutterBluePlus.startScan(timeout: const Duration(seconds: 4));

      var subscription = FlutterBluePlus.scanResults.listen(
        (results) {
          for (ScanResult r in results) {
            print(
                "======================== RESULTS SCAN : ${r.device.advName}");
            if (r.device.advName.startsWith('OptiSun')) {
              var device = BluetoothDeviceWithState(
                  device: r.device, isAssociated: false, isConnected: false);
              print("======================== DEVICE : $device");
              if (!devices.contains(device)) {
                print(
                    "======================== NE CONTIENT PAS OptiSun ========================");
                setState(
                  () {
                    devices.add(device);
                  },
                );
              }
            }
          }
        },
      );
      // cleanup: cancel subscription when scanning stops
      FlutterBluePlus.cancelWhenScanComplete(subscription);
    }
  }```

If I test it on a Android device, all is OK, I can see my bluetooth devices and connect to them
But if I test on a Iphone X, I have this answer :
flutter: ======================== SCAN DEVICES ========================
flutter: ======================== DENIED ========================
flutter: ======================== SCAN STATUS :  false
flutter: ======================== BT CONNECT STATUS :  false

The bluetooth is of course enabled on my Iphone

I don't understand why

Thanks for your answer

### Logs

```shell
[ERROR:flutter/shell/platform/darwin/graphics/FlutterDarwinContextMetalImpeller.mm(42)] Using the Impeller rendering backend.
Connecting to VM Service at ws://127.0.0.1:57287/4uGaf3G3yUo=/ws
flutter: \^[[30m[Localization System] \^[[0m\^[[37mLoaded lib/assets/i18n/fr_FR.js<…>
flutter: ======================== BT ADAPTER STATE :  BluetoothAdapterState.off
flutter: ======================== SCAN DEVICES ========================
flutter: ======================== DENIED ========================
flutter: ======================== SCAN STATUS :  false
flutter: ======================== BT CONNECT STATUS :  false
```

chipweinberger, 2024-04-11:
you dont need to use permission. FBP will ask for permission for you.

closed 2024-04-11 by chipweinberger

SalmanNaghori, 2025-05-02:
> ### Requirements
> * [x]  I've looked at the README 'Common Problems' section
> 
> ### Have you checked this problem on the example app?
> Yes
> 
> ### FlutterBluePlus Version
> 1.32.2
> 
> ### Flutter Version
> 3.86.0
> 
> ### What OS?
> iOS
> 
> ### OS Version
> iOS17
> 
> ### Bluetooth Module
> Raspberry Pi Pico WH
> 
> ### What is your problem?
> Hello Here is my code :
> 
> ```
>     if (await FlutterBluePlus.isSupported == false) {
>       print("Bluetooth not supported by this device");
>       return;
>     }
> // handle bluetooth on & off
> // note: for iOS the initial state is typically BluetoothAdapterState.unknown
> // note: if you have permissions issues you will get stuck at BluetoothAdapterState.unauthorized
>     var subscription =
>         FlutterBluePlus.adapterState.listen((BluetoothAdapterState state) {
>       print(state);
>       if (state == BluetoothAdapterState.on) {
>         setState(() {
>           BTState = state;
>         });
>       } else {
>         ShowDialogAndSnackbar().showAlertDialog(
>             context, "Vous devez activer le Bluetooth".i18n(), "");
>       }
>     });
>     // turn on bluetooth ourself if we can
>     // for iOS, the user controls bluetooth enable/disable
>     if (Platform.isAndroid) {
>       await FlutterBluePlus.turnOn();
>     }
>     if (Platform.isIOS) {
>       var request = await Permission.bluetooth.request();
>     }
>     // cancel to prevent duplicate listeners
>     subscription.cancel();
>   }
> 
>   Future<void> _scanDevices() async {
>     print("======================== SCAN DEVICES ========================");
> 
>     var status = await Permission.bluetooth.status;
>     if (status.isDenied) {
>       print("======================== DENIED ========================");
>     } else if (status.isGranted) {
>       print("======================== GRANTED ========================");
>     } else if (status.isPermanentlyDenied) {
>       print(
>           "======================== PERMANENTLY DENIED ========================");
>     }
>     // if (await Permission.bluetooth.isPermanentlyDenied) {
>     if (await Permission.bluetooth.isDenied) {
>       // If user has permanently denied permission, direct them to app settings
>       openAppSettings();
>     }
>     var scanStatus = await Permission.bluetoothScan.request().isGranted;
>     var bluetoothConnectStatus =
>         await Permission.bluetoothConnect.request().isGranted;
>     print("======================== SCAN STATUS :  $scanStatus");
>     print(
>         "======================== BT CONNECT STATUS :  $bluetoothConnectStatus");
>     // if (!scanStatus || !bluetoothConnectStatus) {
>     //   openAppSettings();
>     // }
>     if (scanStatus && bluetoothConnectStatus) {
>       FlutterBluePlus.startScan(timeout: const Duration(seconds: 4));
> 
>       var subscription = FlutterBluePlus.scanResults.listen(
>         (results) {
>           for (ScanResult r in results) {
>             print(
>                 "======================== RESULTS SCAN : ${r.device.advName}");
>             if (r.device.advName.startsWith('OptiSun')) {
>               var device = BluetoothDeviceWithState(
>                   device: r.device, isAssociated: false, isConnected: false);
>               print("======================== DEVICE : $device");
>               if (!devices.contains(device)) {
>                 print(
>                     "======================== NE CONTIENT PAS OptiSun ========================");
>                 setState(
>                   () {
>                     devices.add(device);
>                   },
>                 );
>               }
>             }
>           }
>         },
>       );
>       // cleanup: cancel subscription when scanning stops
>       FlutterBluePlus.cancelWhenScanComplete(subscription);
>     }
>   }```
> 
> If I test it on a Android device, all is OK, I can see my bluetooth devices and connect to them
> But if I test on a Iphone X, I have this answer :
> flutter: ======================== SCAN DEVICES ========================
> flutter: ======================== DENIED ========================
> flutter: ======================== SCAN STATUS :  false
> flutter: ======================== BT CONNECT STATUS :  false
> 
> The bluetooth is of course enabled on my Iphone
> 
> I don't understand why
> 
> Thanks for your answer
> 
> ### Logs
> 
> ```shell
> [ERROR:flutter/shell/platform/darwin/graphics/FlutterDarwinContextMetalImpeller.mm(42)] Using the Impeller rendering backend.
> Connecting to VM Service at ws://127.0.0.1:57287/4uGaf3G3yUo=/ws
> flutter: \^[[30m[Localization System] \^[[0m\^[[37mLoaded lib/assets/i18n/fr_FR.js<…>
> flutter: ======================== BT ADAPTER STATE :  BluetoothAdapterState.off
> flutter: ======================== SCAN DEVICES ========================
> flutter: ======================== DENIED ========================
> flutter: ======================== SCAN STATUS :  false
> flutter: ======================== BT CONNECT STATUS :  false
> ```

Here is your [Solution](https://github.com/Baseflow/flutter-permission-handler/issues/1333#issuecomment-2539875175)

chipweinberger, 2025-05-02:
you dont need to use Permission library. FBP will ask for permission for you.