chipweinberger/flutter_blue_plus #855: [Help]:

eslamabdelaal, 2024-04-17:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.2

### Flutter Version

3.19.0

### What OS?

Android

### OS Version

android 12 

### Bluetooth Module

HC-08

### What is your problem?

When running the following code, errors appear as shown below
`import 'dart:convert';

import 'package:flutter/material.dart';
import '/control_button.dart';
import 'package:flutter_blue_plus/flutter_blue_plus.dart';
import 'package:permission_handler/permission_handler.dart';

void main() async {
  runApp(const MyApp());
}

class MyApp extends StatelessWidget {
  const MyApp({Key? key}) : super(key: key);

  @override
  Widget build(BuildContext context) {
    return const MaterialApp(
      title: 'Flutter BLE Demo',
      debugShowCheckedModeBanner: false,
      home: MainScreen(),
    );
  }
}

class MainScreen extends StatefulWidget {
  const MainScreen({Key? key}) : super(key: key);

  @override
  State<MainScreen> createState() => _MainScreenState();
}

class _MainScreenState extends State<MainScreen> {
  final FlutterBluePlus flutterBlue = FlutterBluePlus.instance;
  final List<BluetoothDevice> _devicesList = [];
  List<BluetoothService>? bluetoothServices;
  List<ControlButton> controlButtons = [];
  String? readableValue;

  @override
[... 296 lines cut ...]
	]
},{
	"resource": "/c:/Users/lomaux/Desktop/flutter_ble-main/lib/main.dart",
	"owner": "_generated_diagnostic_collection_name_#0",
	"code": {
		"value": "deprecated_member_use",
		"target": {
			"$mid": 1,
			"path": "/diagnostics/deprecated_member_use",
			"scheme": "https",
			"authority": "dart.dev"
		}
	},
	"severity": 2,
	"message": "'name' is deprecated and shouldn't be used. Use platformName instead.\nTry replacing the use of the deprecated member with the replacement.",
	"source": "dart",
	"startLineNumber": 89,
	"startColumn": 30,
	"endLineNumber": 89,
	"endColumn": 34,
	"tags": [
		2
	]
}]
```

chipweinberger, 2024-04-17:
dont use `.instance`

chipweinberger, 2024-04-17:
look at example app

closed 2024-04-17 by chipweinberger