chipweinberger/flutter_blue_plus #859: [Help]: Can't connect again after call device.disconnect.  (FlutterBluePlusException | requestMtu | fbp-code: 1 | Timed out after 15s)

mtiendat, 2024-04-19:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.2

### Flutter Version

3.19.5

### What OS?

Android

### OS Version

Android 10

### Bluetooth Module

nordic nRF5340

### What is your problem?

When I try to connect again after I call device.disconnect. I always got error `FlutterBluePlusException (FlutterBluePlusException | requestMtu | fbp-code: 1 | Timed out after 15s) `. I try to reload /rebuild app it only will be connected in the first time.

### Logs

```shell
D/[FBP-Android](14172): [FBP] onMethodCall: turnOn
D/[FBP-Android](14172): [FBP] onMethodCall: getBondedDevices
D/[FBP-Android](14172): [FBP] onMethodCall: turnOn
D/[FBP-Android](14172): [FBP] onMethodCall: connect
D/[FBP-Android](14172): [FBP] already connected
D/[FBP-Android](14172): [FBP] onMethodCall: requestMtu
D/BluetoothGatt(14172): configureMTU() - device: FE:36:5D:33:4D:04 mtu: 255
```

mtiendat, 2024-04-19:
other behavior: if I turn off BLE device, turn on again, reconnect will be work

chipweinberger, 2024-04-19:
sounds like a problem with your device.

1. try using `connect(mtu:null)`, then call `await device.requestMtu(512)` yourself, with larger timeout.

chipweinberger, 2024-04-23:
did you try?

chipweinberger, 2024-04-24:
no response

closed 2024-04-24 by chipweinberger

mtiendat, 2024-05-09:
sorry, missing your response email. It's working