chipweinberger/flutter_blue_plus #861: [Issue]: Missing Plugin exception when the app is in background. Using home screen widget to start the scan

rohanots, 2024-04-23:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.4

### Flutter Version

3.19.1

### What OS?

iOS

### OS Version

17.3.1

### Bluetooth Module

Esp -32

### What is your problem?

I am using the Home screen widget to start the scan for nearby BLE. Working fine on android but giving error on iOS devices. Checked on multiple devices. The problem occurs when we put the application in background.

### Logs

```shell
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: MissingPluginException(No implementation found for method flutterHotRestart on channel flutter_blue_plus/methods)
#0      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:332:7)
<asynchronous suspension>
#1      FlutterBluePlus._initFlutterBluePlus (package:flutter_blue_plus/src/flutter_blue_plus.dart:416:10)
<asynchronous suspension>
```

chipweinberger, 2024-04-23:
did you look here: https://github.com/boskokg/flutter_blue_plus?tab=readme-ov-file#missingpluginexceptionno-implementation-found-for-method-xxxx-

rohanots, 2024-04-23:
Yes Sir, I did. Actually, this is happening because the invoke method is unable to perform a hot restart since the app is is in background.

Why my app is in background?
Because I am using home screen widget flutter and starting the BLE scan from there.

Log:
MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:332:7)
No implementation found for method flutterHotRestart

chipweinberger, 2024-04-23:
`flutterHotRestart` is called during *any* app launch, not just hot restart.

rohanots, 2024-04-23:
Is there any way we can scan and connect nearby BLE devices with iOS devices while the app is in background.

chipweinberger, 2024-04-23:
yes. implement this PR: https://github.com/boskokg/flutter_blue_plus/issues/846

https://github.com/pauldemarco/flutter_blue/pull/210

chipweinberger, 2024-04-24:
I cannot reproduce this problem btw. Starting from home screen works for me.

closed 2024-04-24 by chipweinberger

rohanots, 2024-04-25:
[image]

Using the home_widget package.
Code on button clicks.
`@pragma('vm:entry-point')
Future<void> interactiveCallback(Uri? uri) async {
// Set AppGroup Id. This is needed for iOS Apps to talk to their WidgetExtensions
  await HomeWidget.setAppGroupId(Constants.appGroupId);
  print("interactiveCallback ${uri?.host}");
// We check the host of the uri to determine which action should be triggered.
  if (uri?.host == 'increment') {
    await connectBTDevice();
  } else if (uri?.host == 'unlock') {
    await unlockDevice();
  }
}`

`connectBTDevice(){
 FlutterBluePlus.startScan(
      timeout: const Duration(seconds: 3),
    );
    
     for (var btDevice in scanResults) {
              if (btDevice.advertisementData.advName.split('-').last ==
                  tempDeviceId) {
                  await btDevice.connect();
      }       
}
`

chipweinberger, 2024-04-25:
Ah, I did not understand what you meant by Home Screen Widget. Thank you for the screenshot.

This is not something I know how to fix. Sorry. If you fix it, please open PR.

rohanots, 2024-04-25:
No i have not been able to fixed this yet.