chipweinberger/flutter_blue_plus #864: [Help]: Not able to perform indicate and writeWithoutResponse operation on IOS, working fine on Android

f22-anurag, 2024-04-24:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.31.0

### Flutter Version

3.19.3

### What OS?

iOS, macOS

### OS Version

17.4.1

### Bluetooth Module

Not sure

### What is your problem?

Not able to perform indicate and writeWithoutResponse operation on IOS, working fine on Android. Any suggestion on how to resolve that? I have also tried to downgrade the fluttter_blue_plus but not luck.

### Logs

```shell
flutter: flutter:
exception
(dart.async/schedule_microtask.dart:49:9)
was thrown:
setStatel) or
markNeedsBuild) called during build.
[FBP]
[FBP]
\^||1;30m<discoverServices>\^[0m args:
\^[|1;35mDDF76B9B-980F-BAA4-4C53-9DB740CAC0<>
\^[[1;30m[[ OnMtuChanged 11\^[[0m
FBP.
^|[1:30m<discoverServices>\^[[0m
result:
\^|[1;33m{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, success: 1,
error_string: success, error_code: 0, mtu: 2<>
result：
flutter:
[FBP]
\^|[1;33mtr<>
\^[[1;30m[[ OnMtuChanged 11\^[[0m
result: \^[[1;33m{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D,
flutter:
[FBPJ
OCACOID,
\^[[1;30m[[ OnDiscoveredServices ]I\^[[0m result: \^[[1;33m{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D,
0,
authenticated
c300,
oSACe, sUccess: 1, rervices: Essenote id: Der6898-9805-64-4C53-908740CAC10, characteristics: Kremote id: DDF76898-980F-BAA-4C53-9D8740CACO1D, descriptors: (remote_ id: DDF76898-980F-BAA-4C53-90574
characteristic uuid:
descriptor_uuid: 2901,
service_uuid: a002,
secondary_service_uuid: nulll,
characteristic_uuid: c300, properties: notify: 0,
indicate_encryption_required: 0, write: 0, read: 1, write without_response: 0, notify_encryption_required: 0,
broadcast: 0,
extended_properties: 0, indicate:
signed writes: 0}, service_uuid: a002,
secondary_service_uuid: null,
fremote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D,
descriptors: Kremote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, characteristic_uuid: c301, descriptor_uuid: 2901, service_uuid: a002,
secondary_service_uuid: nulll,
characteristic_uu
id: c301, properties:
{notify: 0, indicate_encryption_required:
0, write: 0,
read: 1, write_without_response:
0,
notify_encryption_required: 0,
broadcast: 0,
extended_properties: 0,
indicate: 6,
{remote id: DDF76B9B-980F-BAA4-4653-9DB74
OCACOID,
descriptors:
[remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D,
characteristic_uuid: c302,
descriptor_uuid: 2901,
service_uuid:
service uuid:
a082,
secondary service uuid: nult,
characteristic_uuid: C302,
write: 1, read: 0,
write_without_response: 0,
notify_encrypti
on_required: 0,
broadcast: 0, extended_properties:
indicate: 0,
authenticated_signed_writes: 0},
service uuid:
a002, secondary_service_uuid: null,
{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D,
descriptors: Kremote id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D,
characteristic_uuid: c303, descriptor_uuid:
2901, servi
ce_uuid: a002,
secondary_service_uuid: null}l, characteristic _uuid: c303, properties: {notify: 0,
indicate_encryption_required: 0, write:
0, read: 0, write_ without_response: 1, notify_encryption_required: 0, broadcast: 0, extended_properties: 0,
indicate: 0, authenticated_signed_writes: 03,
service_uuid: a002, secondar
y_service_uuid: null},
{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, descriptors:
characteristic_uuid:
c304, properties:
{notify: 0,
descriptors:
indicate_encryption_required: 0, write: 1, read: 0, write_without_response: 0, notify_encryption_required: 0, broadcast: 0, extended_properties: 0,
indicate: 0, authenticated_sign
ed writes:
0},
service_uuid: a002,
secondary_service_uuid: nulls, tremote_id: DDF76B9B-980F-BAA4-4053-9DB740CAC01D,
{notify:
1, indicate_encryption_required: 0, write:
0,
read: 0, write_without_response: 0,
notify_encryption_required: 0,
ast: 0, extended properties: 0, indicate: 8, authenticated signed writes: 07, service Quid: a002, Secondary service uuid.
secondary_service_uuid: null?l, characteristic_uuid: c305, properties:
extended_properties: 0, indicate: 0,
authenticated_signed_writes: 0,
secondary_service_uuid: null,
fremote id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D,
Kremote id:
DDF76B9B-980F-BAA4-4C53-9DB740CAC01D,
characteristic_uuid: c306, descriptor_uuid: 2902,
service_uuid: a002,
_service_uuid: null}l, characteristic uuid: c306, properties:
descriptors:
cast: 0, extended_properties:
0, indicate:
1,
authenticated
service_uuid: a002,
ses: notity: 0, indcateenuryption requared?
secondary_service_uuid: null',
secondarY remote. 2d: DDF76898-980F-BAA4-4C53-908740CAC01D, descriptors:
{notify: 0,
indicate_encryption_required: 0,
write: 0, read: 0,
write without_response: 0, notify_encryption_required: 0,
broad
Kremote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, characteristic_uuid: c307, descriptor_uuid: 2901, service_uuid: a002,
ary_service_uuid:
null}],
characteristic_uuid:
C307, properties:
{notify: 0,
0, write:
0, read:
1,
write without response: 0, notify encryption required: 0, broadcast:
0,
authenticated_signed writes: 0},
second
service_uuid: a002,
secondary_service_uuid: nully
is_primary:
true, service_uuid:
a002, included_services:
{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D,
characteristics: Kremote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, descriptors: Kremote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, characteristic_uuid: 400, descriptor_uuid: 2901, service_uuid: a003,
sec
ondary_service uuid:
nulli
characteristic
uuid:
C400, properties:
{notify: 0,
indicate_encryption_required: 0,
write: 0, read:
write without_response: 0, notify_encryption_required: 0, broadcast: 0, extended_properties: 0, indicate: 0,
authenticated_signed_writes: 0},
service_uuid: a003, secondary_service_uuid: nu
11},
fremote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D,
descriptors:
notify_encryption
Kremote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D,
characteristic_uuid: c401,
descriptor_uuid: 2901, service_uuid: a003,
secondary_service_uuid: null}l, characteristic _uuid: c401, properties: (notify: 0,
indicate_encryption_required: , write: 6, read:
1, write_without_response: 0,
_required: 0,
broadcast: 0, extended_properties: 0, indicate: 0, authenticated_signed writes: 0}, service_uuid: a003, secondary_service_uuid: null}l, is_primary: true, service_uuid: a003, included _services: [1}1, error_string: success, error_code: <>
flutter: The characteristic 303 supports write without response flutter: The characteristic flutter:
[FBP]
c303 supports write without
response
\^[[1;30m<setNotifyValue>\^[[0m args:
\^|[1;35m{remote_id: DDF769B-980F-BAA4-4C53-9DB740CAC01D, service_uuid: a002, secondary_service_uuid: null, characteristic_uuid: 306, force_indications: false, enable: tru<>
[FBP]
\^[[1:30m<setNotifyValue>\^[[0m
[FBPJ
\^[[1:30m[[
result: \^[[1;33mtr<.>
OnDescriptorWritten 11\^[0m result: \*[1;33merror_string: success, remote_id: DDF769B-980F-BAA4-4C53-9DB740CAC®1D, secondary_service _uuid: null, characteristic uuid: 306, success: 1, value: 02000000, error_code: 0, service_uuid: a002, descriptor _uuid: 290<>
Lobject,
11,
exception
was thrown:
setState)
or markNeedsBuild)
called during build.
Another
exception
was thrown:
setState() or markNeedsBuild()
called during build.
flutter:
[FBPJ
flutter:
[FBP]
111;30m<setNotifyValue>/"LLom args:
111;35m remote_id: DDF7689B-980F-BAA4-4053-9D8740CAC01D, service_uuid: aboz, secondary_service_uuid: null, characteristic uuid: 306, force_indications: false, enable: tru<>
[FBP]
\^||1;30m<setNotifyValue>\^|[0m
result: \^|[1;33mtr>
\^|[1;30mlL OnDescriptorWritten 11\^[0m result:
true,
the
verification,
1*11;33m/error_string: success, remote_id: DDF769B-980F-BAA4-4C53-9DB740CAC01D, secondary_service_uuid: null, characteristic _uuid: c306, success: 1, value: 02000000, error_code: 0, service_uuid: a002, descriptor_uuid: 290<>
isNotifying from
[FBPJ
\^111:30m<writeCharacteristic\
[1,
str}
~LIom args:
\^|[1;35m{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, service_uuid: a002, secondary_service_uuid: null, characteristic uuid: c303, write_type: 1, allow_long_write: 0, value: 3939393<>
flutter: flutter:
[FBP]
^ 1:30m<writeCharacteristic^0m
result: 1^||1;33mtr<>
[FBP]
\^|1;30mll OnCharacteristicWritten 1|\^|0m result: \^||1;33m/remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D,
secondary_service_uuid: null, characteristic_uuid: c303, success: 1, value: , error_code: 0, service_uuid: a002, error_string:
```

chipweinberger, 2024-04-24:
I dont understand what that means

chipweinberger, 2024-04-24:
1. you need better logs. why does it look all bad?
2. provide logs for both android & ios

f22-anurag, 2024-04-25:
Iphone Logs:

flutter: [FBP] \^[[1;30m[[ OnMtuChanged ]]\^[[0m result: \^[[1;33m{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, success: 1, error_string: success, error_code: 0, mtu: 2<…>
flutter: [FBP] \^[[1;30m<discoverServices>\^[[0m result: \^[[1;33mtr<…>
flutter: [FBP] \^[[1;30m[[ OnMtuChanged ]]\^[[0m result: \^[[1;33m{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, success: 1, error_string: success, error_code: 0, mtu: 51<…>
flutter: [FBP] \^[[1;30m[[ OnDiscoveredServices ]]\^[[0m result: \^[[1;33m{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, success: 1, services: [{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, characteristics: [{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, descriptors: [{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, characteristic_uuid: c300, descriptor_uuid: 2901, service_uuid: a002, secondary_service_uuid: null}], characteristic_uuid: c300, properties: {notify: 0, indicate_encryption_required: 0, write: 0, read: 1, write_without_response: 0, notify_encryption_required: 0, broadcast: 0, extended_properties: 0, indicate: 0, authenticated_signed_writes: 0}, service_uuid: a002, secondary_service_uuid: null}, {remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, descriptors: [{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, characteristic_uuid: c301, descriptor_uuid: 2901, service_uuid: a002, secondary_service_uuid: null}], characteristic_uuid: c301, properties: {notify: 0, indicate_encryption_required: 0, write: 0, read: 1, write_without_response: 0, notify_encryption_required: 0, broadcast: 0, extended_properties: 0, indicate: 0, authenticated_signed_writes: 0}, service_uuid: a002, secondary_service_uuid: null}, {remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, descriptors: [{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, characteristic_uuid: c302, descriptor_uuid: 2901, service_uuid: a002, secondary_service_uuid: null}], characteristic_uuid: c302, properties: {notify: 0, indicate_encryption_required: 0, write: 1, read: 0, write_without_response: 0, notify_encryption_required: 0, broadcast: 0, extended_properties: 0, indicate: 0, authenticated_signed_writes: 0}, service_uuid: a002, secondary_service_uuid: null}, {remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, descriptors: [{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, characteristic_uuid: c303, descriptor_uuid: 2901, service_uuid: a002, secondary_service_uuid: null}], characteristic_uuid: c303, properties: {notify: 0, indicate_encryption_required: 0, write: 0, read: 0, write_without_response: 1, notify_encryption_required: 0, broadcast: 0, extended_properties: 0, indicate: 0, authenticated_signed_writes: 0}, service_uuid: a002, secondary_service_uuid: null}, {remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, descriptors: [], characteristic_uuid: c304, properties: {notify: 0, indicate_encryption_required: 0, write: 1, read: 0, write_without_response: 0, notify_encryption_required: 0, broadcast: 0, extended_properties: 0, indicate: 0, authenticated_signed_writes: 0}, service_uuid: a002, secondary_service_uuid: null}, {remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, descriptors: [{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, characteristic_uuid: c305, descriptor_uuid: 2902, service_uuid: a002, secondary_service_uuid: null}], characteristic_uuid: c305, properties: {notify: 1, indicate_encryption_required: 0, write: 0, read: 0, write_without_response: 0, notify_encryption_required: 0, broadcast: 0, extended_properties: 0, indicate: 0, authenticated_signed_writes: 0}, service_uuid: a002, secondary_service_uuid: null}, {remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, descriptors: [{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, characteristic_uuid: c306, descriptor_uuid: 2902, service_uuid: a002, secondary_service_uuid: null}], characteristic_uuid: c306, properties: {notify: 0, indicate_encryption_required: 0, write: 0, read: 0, write_without_response: 0, notify_encryption_required: 0, broadcast: 0, extended_properties: 0, indicate: 1, authenticated_signed_writes: 0}, service_uuid: a002, secondary_service_uuid: null}, {remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, descriptors: [{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, characteristic_uuid: c307, descriptor_uuid: 2901, service_uuid: a002, secondary_service_uuid: null}], characteristic_uuid: c307, properties: {notify: 0, indicate_encryption_required: 0, write: 0, read: 1, write_without_response: 0, notify_encryption_required: 0, broadcast: 0, extended_properties: 0, indicate: 0, authenticated_signed_writes: 0}, service_uuid: a002, secondary_service_uuid: null}], is_primary: true, service_uuid: a002, included_services: []}, {remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, characteristics: [{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, descriptors: [{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, characteristic_uuid: c400, descriptor_uuid: 2901, service_uuid: a003, secondary_service_uuid: null}], characteristic_uuid: c400, properties: {notify: 0, indicate_encryption_required: 0, write: 0, read: 1, write_without_response: 0, notify_encryption_required: 0, broadcast: 0, extended_properties: 0, indicate: 0, authenticated_signed_writes: 0}, service_uuid: a003, secondary_service_uuid: null}, {remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, descriptors: [{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, characteristic_uuid: c401, descriptor_uuid: 2901, service_uuid: a003, secondary_service_uuid: null}], characteristic_uuid: c401, properties: { [... cut ...]
flutter: [FBP] \^[[1;30m<writeCharacteristic>\^[[0m args: \^[[1;35m{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, service_uuid: a002, secondary_service_uuid: null, characteristic_uuid: c303, write_type: 1, allow_long_write: 0, value: 3939393<…>
flutter: [FBP] \^[[1;30m<writeCharacteristic>\^[[0m result: \^[[1;33mtr<…>
flutter: [FBP] \^[[1;30m[[ OnCharacteristicWritten ]]\^[[0m result: \^[[1;33m{remote_id: DDF76B9B-980F-BAA4-4C53-9DB740CAC01D, secondary_service_uuid: null, characteristic_uuid: c303, success: 1, value: , error_code: 0, service_uuid: a002, error_string: succes<…>

chipweinberger, 2024-04-25:
thanks. The logs are very different, hard to tell what is supposed to happen.

chipweinberger, 2024-04-30:
yes could be an mtu problem

closed 2024-04-30 by chipweinberger