chipweinberger/flutter_blue_plus #866: [Help]: How to use write indicate

GMarki, 2024-04-29:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.26.4

### Flutter Version

3.16.8

### What OS?

Android

### OS Version

Android 12

### Bluetooth Module

Device OnePlus 9

### What is your problem?

I should create a subscription callback for last value for the characteristic I'm going to write, and then set notifyvalue = true and finally write to characteristic. The write indicate value will then be printed in the subscription callback, correct?
```dart
Map<String, BluetoothCharacteristic> _characteristics = {};

.... //fill characteristics

final subscription =
          _characteristics["UserControlPoint"]?.lastValueStream.listen((value) {
        print('subscription UserControlPoint value : $value');
      });
await _characteristics["UserControlPoint"]?.setNotifyValue(true);
await _characteristics["UserControlPoint"]?.write(1);

### Logs

```shell
no logs
```

chipweinberger, 2024-04-29:
I don't know what write indicate means. Please read these:

[Novel Bits BLE Tutorial](https://novelbits.io/bluetooth-low-energy-ble-complete-guide/)
[All About Circuits BLE Tutorial](https://www.allaboutcircuits.com/technical-articles/exploring-the-basics-of-bluetooth-low-energy-a-beginners-guide-to-ble/)
[Embetronicx BLE Tutorial](https://embetronicx.com/tutorials/tech_devices/bluetooth-low-energy-ble-introduction-part-1/)

chipweinberger, 2024-04-29:
but yes, lastValueStream is updated for writes. It says so in the documentation.