chipweinberger/flutter_blue_plus #867: [Issue]:  GATT_ATTR_NOT_FOUND (10)

kalismeras61, 2024-04-30:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.2

### Flutter Version

3.19.5 

### What OS?

Android

### OS Version

Android 14

### Bluetooth Module

Device it self

### What is your problem?

I have one HR strap device ( Coros HR Strap) and it is working fine with device app and also Polar APP ( most used one). Also i can able to connect with other straps so assume my implementation works fine. But when i try to connect my own app it is returning below error and not connecting. It works fine on IOS but android have an issue. 

E/[FBP-Android](22643): [FBP] onDescriptorWrite:
E/[FBP-Android](22643): [FBP]   chr: 2a05
E/[FBP-Android](22643): [FBP]   desc: 2902
E/[FBP-Android](22643): [FBP]   status: GATT_ATTR_NOT_FOUND (10)
I/flutter (22643): [FBP] [[ OnDescriptorWritten ]] result: {error_string: GATT_ATTR_NOT_FOUND, service_uuid: 1801, success: 0, remote_id: EA:6A:17:FB:E8:97, descriptor_uuid: 2902, error_code: 10, characteristic_uuid: 2a05, value: 0200}

### Logs

```shell
E/[FBP-Android](22643): [FBP] onDescriptorWrite:
E/[FBP-Android](22643): [FBP]   chr: 2a05
E/[FBP-Android](22643): [FBP]   desc: 2902
E/[FBP-Android](22643): [FBP]   status: GATT_ATTR_NOT_FOUND (10)
I/flutter (22643): [FBP] [[ OnDescriptorWritten ]] result: {error_string: GATT_ATTR_NOT_FOUND, service_uuid: 1801, success: 0, remote_id: EA:6A:17:FB:E8:97, descriptor_uuid: 2902, error_code: 10, characteristic_uuid: 2a05, value: 0200}
```

chipweinberger, 2024-04-30:
that's the service changed characteristic 

you can set subscribeToServicesChangedCharacteristic to false in discoverServices 

thanks for reporting

kalismeras61, 2024-05-01:
It is fixed. Thanks

closed 2024-05-01 by kalismeras61