chipweinberger/flutter_blue_plus #869: [Help]:

daniloapr, 2024-05-02:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.29.8

### Flutter Version

3.19.6

### What OS?

All

### OS Version

iOS 1.17

### Bluetooth Module

Samsung S10, iPhone 12, iPad 9th Gen

### What is your problem?

I'm advertising data from one device and reading it in another device using the flutter_blue_plus library. The problem is that those read devices are kept cached. For example: if I change the advertising name, I keep seeing the old name in the scaning process. Is it something in the library or in the OS (Android, iOS) themselves? I coul reproduce it on Android, iPhone and iPad.

Moreover, I'm having difficulties to find advertised services from an iOS device. I'm doing it with a fork from flutter_ble_peripheral. I know this is not supported by this library, but would you have any guesses on why it's happening? Here is the advertising code:

```swift
    private func startPeripheral(_ call: FlutterMethodCall, _ result: @escaping FlutterResult) {
        let map = call.arguments as? Dictionary<String, Any>
  
        let serviceUuidStr = map?["serviceUuid"] as? String ?? ""
        let localName = map?["localName"] as? String ?? "iOS"
        
        let characteristicUuid = CBUUID(string: "251fd7c0-3f1a-4115-a10b-f809f45ad5f1")
        let serviceUuid = CBUUID(string: serviceUuidStr)
        let characteristic = CBMutableCharacteristic(
          type: characteristicUuid,
          properties: [.notify, .indicate, .broadcast, .read],
          value: nil,
          permissions: .readable
        )
        
        let service = CBMutableService(type: serviceUuid, primary: true)
        service.characteristics = [characteristic]
        
        self.peripheralManager = CBPeripheralManager(
            delegate: self,
            queue: nil,
            options: [CBPeripheralManagerOptionShowPowerAlertKey : true]
        )
        self.peripheralManager.add(service)
        self.characteristic = characteristic
        
        self.peripheralManager.startAdvertising([
            CBAdvertisementDataLocalNameKey: localName,
            CBAdvertisementDataServiceUUIDsKey: [serviceUuid]
            ])
        
        result(nil)
    }
```

### Logs

```shell
.
```

chipweinberger, 2024-05-02:
caching happens on the platform level

but you didn't specify which name you are using

daniloapr, 2024-05-02:
@chipweinberger I'm advertising the `CBAdvertisementDataLocalNameKey`. However, I noticed that it affects both `platform name` and `advertised name`. Thinking better about it, any name in my case would be fine, as long as the serviceUUID shows up, something I couldn't do so far. So I wonder if the cache could affect that as well (just keeping the first advertised service)

chipweinberger, 2024-05-02:
platformName is cached by the platform. we don't control is

advName is controlled by FBP, and is updated after every new advertisement

closed 2024-05-06 by chipweinberger