chipweinberger/flutter_blue_plus #870: [Help]: Slow packet transmission time using allowLongWrite

max-h-silverman, 2024-05-02:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.4

### Flutter Version

3.19.1

### What OS?

Android

### OS Version

Android 12

### Bluetooth Module

Unsure (Phomemo M220 thermal printer)

### What is your problem?

When sending image data to the printer using FBP the transmission takes about 3s (about 100ms per 256 bytes, roughly 30 packets). When sending the exact same data using the native Phomemo PrintMaster app, the transmission takes about 1/10th as long.

I've spent a few days digging into this to try and understand if my implementation isn't right, but it ultimately seems that the delay is happening in Flutter_Blue_plus.java. (logs below show a few of the calls.

I added detailed logging at one point to see more granular detail about where the time is being taken, and the relevant log capture is attached.

I retrieved the HCI Log from the device to confirm that the same amount of data is being transmitted by PrintMaster and by my app.

Appreciate any help you can offer.

[image]

### Logs

```shell
2024-05-01 22:29:16.237 25074-25074 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onMethodCall: discoverServices
2024-05-01 22:29:16.237 25074-25074 BluetoothGatt           com.flutterflow.mycofile             D  discoverServices() - device: 04:7F:0E:AC:C9:0F
2024-05-01 22:29:16.240 25074-25189 BluetoothGatt           com.flutterflow.mycofile             D  onSearchComplete() = Device=04:7F:0E:AC:C9:0F Status=0
2024-05-01 22:29:16.241 25074-25189 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onServicesDiscovered:
2024-05-01 22:29:16.241 25074-25189 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   count: 7
2024-05-01 22:29:16.241 25074-25189 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   status: 0GATT_SUCCESS
2024-05-01 22:29:16.261 25074-25134 flutter                 com.flutterflow.mycofile             I  Data sent (hex): 1b 40 1d 76 30 00 28 00 b7 00
2024-05-01 22:29:16.265 25074-25074 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onMethodCall: writeCharacteristic
2024-05-01 22:29:16.394 25074-25189 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onCharacteristicWrite:
2024-05-01 22:29:16.395 25074-25189 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   chr: 49535343-6daa-4d02-abf6-19569aca69fe
2024-05-01 22:29:16.395 25074-25189 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   status: GATT_SUCCESS (0)
2024-05-01 22:29:16.402 25074-25074 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onMethodCall: discoverServices
2024-05-01 22:29:16.402 25074-25074 BluetoothGatt           com.flutterflow.mycofile             D  discoverServices() - device: 04:7F:0E:AC:C9:0F
2024-05-01 22:29:16.405 25074-25189 BluetoothGatt           com.flutterflow.mycofile             D  onSearchComplete() = Device=04:7F:0E:AC:C9:0F Status=0
2024-05-01 22:29:16.405 25074-25189 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onServicesDiscovered:
2024-05-01 22:29:16.405 25074-25189 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   count: 7
2024-05-01 22:29:16.405 25074-25189 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   status: 0GATT_SUCCESS
2024-05-01 22:29:16.422 25074-25134 flutter                 com.flutterflow.mycofile             I  Data sent (hex): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
2024-05-01 22:29:16.425 25074-25074 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onMethodCall: writeCharacteristic
2024-05-01 22:29:16.665 25074-25189 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onCharacteristicWrite:
2024-05-01 22:29:16.666 25074-25189 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   chr: 49535343-6daa-4d02-abf6-19569aca69fe
2024-05-01 22:29:16.666 25074-25189 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   status: GATT_SUCCESS (0)
2024-05-01 22:29:16.671 25074-25074 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onMethodCall: discoverServices
2024-05-01 22:29:16.672 25074-25074 BluetoothGatt           com.flutterflow.mycofile             D  discoverServices() - device: 04:7F:0E:AC:C9:0F
2024-05-01 22:29:16.676 25074-25189 BluetoothGatt           com.flutterflow.mycofile             D  onSearchComplete() = Device=04:7F:0E:AC:C9:0F Status=0
[... 34 lines cut ...]
2024-05-01 22:29:17.667 25074-25189 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   count: 7
2024-05-01 22:29:17.667 25074-25189 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   status: 0GATT_SUCCESS
2024-05-01 22:29:17.696 25074-25134 flutter                 com.flutterflow.mycofile             I  Data sent (hex): 00 00 00 00 1e 00 00 00 00 00 00 00 00 00 00 00 00 00 1f ff e0 00 00 00 00 3c 00 00 00 00 ff ff c0 00 00 00 00 00 ff f0 00 00 00 00 1e 00 00 00 00 00 00 00 00 00 00 00 00 00 1f ff e0 00 00 00 00 3c 00 00 00 00 ff ff c0 00 00 00 00 01 ff f8 00 00 00 00 1e 00 00 00 00 00 00 00 00 00 00 00 00 00 1f 01 e0 00 00 00 00 3c 00 00 00 00 01 e0 00 00 00 00 00 07 c0 f8 00 00 00 00 1e 00 00 00 00 00 00 00 00 00 00 00 00 00 1f 01 f0 0f 80 03 c0 3c 3e 78 1e 00 01 e0 03 e0 1e 3c 00 07 c0 7c 78 1f 07 c0 ff c0 7c 07 ce 00 00 00 00 00 00 00 00 00 1f 01 f0 3f e0 1f f0 3c 3e 78 1e 00 01 e0 0f f8 1e fe 00 0f c0 3c 78 1e 0f f8 ff c0 ff 07 fe 00 00 00 00 00 00 00 00 00 1f 01 f0 7f f0 3f f8 3c 7c 78 1e 00 01 e0 0f fc 1f ff 00 0f 80 3c 3c 1e 1f fc ff c3 ff 87 fe 00 00 00 00 00 00 00
2024-05-01 22:29:17.698 25074-25074 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onMethodCall: writeCharacteristic
2024-05-01 22:29:18.015 25074-25089 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onCharacteristicWrite:
2024-05-01 22:29:18.016 25074-25089 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   chr: 49535343-6daa-4d02-abf6-19569aca69fe
2024-05-01 22:29:18.016 25074-25089 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   status: GATT_SUCCESS (0)
2024-05-01 22:29:18.022 25074-25074 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onMethodCall: discoverServices
2024-05-01 22:29:18.022 25074-25074 BluetoothGatt           com.flutterflow.mycofile             D  discoverServices() - device: 04:7F:0E:AC:C9:0F
2024-05-01 22:29:18.027 25074-25089 BluetoothGatt           com.flutterflow.mycofile             D  onSearchComplete() = Device=04:7F:0E:AC:C9:0F Status=0
2024-05-01 22:29:18.027 25074-25089 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onServicesDiscovered:
2024-05-01 22:29:18.027 25074-25089 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   count: 7
2024-05-01 22:29:18.028 25074-25089 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   status: 0GATT_SUCCESS
2024-05-01 22:29:18.060 25074-25134 flutter                 com.flutterflow.mycofile             I  Data sent (hex): 00 00 1f 01 e0 ff f0 3f f8 3c fc 3c 3e 00 01 e0 1f fe 1f ff 00 0f 80 3c 3c 1e 3f fe 7f c3 ff c7 fe 00 00 00 00 00 00 00 00 00 1f 01 e0 f8 f8 3c 7c 3c f8 3c 3e 00 01 e0 3e 3e 1f 8f 80 0f 80 3e 3c 3c 3c 3e 1e 07 c3 c7 fe 00 00 00 00 00 00 00 00 00 1f ff e1 e0 7c 78 1c 3f e0 1e 3c 00 01 e0 78 1f 1f 07 80 0f 80 3e 1e 3c 3c 00 1e 0f 03 e7 c0 00 00 00 00 00 00 00 00 00 1f ff c1 e0 3c 78 00 3f c0 1e 3c 00 01 e0 78 0f 1e 07 80 0f 80 3e 1e 7c 3e 00 1e 0f 03 e7 c0 00 00 00 00 00 00 00 00 00 1f ff 81 e0 3c 78 00 3f c0 [... cut ...]
2024-05-01 22:29:18.341 25074-25089 BluetoothGatt           com.flutterflow.mycofile             D  onSearchComplete() = Device=04:7F:0E:AC:C9:0F Status=0
2024-05-01 22:29:18.341 25074-25089 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onServicesDiscovered:
2024-05-01 22:29:18.341 25074-25089 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   count: 7
2024-05-01 22:29:18.342 25074-25089 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   status: 0GATT_SUCCESS
```

chipweinberger, 2024-05-02:
FBP only allows a single transaction in progress at one time.

This is for best reliability on all android versions and phones.

this locking happens in dart code.

you can try to remove this lock, or move the lock to native code

this will require careful coding tho.

if you manage it, please open a pr

chipweinberger, 2024-05-02:
however a bigger issue - why are you discovering services so much?

that will destroy performance

max-h-silverman, 2024-05-02:
Ok thanks. I'm using a third party package written for the Phomemo printer and the "sendData" function doesn't accept the characteristic as a parameter. I was able to modify the package locally to send the previously discovered write characteristic but it didn't change the overall time to complete the print.

Which part of the writeCharacteristic function is performing the lock?

Appreciate the quick response.

chipweinberger, 2024-05-02:
you need to not discover services after every write. that is your problem

max-h-silverman, 2024-05-02:
Sorry what I was trying to say in the last note is that's what I tried but didn't see any performance increase. Below are the first few logs from that attempt. You can see the large time gap between "[FBP] onMethodCall: writeCharacteristic" and "[FBP] onCharacteristicWrite:" with no discovery happening after the initial one.

2024-05-02 10:49:41.732 32655-32655 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onMethodCall: discoverServices
2024-05-02 10:49:41.732 32655-32655 BluetoothGatt           com.flutterflow.mycofile             D  discoverServices() - device: 04:7F:0E:AC:C9:0F
2024-05-02 10:49:41.741 32655-32694 BluetoothGatt           com.flutterflow.mycofile             D  onSearchComplete() = Device=04:7F:0E:AC:C9:0F Status=0
2024-05-02 10:49:41.742 32655-32694 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onServicesDiscovered:
2024-05-02 10:49:41.742 32655-32694 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   count: 7
2024-05-02 10:49:41.742 32655-32694 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   status: 0GATT_SUCCESS
2024-05-02 10:49:42.084 32655-32655 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onMethodCall: writeCharacteristic
2024-05-02 10:49:42.170 32655-32694 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onCharacteristicWrite:
2024-05-02 10:49:42.171 32655-32694 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   chr: 49535343-6daa-4d02-abf6-19569aca69fe
2024-05-02 10:49:42.171 32655-32694 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   status: GATT_SUCCESS (0)
2024-05-02 10:49:42.175 32655-32754 flutter                 com.flutterflow.mycofile             I  Data sent successfully
2024-05-02 10:49:42.178 32655-32655 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onMethodCall: writeCharacteristic
2024-05-02 10:49:42.441 32655-32694 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onCharacteristicWrite:
2024-05-02 10:49:42.442 32655-32694 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   chr: 49535343-6daa-4d02-abf6-19569aca69fe
2024-05-02 10:49:42.442 32655-32694 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   status: GATT_SUCCESS (0)
2024-05-02 10:49:42.450 32655-32754 flutter                 com.flutterflow.mycofile             I  Data sent successfully
2024-05-02 10:49:42.453 32655-32655 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onMethodCall: writeCharacteristic
2024-05-02 10:49:42.710 32655-32694 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onCharacteristicWrite:
2024-05-02 10:49:42.711 32655-32694 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   chr: 49535343-6daa-4d02-abf6-19569aca69fe
2024-05-02 10:49:42.711 32655-32694 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   status: GATT_SUCCESS (0)
2024-05-02 10:49:42.716 32655-32754 flutter                 com.flutterflow.mycofile             I  Data sent successfully
2024-05-02 10:49:42.719 32655-32655 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onMethodCall: writeCharacteristic
2024-05-02 10:49:42.981 32655-32694 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onCharacteristicWrite:
2024-05-02 10:49:42.981 32655-32694 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   chr: 49535343-6daa-4d02-abf6-19569aca69fe
2024-05-02 10:49:42.982 32655-32694 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   status: GATT_SUCCESS (0)
2024-05-02 10:49:42.988 32655-32754 flutter                 com.flutterflow.mycofile             I  Data sent successfully
2024-05-02 10:49:42.991 32655-32655 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onMethodCall: writeCharacteristic
2024-05-02 10:49:43.251 32655-32694 [FBP-Android]           com.flutterflow.mycofile             D  [FBP] onCharacteristicWrite:
2024-05-02 10:49:43.251 32655-32694 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   chr: 49535343-6daa-4d02-abf6-19569aca69fe
2024-05-02 10:49:43.251 32655-32694 [FBP-Android]           com.flutterflow.mycofile             D  [FBP]   status: GATT_SUCCESS (0)
2024-05-02 10:49:43.256 32655-32754 flutter                 com.flutterflow.mycofile             I  Data sent successfully

chipweinberger, 2024-05-03:
you should also increase the android PHY speed

closed 2024-05-03 by max-h-silverman

max-h-silverman, 2024-06-14:
Hey I wasn't able to identify the code responsible for this lock in writeCharacteristic. Can you point me to it? Thanks.