chipweinberger/flutter_blue_plus #872: [Help]: System device scan returns empty

Flucadetena, 2024-05-04:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.4

### Flutter Version

3.19.6

### What OS?

macOS

### OS Version

14.4.1

### Bluetooth Module

BCM_4387

### What is your problem?

The system device list comes empty, even though the scan returns devices.

### Logs

```shell
flutter: [FBP] [[ OnScanResponse ]] result: {advertisements: [{remote_id: 3E689256-0832-1EC0-58A1-9D23C8F5E664, manufacturer_data: {76: 1006471d72bf5f78}, rssi: -92, connectable: 1, tx_power_level: 17}]}
2024-05-04 17:56:07.670 blue_control[12788:260721] [FBP-iOS] centralManager didDiscoverPeripheral
flutter: [FBP] [[ OnScanResponse ]] result: {advertisements: [{remote_id: 3E689256-0832-1EC0-58A1-9D23C8F5E664, manufacturer_data: {76: 1006471d72bf5f78}, rssi: -91, connectable: 1, tx_power_level: 17}]}
2024-05-04 17:56:08.548 blue_control[12788:260721] [FBP-iOS] centralManager didDiscoverPeripheral
flutter: [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -96, remote_id: 0686C8D0-BA40-3735-638E-C701CDEF1BD8, manufacturer_data: {117: 420401806604b9e31fdb1706b9e31fdb1601000000000000}}]}
2024-05-04 17:56:08.855 blue_control[12788:260721] [FBP-iOS] centralManager didDiscoverPeripheral
flutter: [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -90, remote_id: 0686C8D0-BA40-3735-638E-C701CDEF1BD8, manufacturer_data: {117: 420401806604b9e31fdb1706b9e31fdb1601000000000000}}]}
2024-05-04 17:56:09.477 blue_control[12788:260721] [FBP-iOS] centralManager didDiscoverPeripheral
flutter: [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -93, remote_id: 0686C8D0-BA40-3735-638E-C701CDEF1BD8, manufacturer_data: {117: 420401806604b9e31fdb1706b9e31fdb1601000000000000}}]}
2024-05-04 17:56:10.070 blue_control[12788:260721] [FBP-iOS] centralManager didDiscoverPeripheral
2024-05-04 17:56:10.071 blue_control[12788:260721] [FBP-iOS] centralManager didDiscoverPeripheral
flutter: [FBP] [[ OnScanResponse ]] result: {advertisements: [{remote_id: 8D656CA1-144D-28DF-2C6C-A342C46DCE91, manufacturer_data: {76: 1005331cf8af09}, rssi: -71, connectable: 1, tx_power_level: 12}]}
flutter: [FBP] [[ OnScanResponse ]] result: {advertisements: [{remote_id: 8D656CA1-144D-28DF-2C6C-A342C46DCE91, manufacturer_data: {76: 1005331cf8af09}, rssi: -70, connectable: 1, tx_power_level: 12}]}
2024-05-04 17:56:10.073 blue_control[12788:260721] [FBP-iOS] centralManager didDiscoverPeripheral
2024-05-04 17:56:10.074 blue_control[12788:260721] [FBP-iOS] centralManager didDiscoverPeripheral
2
flutter: [FBP] [[ OnScanResponse ]] result: {advertisements: [{remote_id: 3716500E-8590-E43F-0765-546D129AD54B, manufacturer_data: {76: 10073d1f48d0b14768}, rssi: -85, connectable: 1, tx_power_level: 12}]}
2024-05-04 17:56:10.349 blue_control[12788:260721] [FBP-iOS] centralManager didDiscoverPeripheral
2024-05-04 17:56:10.350 blue_control[12788:260721] [FBP-iOS] centralManager didDiscoverPeripheral
2
flutter: [FBP] [[ OnScanResponse ]] result: {advertisements: [{remote_id: 3716500E-8590-E43F-0765-546D129AD54B, manufacturer_data: {76: 10073d1f48d0b14768}, rssi: -89, connectable: 1, tx_power_level: 12}]}
2024-05-04 17:56:10.649 blue_control[12788:260721] [FBP-iOS] centralManager didDiscoverPeripheral
flutter: [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -76, remote_id: B2E17F03-D971-B2D0-2BFE-E4A1556D70FB, manufacturer_data: {76: 12020002}}]}
2024-05-04 17:56:10.953 blue_control[12788:260721] [FBP-iOS] centralManager didDiscoverPeripheral
2024-05-04 17:56:10.955 blue_control[12788:260721] [FBP-iOS] centralManager didDiscoverPeripheral
2
flutter: [FBP] [[ OnScanResponse ]] result: {advertisements: [{remote_id: 3E689256-0832-1EC0-58A1-9D23C8F5E664, manufacturer_data: {76: 0c0e00c9075060fae51c7fb1d511933c1006471d72bf5f78}, rssi: -79, connectable: 1}]}
2
2024-05-04 17:56:10.962 blue_control[12788:260721] [FBP-iOS] centralManager didDiscoverPeripheral
2
flutter: [FBP] [[ OnScanResponse ]] result: {advertisements: [{remote_id: 0B0620C7-F72B-7B51-5E6C-106FA866EF75, manufacturer_data: {76: 10057d1cc6df73}, rssi: -76, connectable: 1, tx_power_level: 12}]}
2024-05-04 17:56:11.254 blue_control[12788:260721] [FBP-iOS] centralManager didDiscoverPeripheral
2024-05-04 17:56:11.255 blue_control[12788:260721] [FBP-iOS] centralManager didDiscoverPeripheral
2
flutter: [FBP] [[ OnScanResponse ]] result: {advertisements: [{remote_id: 3E689256-0832-1EC0-58A1-9D23C8F5E664, manufacturer_data: {76: 0c0e00c9075060fae51c7fb1d511933c1006471d72bf5f78}, rssi: -90, connectable: 1}]}
2
2024-05-04 17:56:11.573 blue_control[12788:260721] [FBP-iOS] centralManager didDiscoverPeripheral
2
flutter: [FBP] [[ OnScanResponse ]] result: {advertisements: [{remote_id: 3E689256-0832-1EC0-58A1-9D23C8F5E664, manufacturer_data: {76: 0c0e00c9075060fae51c7fb1d511933c1006471d72bf5f78}, rssi: -76, connectable: 1}]}
2024-05-04 17:56:11.605 blue_control[12788:260721] [FBP-iOS] centralManager didDiscoverPeripheral
flutter: [FBP] [[ OnScanResponse ]] result: {advertisements: [{remote_id: 3E689256-0832-1EC0-58A1-9D23C8F5E664, manufacturer_data: {76: 0c0e00c9075060fae51c7fb1d511933c1006471d72bf5f78}, rssi: -78, connectable: 1}]}
2024-05-04 17:56:11.871 blue_control[12788:260721] [FBP-iOS] centralManager didDiscoverPeripheral
2024-05-04 17:56:11.873 blue_control[12788:260721] [FBP-iOS] centralManager didDiscoverPeripheral
2
flutter: [FBP] [[ OnScanResponse ]] result: {advertisements: [{remote_id: 3E689256-0832-1EC0-58A1-9D23C8F5E664, manufacturer_data: {76: 0c0e00c9075060fae51c7fb1d511933c1006471d72bf5f78}, rssi: -79, connectable: 1}]}
2024-05-04 17:56:12.152 blue_control[12788:260721] [FBP-iOS] centralManager didDiscoverPeripheral
flutter: [FBP] [[ OnScanResponse ]] result: {advertisements: [{remote_id: 3E689256-0832-1EC0-58A1-9D23C8F5E664, manufacturer_data: {76: 0c0e00c9075060fae51c7fb1d511933c1006471d72bf5f78}, rssi: -93, connectable: 1}]}
2024-05-04 17:56:12.477 blue_control[12788:260721] [FBP-iOS] centralManager didDiscoverPeripheral
2024-05-04 17:56:12.479 blue_control[12788:260721] [FBP-iOS] centralManager didDiscoverPeripheral
2
flutter: [FBP] [[ OnScanResponse ]] result: {advertisements: [{remote_id: 3E689256-0832-1EC0-58A1-9D23C8F5E664, manufacturer_data: {76: 0c0e00c9075060fae51c7fb1d511933c1006471d72bf5f78}, rssi: -77, connectable: 1}]}
flutter: [FBP] <stopScan> args: null
2024-05-04 17:56:13.239 blue_control[12788:260721] [FBP-iOS] handleMethodCall: stopScan
flutter: [FBP] <stopScan> result: true
```

Flucadetena, 2024-05-04:
Also, if you could let me know if the plugin can control the bluetooth devices connected to the system? Meaning, can I connect or disconnect the devices to the system? not the app.

Thanks!!

chipweinberger, 2024-05-04:
you can read the documentation here: https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/retrieveconnectedperipherals(withservices:)?language=objc

closed 2024-05-04 by chipweinberger