chipweinberger/flutter_blue_plus #874: [BugFix]:  FBP logs not getting turned off in Android

JagritAndroidDev, 2024-05-07:
### FlutterBluePlus Version

1.32.5

### Flutter Version

 3.19.5 

### What OS?

Android

### OS Version

Android 11

### Bluetooth Module

HC-08

### What is your feature request?

FBP logs not getting turned off even after adding the command
`  FlutterBluePlus.setLogLevel(LogLevel.none);`

### Logs

```shell
D/[FBP-Android](15589): [FBP] onCharacteristicChanged:
D/[FBP-Android](15589): [FBP]   chr: fa00
D/[FBP-Android](15589): [FBP] onMethodCall: readCharacteristic
D/[FBP-Android](15589): [FBP] onCharacteristicRead:
D/[FBP-Android](15589): [FBP]   chr: fa01
D/[FBP-Android](15589): [FBP]   status: GATT_SUCCESS (0)
D/[FBP-Android](15589): [FBP] onMethodCall: readCharacteristic
D/[FBP-Android](15589): [FBP] onCharacteristicRead:
D/[FBP-Android](15589): [FBP]   chr: fa02
D/[FBP-Android](15589): [FBP]   status: GATT_SUCCESS (0)
[BLE_Service] *******************BLE DATA Foreground*******************
[BLE_Service] *******************BLE DATA Foreground*******************
D/[FBP-Android](15589): [FBP] onCharacteristicChanged:
D/[FBP-Android](15589): [FBP]   chr: fa00
D/[FBP-Android](15589): [FBP] onMethodCall: readCharacteristic
D/[FBP-Android](15589): [FBP] onCharacteristicRead:
D/[FBP-Android](15589): [FBP]   chr: fa01
D/[FBP-Android](15589): [FBP]   status: GATT_SUCCESS (0)
D/[FBP-Android](15589): [FBP] onMethodCall: readCharacteristic
D/[FBP-Android](15589): [FBP] onCharacteristicRead:
D/[FBP-Android](15589): [FBP]   chr: fa02
```

chipweinberger, 2024-05-07:
I don't see `setLogLevel` in your logs

are you sure you are calling it? do   `await FlutterBluePlus.setLogLevel(LogLevel.none);`

chipweinberger, 2024-05-13:
closing. no response.

closed 2024-05-13 by chipweinberger