chipweinberger/flutter_blue_plus #875: FlutterBluePlusException | connect | fbp-code: 1

jyljames, 2024-05-10:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.5

### Flutter Version

3.1.3

### What OS?

Android

### OS Version

Harmony2.0

### Bluetooth Module

AIC8800d

### What is your problem?

I used Honor Play to connect to my Bluetooth device. After successfully connecting for the first time, I disconnected the Bluetooth connection. Then, I had the Bluetooth device emit a Bluetooth broadcast, and when I tried to connect to the Bluetooth device again, I couldn't connect. The specific reason for the error, as shown in the log, was due to a connection timeout. However, I couldn't reproduce this issue when using other Android phones, which is a mandatory issue on Honor Play

### Logs

```shell
13:54:21.416][INFO][bluetooth_manager.dart-302][processBluetooth][_scanDeviceBluetooth] ScanResult, _hasScanDevice]
05-10 13:54:21.751 W/agora.io(26602): [STRATEGY RUNNER] strategy chain not found! timer_type type:1
05-10 13:54:22.220 I/flutterPrint(26602): [(LogExecutor.java:54).flutterPrint] [[1715320462217][2024-05-10 13:54:22.217][INFO][bluetooth_manager.dart-298][processBluetooth][_scanDeviceBluetooth] ScanResult, num = 170]
05-10 13:54:22.224 I/flutterPrint(26602): [(LogExecutor.java:54).flutterPrint] [[1715320462219][2024-05-10 13:54:22.219][INFO][bluetooth_manager.dart-302][processBluetooth][_scanDeviceBluetooth] ScanResult, _hasScanDevice]
05-10 13:54:22.252 W/agora.io(26602): [STRATEGY RUNNER] strategy chain not found! timer_type type:1
05-10 13:54:22.457 I/flutterPrint(26602): [(LogExecutor.java:54).flutterPrint] [[1715320462453][2024-05-10 13:54:22.453][INFO][bluetooth_manager.dart-298][processBluetooth][_scanDeviceBluetooth] ScanResult, num = 171]
05-10 13:54:22.459 I/flutterPrint(26602): [(LogExecutor.java:54).flutterPrint] [[1715320462454][2024-05-10 13:54:22.454][INFO][bluetooth_manager.dart-302][processBluetooth][_scanDeviceBluetooth] ScanResult, _hasScanDevice]
05-10 13:54:22.461 I/flutterPrint(26602): [(LogExecutor.java:54).flutterPrint] [[1715320462459][2024-05-10 13:54:22.459][INFO][bluetooth_manager.dart-298][processBluetooth][_scanDeviceBluetooth] ScanResult, num = 172]
05-10 13:54:22.462 I/flutterPrint(26602): [(LogExecutor.java:54).flutterPrint] [[1715320462461][2024-05-10 13:54:22.461][INFO][bluetooth_manager.dart-302][processBluetooth][_scanDeviceBluetooth] ScanResult, _hasScanDevice]
05-10 13:54:22.752 W/agora.io(26602): [STRATEGY RUNNER] strategy chain not found! timer_type type:1
05-10 13:54:22.888 I/flutterPrint(26602): [(LogExecutor.java:54).flutterPrint] [[1715320462885][2024-05-10 13:54:22.885][INFO][bluetooth_manager.dart-298][processBluetooth][_scanDeviceBluetooth] ScanResult, num = 173]
05-10 13:54:22.889 I/flutterPrint(26602): [(LogExecutor.java:54).flutterPrint] [[1715320462887][2024-05-10 13:54:22.887][INFO][bluetooth_manager.dart-302][processBluetooth][_scanDeviceBluetooth] ScanResult, _hasScanDevice]
05-10 13:54:23.252 W/agora.io(26602): [STRATEGY RUNNER] strategy chain not found! timer_type type:1
05-10 13:54:23.536 I/flutterPrint(26602): [(LogExecutor.java:54).flutterPrint] [[1715320463534][2024-05-10 13:54:23.534][ERROR][device_add_bluetooth_connection.dart-133][deviceAdd][_checkConnectOverDue] _jumpToFailPage add device timeout]
05-10 13:54:23.577 I/flutterPrint(26602): [(LogExecutor.java:54).flutterPrint] [[1715320463576][2024-05-10 13:54:23.576][INFO][bluetooth_manager.dart-298][processBluetooth][_scanDeviceBluetooth] ScanResult, num = 174]
05-10 13:54:23.578 I/flutterPrint(26602): [(LogExecutor.java:54).flutterPrint] [[1715320463577][2024-05-10 13:54:23.577][INFO][bluetooth_manager.dart-302][processBluetooth][_scanDeviceBluetooth] ScanResult, _hasScanDevice]
05-10 13:54:23.752 W/agora.io(26602): [STRATEGY RUNNER] strategy chain not found! timer_type type:1
05-10 13:54:23.805 D/[FBP-Android](26602): [FBP] onMethodCall: disconnect
05-10 13:54:23.805 D/[FBP-Android](26602): [FBP] disconnect: cancelling connection in progress
05-10 13:54:23.805 D/BluetoothGatt(26602): cancelOpen() - device: B4:6D:C2:FC:54:F0
05-10 13:54:23.810 D/BluetoothGatt(26602): close()
05-10 13:54:23.810 D/BluetoothGatt(26602): unregisterApp() - mClientIf=8
05-10 13:54:23.816 I/flutterPrint(26602): [(LogExecutor.java:54).flutterPrint] [[1715320463814][2024-05-10 13:54:23.814][ERROR][bluetooth_manager.dart-404][processBluetooth][_connectDeviceBluetooth] error, name = Cam_I5AP2B, id = B4:6D:C2:FC:54:F0, error = FlutterBluePlusException | connect | fbp-code: 1 | Timed out after 30s]
05-10 13:54:23.816 I/flutterPrint(26602): [(LogExecutor.java:54).flutterPrint] [[1715320463814][2024-05-10 13:54:23.814][INFO][bluetooth_manager.dart-478][processBluetooth][finish] ]
05-10 13:54:23.817 I/flutterPrint(26602): [(LogExecutor.java:54).flutterPrint]
```

chipweinberger, 2024-05-10:
increase the timeout to 90 seconds

jyljames, 2024-05-10:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

jyljames, 2024-05-13:
Thanks，After setting the time to 90 seconds, I still couldn't connect successfully, but another Android 133 error was reported. Then I searched for relevant issues and added a 300ms delay to the Bluetooth of the device on my phone before connecting again, and it was successful！

closed 2024-05-13 by jyljames