chipweinberger/flutter_blue_plus #876: [Bug]: Frequent start/stop scan operations break the scan and return empty results

wcoder, 2024-05-12:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.5

### Flutter Version

3.19.6

### What OS?

Android

### OS Version

Android 11

### Bluetooth Module

Not related yet

### What is your problem?

startScan and stopScan are not synced between Dart and native layers. Android StopScan doesn't have a proper mechanism to track its scanning state.

As a result, scanResult is broken and returns empty list.

Sometimes, new scanning can't be started because it returns the error `SCAN_FAILED_ALREADY_STARTED` and then can't be stopped for restart scanning (fixes by restart app).

StopScan on Android looks like it is not synchronous under the hood and needs some time for all the next startScan, but FBP doesn't handle this.

See a demo of this issue on the example app:

https://github.com/boskokg/flutter_blue_plus/assets/766193/d7b8132c-37fe-445e-a416-c1b97e7f718d

### Logs

```shell
App logs:

Restarted application in 1,544ms.
D/[FBP-Android](23180): [FBP] onMethodCall: flutterRestart
D/[FBP-Android](23180): [FBP] disconnectAllDevices(flutterRestart)
D/[FBP-Android](23180): [FBP] connectedPeripherals: 0
I/flutter (23180): [FBP] <setLogLevel> args: 5
D/[FBP-Android](23180): [FBP] onMethodCall: setLogLevel
I/flutter (23180): [FBP] <setLogLevel> result: true
I/flutter (23180): [FBP] <getAdapterState> args: null
D/[FBP-Android](23180): [FBP] onMethodCall: getAdapterState
I/flutter (23180): [FBP] <getAdapterState> result: {adapter_state: 4}
I/flutter (23180): [FBP] <startScan> args: {with_services: [], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: false, continuous_divisor: 1, android_scan_mode: 2, android_uses_fine_location: false}
D/[FBP-Android](23180): [FBP] onMethodCall: startScan
D/BluetoothAdapter(23180): isLeEnabled(): ON
D/BluetoothLeScanner(23180): onScannerRegistered() - status=0 scannerId=8 mScannerId=0
I/flutter (23180): [FBP] <startScan> result: true
D/[FBP-Android](23180): [FBP] onScanResult
D/[FBP-Android](23180): [FBP] onScanResult
D/[FBP-Android](23180): [FBP] onScanResult
I/flutter (23180): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -47, connectable: 1, *** ]}
D/[FBP-Android](23180): [FBP] onScanResult
I/flutter (23180): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -76, *** ]}
I/flutter (23180): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -75, *** ]}
I/flutter (23180): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -34, connectable: 1, *** ]}
[... 612 lines cut ...]
D/[FBP-Android](23180): [FBP] onScanResult
D/[FBP-Android](23180): [FBP] onScanResult
D/[FBP-Android](23180): [FBP] onScanResult
D/[FBP-Android](23180): [FBP] onScanResult
D/[FBP-Android](23180): [FBP] onScanResult
D/[FBP-Android](23180): [FBP] onScanResult
D/[FBP-Android](23180): [FBP] onScanResult
D/[FBP-Android](23180): [FBP] onScanResult
D/[FBP-Android](23180): [FBP] onScanResult
I/flutter (23180): [FBP] [[ OnScanResponse ]] result: {advertisements: [{rssi: -76, *** ]}
D/[FBP-Android](23180): [FBP] onScanResult
D/[FBP-Android](23180): [FBP] onScanResult
D/[FBP-Android](23180): [FBP] onScanResult
D/[FBP-Android](23180): [FBP] onScanResult
D/[FBP-Android](23180): [FBP] onScanResult
D/[FBP-Android](23180): [FBP] onScanResult
D/[FBP-Android](23180): [FBP] onScanResult
D/[FBP-Android](23180): [FBP] onScanResult
D/[FBP-Android](23180): [FBP] onScanResult
D/[FBP-Android](23180): [FBP] onScanResult
I/flutter (23180): [FBP] <stopScan> args: null
D/[FBP-Android](23180): [FBP] onMethodCall: stopScan
D/BluetoothAdapter(23180): isLeEnabled(): ON
I/flutter (23180): [FBP] <stopScan> result: true
```

chipweinberger, 2024-05-12:
thanks for filing issue. 

can you fix it and open PR? thanks.

closed 2024-05-12 by chipweinberger

chipweinberger, 2024-05-12:
fixed in 1.32.6

alisinaee, 2024-05-21:
I'm using flutter_blue_plus: ^1.32.7  on android and got same bug!!

PR #877 (2024-05-12): Fix isScanning flag when scan has already started and should be stopped before start new scan
Fixes #876 

Test case:
```dart
print('XXX: #1 scan start: isScanningNow:${FlutterBluePlus.isScanningNow}'); // false
await FlutterBluePlus.startScan(timeout: const Duration(seconds: 15)); // SUCCESS
print('XXX: #1 scan done: isScanningNow:${FlutterBluePlus.isScanningNow}'); // true

await Future.delayed(const Duration(seconds: 2));

print('XXX: #2 scan start: isScanningNow:${FlutterBluePlus.isScanningNow}'); // true
await FlutterBluePlus.startScan(timeout: const Duration(seconds: 15)); // SUCCESS
pri [... cut]