chipweinberger/flutter_blue_plus #881: [Help]: last value in MSD is always 0 after 1.32.5

William-Geng, 2024-05-14:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.6

### Flutter Version

3.19.3

### What OS?

Android

### OS Version

13

### Bluetooth Module

nRF52832

### What is your problem?

Hello guys! The last int in scanResult.advertisementData.manufacturerData is always 0, has been working up to 1.32.4, after the update the last value is always 0.
Please see advertisement data in the attached image, here the 0x1C at the end will always be 0

[image]

### Logs

```shell
none
```

chipweinberger, 2024-05-14:
@MrCsabaToth

chipweinberger, 2024-05-14:
so youre saying instead of

Ox17040E785D13B0A1AB4FDCB1045DEE6F6A958D361C

you get

Ox17040E785D13B0A1AB4FDCB1045DEE6F6A958D3600

?

chipweinberger, 2024-05-14:
what is the MSD data you are getting in FBP. please post it exactly.

chipweinberger, 2024-05-14:
the problem is this line

`System.arraycopy(bytes, n + 4, mergedData, existingData.length, fieldLen - 4);
`
should be

`System.arraycopy(bytes, n + 4, mergedData, existingData.length, fieldLen - 3);`

chipweinberger, 2024-05-15:
fixed 1.32.7

closed 2024-05-15 by chipweinberger