chipweinberger/flutter_blue_plus #882: [Help]:  Unstable connection

smithFlutter, 2024-05-15:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.31.14

### Flutter Version

3.7.12

### What OS?

All

### OS Version

IOS 16.7

### Bluetooth Module

unknown

### What is your problem?

Hello everyone, when connecting the device, after the scanner, it takes a long time or it is random for it to work. Or if I use Auto connect it never stays in the connected state, I have tried everything but it is still unstable, has this happened to anyone?

**await device.connectAndUpdateStream();**
It works randomly

**await device.connect(autoConnect: true, mtu: null);**
It doesn't stay connected, if I add a delay sometimes yes.

### Logs

```shell
No log.
```

chipweinberger, 2024-05-15:
FBP cannot controll the connection quality. This is controlled by system bluetooth driver, wireless quality, etc.

closed 2024-05-15 by chipweinberger

smithFlutter, 2024-05-15:
Thanks for your response, the problem is that it runs immediately and remains disconnected

await device.connect(autoConnect: true, mtu: null, timeout: Duration(seconds: 90));

The connection code is in Mian.dart. Sometimes it works well and sometimes it doesn't, how can I send the log?

chipweinberger, 2024-05-15:
read the README about autoconnect....

smithFlutter, 2024-05-16:
I have not been able to solve the problem, I read the readme but it does not help me much, I will evaluate changing the component, I have been doing this for a long time and it should be simple. Thanks for the help.