chipweinberger/flutter_blue_plus #884: [Help]: FlutterBluePlus.systemDevices isn't showing already paired devices to the BLE system

samnbd, 2024-05-15:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.7

### Flutter Version

3.7.11

### What OS?

Android

### OS Version

13

### Bluetooth Module

Nordic

### What is your problem?

When scanning for devices on Android and using both `FlutterBluePlus.onScanResults.listen(...)` and ` FlutterBluePlus.systemDevices`, the already paired device to Android's BLE system will not show up. I even tried using `FlutterBluePlus.connectedDevices` but it's the same thing. 
Please note that I tried to scan the same paired device with flutter_reactive_ble on Android and it showed the device as expected. 

### Logs

```shell
I/flutter (25804): [FBP] <getSystemDevices> args: null
D/[FBP-Android](25804): [FBP] onMethodCall: getSystemDevices
I/flutter (25804): [FBP] <getSystemDevices> result: {devices: []}
433
E/gralloc4(25804): Empty SMPTE 2094-40 data
```

chipweinberger, 2024-05-15:
you can look at FlutterBluePlus.bondedDevices too

and you should look at how they implement their code in android native code and compare

closed 2024-05-15 by chipweinberger

samnbd, 2024-05-16:
Thanks for the reply