chipweinberger/flutter_blue_plus #887: [Help]: Can't get sendData on iOS

kirasoga, 2024-05-17:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.7

### Flutter Version

3.19.1

### What OS?

iOS

### OS Version

17.5.1

### Bluetooth Module

general

### What is your problem?

I created GattServiceProvider with dotnet.   
On Android, the value in ServiceData can be retrieved normally, but on Android, it is not possible to retrieve it.

dotnet is https://qiita.com/gpsnmeajp/items/481fd731851e0e538263

Is this an iOS only specification?

### Logs

```shell
sorry. notting log
```

chipweinberger, 2024-05-17:
?

chipweinberger, 2024-05-17:
need logs

chipweinberger, 2024-05-21:
closing. no response.

closed 2024-05-21 by chipweinberger