chipweinberger/flutter_blue_plus #888: [Help]: I/flutter (10612): [FBP] <setLogLevel> result: true I/flutter (10612): [FBP] <startScan> args: {with_services: [], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: true, continuous_divisor: 1, android_scan_mode: 2, android_uses_fine_location: true} D/[FBP-Android](10612): [FBP] onMethodCall: startScan D/BluetoothAdapter(10612): isLeEnabled(): ON D/BluetoothLeScanner(10612): onScannerRegistered() - status=0 scannerId=7 mScannerId=0 I/flutter (10612): [FBP] <startScan> result: true I/flutter (10612): [FBP] <stopScan> args: null D/[FBP-Android](10612): [FBP] onMethodCall: stopScan D/BluetoothAdapter(10612): isLeEnabled(): ON I/flutter (10612): [FBP] <stopScan> result: true                        i am unable to get nearby device using thi flutter package i dnt know why , can anybody please help me

iqraashraf4457, 2024-05-20:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.7

### Flutter Version

3.22.0 

### What OS?

Android

### OS Version

android 12

### Bluetooth Module

A2DP

### What is your problem?

i am unable to get bluetooth devices as they available 

### Logs

```shell
Debug service listening on ws://127.0.0.1:50856/eCyJWjbgp08=/ws
Syncing files to device TECNO CG6j...
E/ion     (10612): ioctl c0044901 failed with code -1: Invalid argument
D/hw-ProcessState(10612): Binder ioctl to enable oneway spam detection failed: Invalid argument
E/OpenGLRenderer(10612): fbcNotifyFrameComplete error: undefined symbol: fbcNotifyFrameComplete
E/OpenGLRenderer(10612): fbcNotifyNoRender error: undefined symbol: fbcNotifyNoRender
D/[FBP-Android](10612): [FBP] onMethodCall: flutterRestart
D/[FBP-Android](10612): [FBP] initializing BluetoothAdapter
D/[FBP-Android](10612): [FBP] disconnectAllDevices(flutterRestart)
D/[FBP-Android](10612): [FBP] connectedPeripherals: 0
D/[FBP-Android](10612): [FBP] onMethodCall: isSupported
D/[FBP-Android](10612): [FBP] onMethodCall: turnOn
D/[FBP-Android](10612): [FBP] onMethodCall: getAdapterState
D/[FBP-Android](10612): [FBP] onMethodCall: isSupported
D/[FBP-Android](10612): [FBP] onMethodCall: turnOn
I/flutter (10612): BluetoothAdapterState.on
I/flutter (10612): [FBP] <turnOn> result: false
I/flutter (10612): [FBP] <setLogLevel> args: 5
D/[FBP-Android](10612): [FBP] onMethodCall: setLogLevel
I/flutter (10612): [FBP] <setLogLevel> result: true
I/flutter (10612): [FBP] <startScan> args: {with_services: [], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: true, continuous_divisor: 1, android_scan_mode: 2, android_uses_fine_location: true}
D/[FBP-Android](10612): [FBP] onMethodCall: startScan
D/BluetoothAdapter(10612): isLeEnabled(): ON
D/BluetoothLeScanner(10612): onScannerRegistered() - status=0 scannerId=7 mScannerId=0
I/flutter (10612): [FBP] <startScan> result: true
I/flutter (10612): [FBP] <stopScan> args: null
D/[FBP-Android](10612): [FBP] onMethodCall: stopScan
D/BluetoothAdapter(10612): isLeEnabled(): ON
I/flutter (10612): [FBP] <stopScan> result: true
```

chipweinberger, 2024-05-20:
try example app

chipweinberger, 2024-05-20:
and try other scanning apps on App Store

iqraashraf4457, 2024-05-20:
Can you share example app link?

chipweinberger, 2024-05-20:
sorry. you have not read the README, which is required for positing issues. 

this is all addressed in the README. don't waste people's time.

closed 2024-05-20 by chipweinberger

iqraashraf4457, 2024-05-21:
i have read readme , and tried example but not showing device,

linfulongnet, 2024-10-30:
@chipweinberger  Location information must be turned on to scan devices? If it is closed, no device can be scanned.