chipweinberger/flutter_blue_plus #889: [Help]: Verify example for double pairing request on Android

olo95, 2024-05-23:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.7

### Flutter Version

3.19.1

### What OS?

Android

### OS Version

11

### Bluetooth Module

Nordic

### What is your problem?

I set up an approach for resolving double pairing request popup on Android which is apparently common issue. I'm trying to use the suggested fix from documentation by invoking createBond right after connect is finished but it works very random. Sometimes alert is displayed once sometimes twice. I'm wondering if example below properly represents the idea behind suggestion from doc. FYI I listen to bond state for debugging purposes for now. 

Do you maybe have some verified examples for this issue?

`Future<void> connect(BluetoothDevice device, Duration timeout) async {
    final Completer<void> completer = Completer<void>();

    await device.connect(timeout: timeout);
    if (Platform.isAndroid) unawaited(device.createBond());

    device.bondState.listen((bondState) {
      if (bondState == BluetoothBondState.bonded) {
        completer.complete();
      }
    });

    await completer.future;
  }`

### Logs

```shell
D/[FBP-Android](18433): [FBP] onMethodCall: setOptions
D/[FBP-Android](18433): [FBP] initializing BluetoothAdapter
D/EventGDTLogger(18433): Session Event: {"eventType":1,"sessionData":{"sessionId":"asdafaas","firstSessionId":"asdasdasd","sessionIndex":0,"eventTimestampUs":1716473658886000,"dataCollectionStatus":{"performance":1,"crashlytics":2,"sessionSamplingRate":1.0},"firebaseInstallationId":"asdsd-asdasd"},"applicationInfo":{"appId":"asdasdsd","deviceModel":"moto g(50)","sessionSdkVersion":"1.2.1","osVersion":"11","logEnvironment":3,"androidAppInfo":{"packageName":"bundle_identifier","versionName":"1.0.0","appBuildVersion":"1","deviceManufacturer":"motorola","currentProcessDetails":{"processName":"bundle_identifier","pid":18433,"importance":100,"defaultProcess":true},"appProcessDetails":[{"processName":"bundle_identifier","pid":18433,"importance":100,"defaultProcess":true}]}}}
D/SessionFirelogPublisher(18433): Successfully logged Session Start event: 3da5ce5132ba4fcaaae290cba1ff1324
D/[FBP-Android](18433): [FBP] onMethodCall: flutterRestart
D/[FBP-Android](18433): [FBP] disconnectAllDevices(flutterRestart)
D/[FBP-Android](18433): [FBP] connectedPeripherals: 0
D/[FBP-Android](18433): [FBP] onMethodCall: getAdapterState
I/TetheringManager(18433): registerTetheringEventCallback:bundle_identifier
D/[FBP-Android](18433): [FBP] onMethodCall: isSupported
D/[FBP-Android](18433): [FBP] onMethodCall: setOptions
D/[FBP-Android](18433): [FBP] onMethodCall: isSupported
D/[FBP-Android](18433): [FBP] onMethodCall: isSupported
I/flutter (18433): 2024-05-23T16:14:23.243052	D	BleDataSourceImpl._connect:	 Cannot pair with X device automatically, scanning devices
I/flutter (18433): [FBP] stopScan: already stopped
D/[FBP-Android](18433): [FBP] onMethodCall: startScan
D/BluetoothAdapter(18433): isLeEnabled(): ON
D/BluetoothLeScanner(18433): onScannerRegistered() - status=0 scannerId=12 mScannerId=0
D/[FBP-Android](18433): [FBP] onMethodCall: connect
D/BluetoothGatt(18433): connect() - device: 4F:F8:D6:C2:5B:3D, auto: false
D/BluetoothGatt(18433): registerApp()
D/BluetoothGatt(18433): registerApp() - UUID=91f9031e-c9f5-4dd8-8116-fe8c1ca08f04
D/BluetoothGatt(18433): onClientRegistered() - status=0 clientIf=13
D/BluetoothGatt(18433): onClientConnectionState() - status=0 clientIf=13 device=4F:F8:D6:C2:5B:3D
D/[FBP-Android](18433): [FBP] onConnectionStateChange:connected
D/[FBP-Android](18433): [FBP]   status: SUCCESS
D/[FBP-Android](18433): [FBP] OnBondStateChanged: bonding prev: bond-none
D/[FBP-Android](18433): [FBP] onMethodCall: requestMtu
D/[FBP-Android](18433): [FBP] [FBP] waiting for bonding to complete...
D/[FBP-Android](18433): [FBP] [FBP] bonding completed
D/BluetoothGatt(18433): configureMTU() - device: 4F:F8:D6:C2:5B:3D mtu: 512
D/BluetoothGatt(18433): onConnectionUpdated() - Device=4F:F8:D6:C2:5B:3D interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(18433): onPhyUpdate() - status=0 address=4F:F8:D6:C2:5B:3D txPhy=2 rxPhy=2
D/BluetoothGatt(18433): onConfigureMTU() - Device=4F:F8:D6:C2:5B:3D mtu=182 status=0
D/[FBP-Android](18433): [FBP] onMtuChanged:
D/[FBP-Android](18433): [FBP]   mtu: 182
D/[FBP-Android](18433): [FBP]   status: GATT_SUCCESS (0)
D/[FBP-Android](18433): [FBP] onMethodCall: createBond
W/[FBP-Android](18433): [FBP] bonding already in progress
D/BluetoothGatt(18433): onConnectionUpdated() - Device=4F:F8:D6:C2:5B:3D interval=36 latency=0 timeout=500 status=0
D/SessionLifecycleClient(18433): Sending lifecycle 2 to service
D/SessionLifecycleService(18433): Activity backgrounding at 336705370
D/SessionLifecycleClient(18433): Sending lifecycle 1 to service
D/SessionLifecycleService(18433): Activity foregrounding at 336706378.
I/TRuntime.CctTransportBackend(18433): Making request to: https://firebaselogging-pa.googleapis.com/v1/firelog/legacy/batchlog
I/TRuntime.CctTransportBackend(18433): Status Code: 200
D/BluetoothGatt(18433): onConnectionUpdated() - Device=4F:F8:D6:C2:5B:3D interval=36 latency=0 timeout=42 status=0
D/SessionLifecycleClient(18433): Sending lifecycle 2 to service
D/SessionLifecycleService(18433): Activity backgrounding at 336710336
D/SessionLifecycleClient(18433): Sending lifecycle 1 to service
D/SessionLifecycleService(18433): Activity foregrounding at 336711161.
D/[FBP-Android](18433): [FBP] OnBondStateChanged: bonded prev: bonding
I/flutter (18433): 2024-05-23T16:14:35.315082	D	BleRepositoryImpl.connectToX:	 Connected to X device: 4F:F8:D6:C2:5B:3D
```

olo95, 2024-05-23:
I assume that the simplest example could be just:

`await device.connect(timeout: timeout);
await device.createBond()` 

But the behaviour for me is just the same

chipweinberger, 2024-05-31:
there is nothing more FBP can do, that I know of. This is an android OS issue. good. luck.

I think maybe you can try to change device firmware, but I am not sure.

closed 2024-05-31 by chipweinberger