chipweinberger/flutter_blue_plus #891: [Help]: Problem connecting with Android 11

smithFlutter, 2024-05-26:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.6

### Flutter Version

3.7.12

### What OS?

Android

### OS Version

Ventura 13.6.5

### Bluetooth Module

Android 11 - BLE

### What is your problem?

After leaving the component working well, we tested the application on Android 11, even the one that comes as tests in this component, and it does not connect, in any way, has this happened to anyone? A very generic error appears.

[image]

### Logs

```shell
There is no log, only the generic error mentioned.
```

smithFlutter, 2024-05-27:
It only works with Android 11, with other versions of Android it connects well, I have tried it on 3 devices with Android 11 and it is the same problem.

chipweinberger, 2024-05-27:
see readme. no known solution.

closed 2024-05-28 by chipweinberger

chipweinberger, 2024-06-10:
specific devices

their hitting some issue, but impossible to know without good debugging skills