chipweinberger/flutter_blue_plus #892: [Help]: onValueReceived not work

Huy1908, 2024-05-29:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.7

### Flutter Version

3.19.5

### What OS?

Android, iOS

### OS Version

IOS 14.4 and Android 13

### Bluetooth Module

No

### What is your problem?

I use python code to send a string to the flutter application. But in the application I use onValueReceived to listen to the string from python code but do not receive information

### Logs

```shell
await device.connect();
      setState(() {
        connectedDevice = device;
      });
      List<BluetoothService> services = await device.discoverServices();
      BluetoothCharacteristic characteristic = services.first.characteristics.first;

      for (BluetoothService service in services) {
        for (BluetoothCharacteristic c in service.characteristics) {
          if (c.uuid == Guid("00002a24-0000-1000-8000-00805f9b34fb")) {
            characteristic = c;
            c.onValueReceived.listen((event) {
              print("VALUE Received: ${utf8.decode(event)}");
            });
            List<int> valueRead = await c.read();
            await c.setNotifyValue(true);
            break;
          }
        }
      }
```

chipweinberger, 2024-05-29:
probably a problem in your python code.

Huy1908, 2024-05-29:
The python code I sent was successful. Please refer to the code in the attached file

[py.txt](https://github.com/boskokg/flutter_blue_plus/files/15482709/py.txt)

chipweinberger, 2024-05-29:
Does it work using other ble apps?

https://apps.apple.com/us/app/nrf-connect-for-mobile/id1054362403

https://play.google.com/store/apps/details?id=com.macdom.ble.blescanner

chipweinberger, 2024-05-29:
I looked at your python and I see the problem 

your python code is for BLE Peripheral mode

FBP is also for BLE Peripheral mode

To make a ble connection, one device must be in BLE Central mode

closed 2024-05-29 by chipweinberger