chipweinberger/flutter_blue_plus #893: [Help]: data longer than allowed.MTU write length exceeds 512

sumaro, 2024-05-30:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.7

### Flutter Version

3.19.5

### What OS?

Android

### OS Version

android 12

### Bluetooth Module

Unknown

### What is your problem?

On the Android platform, the MTU value obtained by the phone and another Bluetooth BLE device is 517. When I wrote 514 bytes, this error occurred. Why?
 Can't the MTU length of a single write on the Android platform exceed 512?

### Logs

```shell
configureMTU() - device: 54:FA:8D:25:34:81 mtu: 517

writeMtu fail :PlatformException(writeCharacteristic, data longer than allowed. dataLen: 514 > max: 512 (withoutResponse), null, null)
```

chipweinberger, 2024-05-30:
in theory, yes.

in practice its a bad idea for reliability across devices,

we enforce a limit of 512

sumaro, 2024-05-31:
So the maximum value of requestMtu on the Android platform should be 515, right?
Is there a plan to remove the MTU write length limit of 512 on the Android platform?
Can you help me explain the reason for limiting the MTU length to 512 on the Android platform? Or provide some website links for related information.
Thanks.

chipweinberger, 2024-05-31:
the limit is because technically characteristics should have a maximum size of 512

closed 2024-06-03 by chipweinberger