chipweinberger/flutter_blue_plus #894: [Help]: App Crash on Android Device

huongrealest, 2024-05-31:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.7

### Flutter Version

3.10.6

### What OS?

Android

### OS Version

All OS Version

### Bluetooth Module

Any Device

### What is your problem?

When I call startScan function, android device promt permission dialog, i acepted and app crash immediately

### Logs

```shell
E/AndroidRuntime( 5235): java.lang.RuntimeException: Failure delivering result ResultInfo{who=@android:requestPermissions:, request=1452, result=-1, data=Intent { act=android.content.pm.action.REQUEST_PERMISSIONS (has extras) }} to activity {com.boxoffice.org.app/com.boxoffice.app.MainActivity}: java.lang.NullPointerException: Attempt to invoke interface method 'void io.flutter.plugin.common.MethodChannel$Result.error(java.lang.String, java.lang.String, java.lang.Object)' on a null object reference
E/AndroidRuntime( 5235): 	at android.app.ActivityThread.deliverResults(ActivityThread.java:5673)
E/AndroidRuntime( 5235): 	at android.app.ActivityThread.handleSendResult(ActivityThread.java:5712)
E/AndroidRuntime( 5235): 	at android.app.servertransaction.ActivityResultItem.execute(ActivityResultItem.java:67)
E/AndroidRuntime( 5235): 	at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:45)
E/AndroidRuntime( 5235): 	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
E/AndroidRuntime( 5235): 	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
E/AndroidRuntime( 5235): 	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2470)
E/AndroidRuntime( 5235): 	at android.os.Handler.dispatchMessage(Handler.java:106)
E/AndroidRuntime( 5235): 	at android.os.Looper.loopOnce(Looper.java:240)
E/AndroidRuntime( 5235): 	at android.os.Looper.loop(Looper.java:351)
E/AndroidRuntime( 5235): 	at android.app.ActivityThread.main(ActivityThread.java:8423)
E/AndroidRuntime( 5235): 	at java.lang.reflect.Method.invoke(Native Method)
E/AndroidRuntime( 5235): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:584)
E/AndroidRuntime( 5235): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1013)
E/AndroidRuntime( 5235): Caused by: java.lang.NullPointerException: Attempt to invoke interface method 'void io.flutter.plugin.common.MethodChannel$Result.error(java.lang.String, java.lang.String, java.lang.Object)' on a null object reference
E/AndroidRuntime( 5235): 	at com.example.bluetooth_print.BluetoothPrintPlugin.startScan(BluetoothPrintPlugin.java:275)
E/AndroidRuntime( 5235): 	at com.example.bluetooth_print.BluetoothPrintPlugin.onRequestPermissionsResult(BluetoothPrintPlugin.java:448)
E/AndroidRuntime( 5235): 	at io.flutter.embedding.engine.FlutterEngineConnectionRegistry$FlutterEngineActivityPluginBinding.onRequestPermissionsResult(FlutterEngineConnectionRegistry.java:781)
E/AndroidRuntime( 5235): 	at io.flutter.embedding.engine.FlutterEngineConnectionRegistry.onRequestPermissionsResult(FlutterEngineConnectionRegistry.java:413)
E/AndroidRuntime( 5235): 	at io.flutter.embedding.android.FlutterActivityAndFragmentDelegate.onRequestPermissionsResult(FlutterActivityAndFragmentDelegate.java:820)
E/AndroidRuntime( 5235): 	at io.flutter.embedding.android.FlutterActivity.onRequestPermissionsResult(FlutterActivity.java:941)
E/AndroidRuntime( 5235): 	at android.app.Activity.dispatchRequestPermissionsResult(Activity.java:8902)
E/AndroidRuntime( 5235): 	at android.app.Activity.dispatchActivityResult(Activity.java:8760)
E/AndroidRuntime( 5235): 	at android.app.ActivityThread.deliverResults(ActivityThread.java:5666)
E/AndroidRuntime( 5235): 	... 13 more
I/Process ( 5235): Sending signal. PID: 5235 SIG: 9
Lost connection to device.
```

chipweinberger, 2024-05-31:
this is not flutter blue plus.

you are using `BluetoothPrintPlugin`

closed 2024-05-31 by chipweinberger

huongrealest, 2024-06-03:
@chipweinberger The problem occurs when both libraries use the same request code. please change request code on upcoming version