chipweinberger/flutter_blue_plus #895: [Help]:

dmkk3r, 2024-06-02:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.7

### Flutter Version

3.22.1

### What OS?

Android, iOS

### OS Version

Android 14

### Bluetooth Module

esp32c3

### What is your problem?

Is it possible to get direct access to the Advertisement Packet?
I am currently trying to build an integration for BTHome into my app: https://bthome.io/format/

BTHome uses the advertisement packet to send the sensor data with its own protocol.

A field for the RawPDU would be useful in the AdvertisementData of the ScanResult. Is it perhaps possible via detours?

### Logs

```shell
No logs
```

chipweinberger, 2024-06-02:
only possible on android, not ios

chipweinberger, 2024-06-02:
https://github.com/boskokg/flutter_blue_plus/blob/75e871e400483da3128219baa506f846323db2c2/lib/src/flutter_blue_plus.dart#L777

closed 2024-06-02 by chipweinberger

chipweinberger, 2024-06-03:
those are both the same exact data, just showing in different formats.

dmkk3r, 2024-06-03:
Okay, now I understand what you mean, but that's not what I was aiming for. The decisive factor for me would be the value:
"0x020106 0616d2fc402f420b096772656 56e67756172640000000000" from the advertisement packet.
But I suspect I won't get there this way.