chipweinberger/flutter_blue_plus #898: bondedDevices isConnected

save-us-prophet, 2024-06-04:
### FlutterBluePlus Version

1.32.7

### Flutter Version

3.19.6

### What OS?

Android

### OS Version

android 12

### Bluetooth Module

Exynos

### What is your feature request?

```dart
FlutterBluePlus.bondedDevices.then((bondedDevices) {
      for (final device in bondedDevices) {
        print(device.isConnected);
      }
    });
```

All connected devices also return **FALSE**.

### Logs

```shell
FlutterBluePlus.bondedDevices.then((bondedDevices) {
      for (final device in bondedDevices) {
        print(device.isConnected);
      }
    });
```

chipweinberger, 2024-06-04:
you can be bonded but not connected 

bonded just means "previously paired"

save-us-prophet, 2024-06-05:
```kotlin
fun isConnected(device: BluetoothDevice): Boolean {
        return try {
            val method = device.javaClass.getMethod("isConnected")
            method.invoke(device) as Boolean
        } catch (e: Exception) {
            throw IllegalStateException(e)
        }
    }
```
It would be nice to have a function that could check the connection status.

What do you think?

chipweinberger, 2024-06-05:
i dont understand

closed 2024-06-06 by chipweinberger

efraespada, 2024-09-09:
@save-us-prophet is talking about the connection with the system, not the app.

This method access to the private field `isConnected` which allows to know if a BT device is connected to the smartphone: 

```kotlin
fun isConnected(device: BluetoothDevice): Boolean {
        return try {
            val method = device.javaClass.getMethod("isConnected")
            method.invoke(device) as Boolean
        } catch (e: Exception) {
            throw IllegalStateException(e)
        }
    }
```

I think it is not implemented in this library. Sadly 😞