chipweinberger/flutter_blue_plus #899: [Help]: In startScan(withRemoteIds), I have a question about the behavior of with RemoteIds.

Louis-Wave, 2024-06-05:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.2

### Flutter Version

3.22.1

### What OS?

Android

### OS Version

Android 14

### Bluetooth Module

Nordic NRF52832

### What is your problem?

I save 4 remoteIds in a variable called SavedAdd, and when scanning, I try to connect when the corresponding device is identified.
However, the device was discovered only when the device corresponding to SavedAdds[0] was found, and the rest, including SavedAdds[1], could not be found.

This is my code.

```
List<String> SavedAdds = ['C3:A5:61:DD:F1:9F', 'EB:60:4C:9E:3F:10', 'None' , 'D4:CF:1A:6F:26:7D']

...

await FlutterBluePlus.startScan(
          withRemoteIds: [SavedAdds]
      );

```

Here I tried to connect a device with remoteID EB:60:4C:9E:3F:10, but the App couldn't find the device.

### Logs

```shell
Scanning, Ids
```

chipweinberger, 2024-06-05:
I'm not sure the cause. Try to debug it further.

btw, you can connect without scanning. 

1. create BluetoothDevice with remoteId
2. await device.connect()

chipweinberger, 2024-06-06:
the code looks correct to me. probably just a coincidence. and your devices are having problems.

```
                        // remoteIds
                        for (int i = 0; i < withRemoteIds.size(); i++) {
                            String address = withRemoteIds.get(i);
                            ScanFilter f = new ScanFilter.Builder().setDeviceAddress(address).build();
                            filters.add(f);
                        }
```

closed 2024-06-06 by chipweinberger