chipweinberger/flutter_blue_plus #900: [Help]: Write without Response Sends Empty Value

Antonwy, 2024-06-07:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

^1.32.7

### Flutter Version

3.19.6

### What OS?

iOS

### OS Version

17.5.1

### Bluetooth Module

Greensolar Battery Module

### What is your problem?

I am trying to send data to my bluetooth module, but somehow I don't get a response or at least the listeners don't get triggered.

Especially interesting is, that

```
flutter: Written global event: []
flutter: Sender value: []
```

here is empty, but the data I put into `.write` is definitely not empty.

Also I tried to send exactly the same data using the Lightblue app and as seen in the logs, I get a valid response.

Thanks for your help in advance :)

```dart
try {
      await device.connect();

      final services = await device.discoverServices();

      final service = services.firstWhereOrNull(
        (service) => service.uuid.toString() == _bluetoothDeviceServiceUuidHex,
      );

      if (service == null) {
        throw Exception("Service not found");
      }

      final senderCharacteristic = service.characteristics.firstWhereOrNull(
        (characteristic) =>
            characteristic.uuid.toString() ==
            _bluetoothDeviceSenderCharacteristicUuidHex,
      );

      final receiverCharacteristic = service.characteristics.firstWhereOrNull(
        (characteristic) =>
            characteristic.uuid.toString() ==
            _bluetoothDeviceReceiverCharacteristicUuidHex,
      );

      if (senderCharacteristic == null || receiverCharacteristic == null) {
        throw Exception("Sender or receiver characteristic not found");
      }

      final receiverStream =
          receiverCharacteristic.lastValueStream.listen((event) {
        print("Receiver value: $event");
      });

      final senderStream = senderCharacteristic.lastValueStream.listen((event) {
        print("Sender value: $event");
      });

      await receiverCharacteristic.setNotifyValue(true);
      await senderCharacteristic.setNotifyValue(true);

      device.cancelWhenDisconnected(receiverStream);
      device.cancelWhenDisconnected(senderStream);

      final data = _buildMessage([0x73, 0x06, 0x23, 0x04, 0x01]);

      print("Data to send: ${data.map((e) => e.toRadixString(16))}");

      String hexEncode(List<int> numbers) {
        return numbers
            .map((n) => (n & 0xFF).toRadixString(16).padLeft(2, '0'))
            .join();
      }

      print("Data to send hex: ${hexEncode(data)}");

      await senderCharacteristic.write(
        data,
        withoutResponse: true,
      );
      print("Data sent");

      await senderCharacteristic.setNotifyValue(true);
      await receiverCharacteristic.setNotifyValue(true);

      print("Is subscribed: ${receiverCharacteristic.isNotifying}");
    } catch (e) {
      print("Error: $e");
    } finally {
      await Future.delayed(5.seconds);

      await device.disconnect();

      print("Done");
    }
```

### Logs

```shell
## My Logs
flutter: Receiver value: []
flutter: Sender value: []
flutter: Data to send: (73, 6, 23, 4, 1, 53)
flutter: Data to send hex: 730623040153
flutter: Written global event: []
flutter: Sender value: []
flutter: Data sent
flutter: Is subscribed: true
flutter: Done

## Lightblue Logs
11:56:54.179 - Connected to nearby peripheral: HM_B2500_c89c 
11:56:55.066 - Stopping search for nearby peripherals 
11:57:14.245 - Characteristic (FF01) wrote new value: <73062304 0153> 
11:57:14.335 - Characteristic (FF02) notified: <733c2304 74797065 3d484d4b 2d322c69 643d3336 30313131 35303462 35373331 33343335 31343464 34632c6d 61633d39 34373036 63636463 38396373>
```

chipweinberger, 2024-06-07:
thanks for the detailed issue 

 ab you investigate further and open a PR?

chipweinberger, 2024-06-10:
fixed: 81cfc12bde223f82da82dbf5354f83b37519b591

please try master branch by updating your `pubspecl.yaml`

Antonwy, 2024-06-10:
hi @chipweinberger :)

huge thank you already for responding this quickly!

this already fixed the problem with the sender value. but still I don't receive anything back, although with the same setup I receive an value back when using Lightblue.

[image]
Here you can see that we have three characteristics.

[image]
Here I am listening to our `ff02` characteristics.

[image]
And here I am writing `730623040153` and also directly getting back a result.

Antonwy, 2024-06-10:
https://github.com/boskokg/flutter_blue_plus/blob/81cfc12bde223f82da82dbf5354f83b37519b591/ios/Classes/FlutterBluePlusPlugin.m#L1394-L1434

this callback should also get triggered when we send something with `withoutResponse: true`. At least in the Swift implementation, this callback gets triggered.

Antonwy, 2024-06-10:
hmm so I found a solution. somehow when using `withoutResponse: true` it can take multiple tries until the `updateCharacteristics` is called. So now I am just trying it multiple times.

chipweinberger, 2024-06-10:
please send logs

chipweinberger, 2024-06-14:
fixed 1.32.8

closed 2024-06-14 by chipweinberger