chipweinberger/flutter_blue_plus #902: [Help]: [ANDROID] Code stalls after setting notification value after the first few times within loop

SyedKaz00, 2024-06-11:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.5

### Flutter Version

3.19.1

### What OS?

Android

### OS Version

Android 10 (API 29)

### Bluetooth Module

/

### What is your problem?

I am running Flutter Blue plus on a loop, where it scans and connects to devices based off the manufacturer ID. This works for the first few times it runs, however after successive succesful runs, it begins to stall after setting the notification value and it no longer reads or writes to the characteristics. The loop is running within the foreground service using the flutter_foreground_service package, however I dont think this is the issue as it works the first few times it is called and then it suddenly stops working. 
Here is the code that I am using that calls flutter blue plus
```
class BLEManager {
  // Singleton instance
  static final BLEManager _singleton = BLEManager._internal();

  // Factory method to return the same instance
  factory BLEManager() {
    return _singleton;
  }

  // Named constructor
  BLEManager._internal() {
    print('BLEManager instance created');
  }

  //Has implementation of both Central and Peripheral modes.
  Future<List<List<int>>> performBleOperationCentral(
      List<List<int>> outgoingPackets) async {
    Guid serviceUuid = Guid('00000abc-0000-1000-8000-00805f9b34fb');
    Guid characteristicUuidPC = Guid('0000000a-0000-1000-8000-00805f9b34fb');
    Guid characteristicUuidCP = Guid('0000000b-0000-1000-8000-00805f9b34fb');

    List<ScanResult> scanResults = [];
    List<List<int>> incomingPacketsPC = [];
    late StreamSubscription<List<ScanResult>> scanResultsSubscription;

    Completer<List<List<int>>> completer = Completer();

    Future<void> stopScan() async {
      await FlutterBluePlus.stopScan();
    }

    Future<void> writeToCharacteristicCP(
        BluetoothDevice device, List<int> data) async {
      try {
        for (BluetoothService service in await device.discoverServices()) {
          if (service.uuid == serviceUuid) {
            for (BluetoothCharacteristic characteristic
[... 400 lines cut ...]
I/flutter (18168): [FBP] stopScan: already stopped
I/flutter (18168): Scan Results: [ScanResult{device: BluetoothDevice{remoteId: 68:24:72:DC:AE:12, platformName: Baba, services: Instance of 'BmDiscoverServicesResult'}, advertisementData: AdvertisementData{advName: Baba, txPowerLevel: null, appearance: null, connectable: true, manufacturerData: {1: [1, 2, 3]}, serviceData: {}, serviceUuids: []}, rssi: -53, timeStamp: 2024-06-11 18:40:18.690235}]
D/[FBP-Android](18168): [FBP] onMethodCall: connect
D/[FBP-Android](18168): [FBP] already connected
D/[FBP-Android](18168): [FBP] onMethodCall: requestMtu
D/[FBP-Android](18168): [FBP] onMtuChanged:
D/[FBP-Android](18168): [FBP]   mtu: 512
D/[FBP-Android](18168): [FBP]   status: GATT_SUCCESS (0)
D/[FBP-Android](18168): [FBP] onMethodCall: requestMtu
D/[FBP-Android](18168): [FBP] onMtuChanged:
D/[FBP-Android](18168): [FBP]   mtu: 512
D/[FBP-Android](18168): [FBP]   status: GATT_SUCCESS (0)
D/[FBP-Android](18168): [FBP] onMethodCall: discoverServices
D/[FBP-Android](18168): [FBP] onServicesDiscovered:
D/[FBP-Android](18168): [FBP]   count: 2
D/[FBP-Android](18168): [FBP]   status: 0GATT_SUCCESS
D/[FBP-Android](18168): [FBP] onMethodCall: setNotifyValue
W/[FBP-Android](18168): [FBP] CCCD descriptor for characteristic not found: 2a05
I/flutter (18168): Error during timeout: TimeoutException: Timeout occurred while waiting for packets.
I/flutter (18168): Central queue after BLE operation: [[6, 9, 6, 9]]
I/flutter (18168): Central queue is empty, no packets to process
I/flutter (18168): Current values are peripheralReceived: true and centralReceived: false      
I/flutter (18168): The packets have been sent out by atleast on of the peripheral or central queues
I/flutter (18168): Wait 10 seconds before next iteration
```

chipweinberger, 2024-06-11:
FBP only does 1 operations at a time.

Maybe you are queueing too much work.

SyedKaz00, 2024-06-14:
What would I do to solve this, it seems to me that in my code only ever one operation is called, but would you recommend keeping all the flutter_blue_plus commands in a separate singleton class and sending requests there sequentially or something similar to that. 
Please help I've been working on this project for quite a while and would hate to have to go back to android native from flutter. 
Thanks in advance. :)

chipweinberger, 2024-06-14:
I would try and figure out the cause first.

Just keep playing with your code until you can figure out the cause.

SyedKaz00, 2024-06-14:
Yea I will give the code a refactor and see if I run into the same issue.
Thanks for the prompt responses.
Just while I'm here, do you know what this line is, googling it doesn't get me anywhere.
```
W/[FBP-Android](18168): [FBP] CCCD descriptor for characteristic not found: 2a05
```
The peripheral device is also a android phone if that helps

chipweinberger, 2024-06-14:
it's just a warning. You can google what a CCCD is.

chipweinberger, 2024-06-19:
closing. not enough info.

closed 2024-06-19 by chipweinberger

chipweinberger, 2026-03-28:
please open a new issue