chipweinberger/flutter_blue_plus #909: [Help]: Unable to get real MTU

Jackjack47, 2024-06-20:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.31.4

### Flutter Version

3.19.6

### What OS?

iOS

### OS Version

IOS 16.7.8

### Bluetooth Module

none

### What is your problem?

When I use an IOS phone as the central device to connect an Android phone as a peripheral device through `BluetoothDevice.connect`, the mtu I got on my ios phone via the `BluetoothDevice.mtu`  is 515, and mtu is 517 on Android phone.

I think the problem stems from the following two method in FlutterBluePlusPlugin.m:
```
- (int)getMaxPayload:(CBPeripheral *)peripheral forType:(CBCharacteristicWriteType)writeType allowLongWrite:(bool)allowLongWrite
{
    // if allowLongWrite is disabled, we can only write up to MTU-3
    if (allowLongWrite == false) {
        writeType = CBCharacteristicWriteWithoutResponse;
    }

    // For withoutResponse, or allowLongWrite == false
    //   iOS returns MTU-3. In theory, MTU can be as high as 65535 (16-bit).
    //   I've seen iOS return 524 for this value. But typically it is lower.
    //   The MTU negotiated by the OS depends on iOS version.
    //
    // For withResponse, 
    //   iOS typically returns a constant value of 512, regardless of MTU. 
    //   This is because iOS will autosplit large writes
    int maxForType = (int) [peripheral maximumWriteValueLengthForType:writeType];

    // In order to operate the same on both iOS & Android, we enforce a 
    // maximum of 512, which is the same as android. This is also the
    // maxAttrLen of a characteristic in the BLE specification.
    return MIN(maxForType, 512);
}

- (int)getMtu:(CBPeripheral *)peripheral
{
    int maxPayload = [self getMaxPayload:peripheral forType:CBCharacteristicWriteWithoutResponse allowLongWrite:false];
    return maxPayload + 3; // ATT overhead
}
```

When the real MTU is 517, `getMaxPayload` return 512, then `getMtu` return 515.
Maybe we can make the `getMtu` method return the real mtu?

### Logs

```shell
none
```

chipweinberger, 2024-06-21:
why do you need real mtu?

Jackjack47, 2024-06-21:
I didn't read the notes carefully enough, I thought when mtu was greater than 512, IOS won't autosplit large writes...

closed 2024-06-21 by Jackjack47