chipweinberger/flutter_blue_plus #912: [Help]: can't scan in Galaxy Z Flip

rogerjin1206, 2024-06-20:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.8

### Flutter Version

3.22.2

### What OS?

Android

### OS Version

Android 14.0

### Bluetooth Module

BoT-nLE522

### What is your problem?

i am currently testing the Galaxy S10 and Z Flip3. For Galaxy S10, peripheral devices can be scanned normally, but for Flip3, peripheral devices will not come out even after scanning. Is it possible that the scanning function depending on the device does not work properly?

### Logs

```shell
void subscribeToScanResults() {
    scanSubscription = FlutterBluePlus.onScanResults.listen(
      (results) {
        logger.i(results);
        devices.value = results.map((result) => result.device).toList();
      },
      onError: (e) => logger.e(e.toString()),
    );
    FlutterBluePlus.cancelWhenScanComplete(scanSubscription!);
  }
```

chipweinberger, 2024-06-20:
missing logs

chipweinberger, 2024-06-20:
if your device is already connected to the Galaxy S10, it might stop advertising.

closed 2024-06-20 by chipweinberger

rogerjin1206, 2024-06-20:
> missing logs

sorry, i gonna give you logs