chipweinberger/flutter_blue_plus #913: [Feature]: Android BLE Scan Stops Automatically After 10 Minutes

codercengiz, 2024-06-20:
### FlutterBluePlus Version

1.32.8

### Flutter Version

3.22

### What OS?

Android

### OS Version

Android 14

### Bluetooth Module

esp32

### What is your feature request?

After starting the BLE scan with FlutterBluePlus.startScan, it automatically stops after 10 minutes by android. 
If this behavior can be detected, it would be great if `FlutterBluePlus.isScanning` is also updated accordingly.

### Logs

```shell
The android log after 10 minutes shows:

2024-06-20 11:00:49.613  1766-2203  BtGatt.ScanManager      com.android.bluetooth                W  Moving unfiltered scan client to opportunistic scan (scannerId 3)
2024-06-20 11:00:49.613  1766-2203  bt_stack                com.android.bluetooth                I  [INFO:le_scanning_manager.cc(214)] ScanFilterParamSetup in shim layer
2024-06-20 11:00:49.615   664-17300 android.ha...h@1.0-impl and...ardware.bluetooth@1.0-service  I  Send: VendorInterface::Send
2024-06-20 11:00:49.617  1766-2203  bt_stack                com.android.bluetooth                I  [INFO:le_scanning_manager.cc(178)] Scan in shim layer stopped
2024-06-20 11:00:49.617  1766-1975  bluetooth               com.android.bluetooth                I  packages/modules/Bluetooth/system/gd/hci/le_address_manager.cc:203 unregister_client: Client unregistered
2024-06-20 11:00:49.617  1766-2203  bt_stack                com.android.bluetooth                I  [INFO:le_scanning_manager.cc(394)] SetScanParameters with scanning phys in shim layer: phy_cnt=1
2024-06-20 11:00:49.617  1766-2203  bt_stack                com.android.bluetooth                I  [INFO:le_scanning_manager.cc(413)] SetScanParameters: scan_type=1, scan_interval=2977, scan_window=324
```
```

chipweinberger, 2024-06-20:
please submit PR

chipweinberger, 2024-06-20:
I dont think it stops. It just goes into opportunistic mode.

closed 2024-06-20 by chipweinberger

VUNO-wonwoohan, 2024-12-19:
@codercengiz
I had the same issue. I was intentionally scanning without a timeout while the app was running, but the OS automatically stopped it. This occurred on both iOS and Android.
I resolved it by adding a timeout when calling FlutterBluePlus.startScan and restarting the scan before the OS stops it.

aqoong, 2024-12-19:
@codercengiz 
I had the same issue. I was intentionally scanning without a timeout while the app was running, but the OS automatically stopped it. This occurred on both iOS and Android.
I resolved it by adding a timeout when calling FlutterBluePlus.startScan and restarting the scan before the OS stops it.