chipweinberger/flutter_blue_plus #914: [Feature]: if the connection did not occur, disconnect command stuck.

keskink, 2024-06-20:
### FlutterBluePlus Version

1.32.8

### Flutter Version

3.22.2

### What OS?

iOS

### OS Version

17.5.1

### Bluetooth Module

TI2640

### What is your feature request?

if the connection did not occur, disconnect command stuck

await rdt.device.disconnect(timeout: 5, queue: false); stuck after.
await this.device.connect(autoConnect: false). 
device not power-on.

the code not passing bluetooth_device.dart (line 184). await dtx.take();

### Logs

```shell
no logs
```

chipweinberger, 2024-06-21:
can you repro is on the example app?

logs would be really useful

also, can you add a lot of print statements to the connect & disconnect code in FBP so we can see where it gets stuck and who still has the lock?

keskink, 2024-06-21:
The main problem is that I wanted to connect without the device being turned on (and w/o scan). As you know, when I want to connect without the device being turned on, there is no problem. When the device is turned on, it connects automatically. But, when I disconnect without connecting, the command hangs.

I don't use example, because of, I am using my app in connect w/o scan mode.

it is working good in this condition. I think no issue after this.

**await this.device.connect(autoConnect: true, mtu: null);
.
.
.
onDoubleTap: () async {
                  await fxRdt.device.disconnect();
                  print("${fxRdt.name} ddd disconnected");
                  await fxRdt.connect();
                  print("${fxRdt.name} ccc connected");
                },**

chipweinberger, 2024-06-21:
what you describe should work.

you should be able to cancel an in-progress connection attempt using `disconnect`

since it seems you decided to use autoconnect instead, im going to close this.

closed 2024-06-21 by chipweinberger