chipweinberger/flutter_blue_plus #915: [Feature]: withoutResponse problem

keskink, 2024-06-20:
### FlutterBluePlus Version

1.32.8

### Flutter Version

3.22.2

### What OS?

iOS

### OS Version

17.5.1

### Bluetooth Module

ti 2460

### What is your feature request?

withoutResponse assuming connection dropped 

### Logs

```shell
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: FlutterBluePlusException | writeCharacteristic | fbp-code: 6 | Device is disconnected
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(writeCharacteristic, device is disconnected, A37A0D0F-FF3F-161F-47CC-42AA39238331, null)
#0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:648:7)
#1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:334:18)
<asynchronous suspension>
#2      FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:603:13)
<asynchronous suspension>
#3      BluetoothCharacteristic.write (package:flutter_blue_plus/src/bluetooth_characteristic.dart:204:7)
<asynchronous suspension>
#4      WfuStatePage.programmingTick (package:fxv2/Device/TimerDetail/firmwarePage.dart:342:13)
<asynchronous suspension>
```

chipweinberger, 2024-06-21:
If you get this error, your device is disconected

keskink, 2024-06-21:
No way.
1- my app is not new developing. Codes are running with 1.20.8 w/o any problem. I upgraded to 1.32.8 in few days.
2- if I set withoutResponse=false (or use default setting), it is working fine.

Thanks in advance

chipweinberger, 2024-06-21:
okay.

keskink, 2024-06-21:
This issue should not close. the issue still exists when I set withoutResponse = true.
I need to use this withoutResponse.

should fix this issue.

chipweinberger, 2024-06-21:
What is "the issue"?

keskink, 2024-06-21:
same issue, nothing changed.
withoutResponse assuming connection closed. But device not disconnected. 
When I set withoutResponse=false it is working. but I need to use withoutResponse=true.

chipweinberger, 2024-06-21:
how do you know the device is not disconnected?

chipweinberger, 2024-06-21:
also, can you find the first version where this started becoming a problem?

keskink, 2024-06-21:
I know is not disconnect because, device working and don't send disconnect commend. and same code working with 1.20.8
and I don't know where the issue started.

chipweinberger, 2024-06-22:
without more info this cannot be fixed and the issue must be closed

chipweinberger, 2024-06-25:
not enough info.

closed 2024-06-25 by chipweinberger