chipweinberger/flutter_blue_plus #916: IOS connect well when pairing from the same source, but not on Android.

freecions, 2024-06-26:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.8

### Flutter Version

3.22.0

### What OS?

Android

### OS Version

Android API 14

### Bluetooth Module

Bluetooth BLE 4.2

### What is your problem?

When pairing external Bluetooth from the same source, the IOS is very connected
On Android, it connects first and then automatically disconnects and does not reconnect.
You can't fix it no matter how hard you try.
Could you please help me? Please take a look at the log file.

### Logs

```shell
D/BluetoothGatt(28781): onClientConnectionState() - status=0 clientIf=12 device=XX:XX:XX:XX:BD:C5
D/[FBP-Android](28781): [FBP] onConnectionStateChange:connected
D/[FBP-Android](28781): [FBP]   status: SUCCESS
I/flutter (28781): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 0, disconnect_reason_string: SUCCESS, remote_id: 04:91:62:A4:BD:C5, connection_state: 1}
I/flutter (28781): [FBP] <readRssi> args: 04:91:62:A4:BD:C5
D/[FBP-Android](28781): [FBP] onMethodCall: readRssi
D/BluetoothGatt(28781): readRssi() - device: XX:XX:XX:XX:BD:C5
D/[FBP-Android](28781): [FBP] onReadRemoteRssi:
D/[FBP-Android](28781): [FBP]   rssi: -67
D/[FBP-Android](28781): [FBP]   status: GATT_SUCCESS (0)
D/[FBP-Android](28781): [FBP] onScanResult
I/flutter (28781): [FBP] <readRssi> result: true
I/flutter (28781): [FBP] [[ OnReadRssi ]] result: {error_string: GATT_SUCCESS, rssi: -67, success: 1, remote_id: 04:91:62:A4:BD:C5, error_code: 0}
D/[FBP-Android](28781): [FBP] onScanResult
D/[FBP-Android](28781): [FBP] OnBondStateChanged: bonding prev: bond-none
I/flutter (28781): [FBP] [[ OnBondStateChanged ]] result: {bond_state: 1, remote_id: 04:91:62:A4:BD:C5, prev_state: 0}
D/[FBP-Android](28781): [FBP] onScanResult
I/ViewRootImpl@64e2293[MainActivity](28781): handleWindowFocusChanged: 0 0 call from android.view.ViewRootImpl.-$$Nest$mhandleWindowFocusChanged:0
I/ImeFocusController(28781): onPreWindowFocus: skipped, hasWindowFocus=false mHasImeFocus=true
I/ImeFocusController(28781): onPostWindowFocus: skipped, hasWindowFocus=false mHasImeFocus=true
D/[FBP-Android](28781): [FBP] onScanResult
D/[FBP-Android](28781): [FBP] onScanResult
D/BluetoothGatt(28781): onConnectionUpdated() - Device=XX:XX:XX:XX:BD:C5 interval=6 latency=0 timeout=500 status=0
D/[FBP-Android](28781): [FBP] onScanResult
D/[FBP-Android](28781): [FBP] onScanResult
D/[FBP-Android](28781): [FBP] onScanResult
D/[FBP-Android](28781): [FBP] onScanResult
D/[FBP-Android](28781): [FBP] onScanResult
D/[FBP-Android](28781): [FBP] onScanResult
D/BluetoothGatt(28781): onConnectionUpdated() - Device=XX:XX:XX:XX:BD:C5 interval=24 latency=0 timeout=500 status=0
D/[FBP-Android](28781): [FBP] onScanResult
D/[FBP-Android](28781): [FBP] onScanResult
[... 125 lines cut ...]
D/[FBP-Android](28781): [FBP] onScanResult
I/InsetsController(28781): onStateChanged: host=com.lib.flutter_blue_plus_example/com.lib.flutter_blue_plus_example.MainActivity, from=android.view.ViewRootImpl$ViewRootHandler.handleMessageImpl:7209, state=InsetsState: {mDisplayFrame=Rect(0, 0 - 1080, 2640), mDisplayCutout=DisplayCutout{insets=Rect(0, 94 - 0, 0) waterfall=Insets{left=0, top=0, right=0, bottom=0} boundingRect={Bounds=[Rect(0, 0 - 0, 0), Rect(505, 0 - 576, 94), Rect(0, 0 - 0, 0), Rect(0, 0 - 0, 0)]} cutoutPathParserInfo={CutoutPathParserInfo{displayWidth=1080 displayHeight=2640 physicalDisplayWidth=1080 physicalDisplayHeight=2640 density={3.0} cutoutSpec={M 0,0 H -11.83333333333333 V 31.33333333333333 H 11.83333333333333 V 0 H 0 Z @dp} rotation={0} scale={1.0} physicalPixelDisplaySizeRatio={1.0}}}}, mRoundedCorners=RoundedCorners{[RoundedCorner{position=TopLeft, radius=95, center=Point(95, 95)}, RoundedCorner{position=TopRight, radius=95, center=Point(985, 95)}, RoundedCorner{position=BottomRight, radius=95, center=Point(985, 2545)}, RoundedCorner{position=BottomLeft, radius=95, center=Point(95, 2545)}]}  mRoundedCornerFrame=Rect(0, 0 - 1080, 2640), mPrivacyIndicatorBounds=PrivacyIndicatorBounds {static bounds=Rect(964, 0 - 1080, 94) rotation=0}, mDisplayShape=DisplayShape{ spec=-311912193 displayWidth=1080 displayHeight=2640 physicalPixelDisplaySizeRatio=1.0 rotation=0 offsetX=0 offsetY=0 scale=1.0}, mSources= { InsetsSource: {fbb80001 mType=navigationBars mFrame=[0,2514][1080,2640] mVisible=true mFlags=[]}, InsetsSource: {fbb80004 mType=systemGestures mFrame=[0,0][0,0] mVisible=true mFlags=[]}, InsetsSource: {fbb80005 mType=mandatorySystemGestures mFrame=[0,2514][1080,2640] mVisible=true mFlags=[]}, InsetsSource: {fbb80006 mType=tappableElement mFrame=[0,2514][1080,2640] mVisible=true mFlags=[]}, InsetsSource: {fbb80024 mType=systemGestures mFrame=[0,0][0,0] mVisible=true mFlags=[]}, InsetsSource: {3 mType=ime mFrame=[0,1649][1080,2640] mVisible=false mFlags=[]}, InsetsSource: {27 mType=displayCutout mFrame=[0,0][1080,94] mVisible=true mFlags=[]}, InsetsSource: {2b7d0000 mType=statusBars mFrame=[0,0][1080,94] mVisible=true mFlags=[]}, InsetsSource: {2b7d0005 mType=mandatorySystemGestures mFrame=[0,0][1080,126] mVisible=true mFlags=[]}, InsetsSource: {2b7d0006 mType=tappableElement mFrame=[0,0][1080,94] mVisible=true mFlags=[]} }
I/InsetsSourceConsumer(28781): applyRequestedVisibilityToControl: visible=true, type=statusBars, host=com.lib.flutter_blue_plus_example/com.lib.flutter_blue_plus_example.MainActivity
I/InsetsSourceConsumer(28781): applyRequestedVisibilityToControl: visible=true, type=navigationBars, host=com.lib.flutter_blue_plus_example/com.lib.flutter_blue_plus_example.MainActivity
I/ViewRootImpl@64e2293[MainActivity](28781): handleResized, msg = 4 frames=ClientWindowFrames{frame=[0,0][1080,2640] display=[0,0][1080,2640] parentFrame=[0,0][0,0]} forceNextWindowRelayout=false displayId=0 dragResizing=false compatScale=1.0 frameChanged=false attachedFrameChanged=false configChanged=false displayChanged=false compatScaleChanged=false
I/ViewRootImpl@64e2293[MainActivity](28781): handleWindowFocusChanged: 1 0 call from android.view.ViewRootImpl.-$$Nest$mhandleWindowFocusChanged:0
D/ViewRootImpl@64e2293[MainActivity](28781): mThreadedRenderer.initializeIfNeeded()#2 mSurface={isValid=true 0xb400007b60cea420}
D/InputMethodManagerUtils(28781): startInputInner - Id : 0
I/InputMethodManager(28781): startInputInner - IInputMethodManagerGlobalInvoker.startInputOrWindowGainedFocus
D/[FBP-Android](28781): [FBP] onScanResult
I/InsetsSourceConsumer(28781): applyRequestedVisibilityToControl: visible=false, type=ime, host=com.lib.flutter_blue_plus_example/com.lib.flutter_blue_plus_example.MainActivity
D/[FBP-Android](28781): [FBP] onScanResult
D/[FBP-Android](28781): [FBP] OnBondStateChanged: bond-none prev: bonding
I/flutter (28781): [FBP] [[ OnBondStateChanged ]] result: {bond_state: 0, remote_id: 04:91:62:A4:BD:C5, prev_state: 1}
D/BluetoothGatt(28781): onClientConnectionState() - status=19 clientIf=12 device=XX:XX:XX:XX:BD:C5
D/[FBP-Android](28781): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android](28781): [FBP]   status: REMOTE_USER_TERMINATED_CONNECTION
D/BluetoothGatt(28781): close()
D/BluetoothGatt(28781): unregisterApp() - mClientIf=12
D/[FBP-Android](28781): [FBP] onScanResult
D/[FBP-Android](28781): [FBP] onScanResult
D/[FBP-Android](28781): [FBP] onScanResult
I/flutter (28781): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 19, disconnect_reason_string: REMOTE_USER_TERMINATED_CONNECTION, remote_id: 04:91:62:A4:BD:C5, connection_state: 0}
D/[FBP-Android](28781): [FBP] onScanResult
```

chipweinberger, 2024-06-26:
your ble device is triggering disconnection

REMOTE_USER_TERMINATED_CONNECTION

you need to check ble device logs

freecions, 2024-06-26:
REMOTE_USER_TERMINATED_CONNECTION does not appear on IOS. REMOTE_USER_TERMINATED_CONNECTION appears on Android. FYI, my phone is using Galay Z Flip4 and my iPhone is 14. iPhone is working normally.
Is it because of the Android API 14 version?
And where can I find out the REMOTE_USER_TERMINATED_CONNECTION phenomenon?

chipweinberger, 2024-06-26:
android is doing something your device does not like 

i had a similar issue on ios

https://github.com/espressif/esp-idf/issues/14053

chipweinberger, 2024-06-27:
but we cant fix this for you

closed 2024-06-27 by chipweinberger