chipweinberger/flutter_blue_plus #918: [Feature]: Ability to check if its bluetooth 5 or not.

cimplesid-tekkon, 2024-06-27:
### FlutterBluePlus Version

latest

### Flutter Version

latest

### What OS?

All

### OS Version

latest

### Bluetooth Module

bluetooth 5

### What is your feature request?

Ability to check if its bluetooth 5 or not.Since isLe2MPhySupport is only from Bluetooth 5

### Logs

```shell
no logs
```

chipweinberger, 2024-06-27:
not possible.

if you know of an api, let me know

but there is no api for this afaik

chipweinberger, 2024-06-27:
but we already have code to get phy support

FlutterBluePlus.getSupportedPhy

cimplesid-tekkon, 2024-06-27:
> but we already have code to get phy support
> 
> FlutterBluePlus.getPhySupport

do we also have the `isLeCodedPhySupported` , `isLeExtendedAdvertisingSupported` ,  .isLePeriodicAdvertisingSupported'?

chipweinberger, 2024-06-27:
The code is open source

Please be self reliant and answer your own simple questions

closed 2024-06-27 by cimplesid-tekkon

sqcsabbey, 2024-08-14:
@chipweinberger Doesn't look like isLeExtendedAdvertisingSupported et al are supported; would you be interested in a PR with that?