chipweinberger/flutter_blue_plus #921: [Help]:  Autoconnect does not work on Android

smithFlutter, 2024-07-01:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.8

### Flutter Version

3.7.12

### What OS?

Android

### OS Version

10

### Bluetooth Module

Android <=10 - BLE

### What is your problem?

Hello everyone. The "Auto Connect" option does not work in previous versions of Android 11, it does not work in any way, it never connects, anything to consider? In IOS and later versions of Android 10 it works.

```
 await device.connect(autoConnect: true, mtu: null, timeout: Duration(seconds: 35));
//I listen to the "device.connectionState" method, but it never connects.
```

What is needed for it to work?

### Logs

```shell
(I don't have log)
```

chipweinberger, 2024-07-02:
logs required

closed 2024-07-03 by chipweinberger

xOldeVx, 2024-07-17:
Same here, it doesn't reconnect when I destroy the app and open it again.
Nothing is displayed in Logs (exactly like `autoConnect: false` behavior).

**Code:**
```
  @override
  void initState() {
    startScan();
    super.initState();
  }

  void startScan() async {
    FlutterBluePlus.startScan(
      timeout: Duration(seconds: 30),
    );
    _scanResultsSubscription = FlutterBluePlus.scanResults.listen((results) {
      results.forEach(
        (e) {
          print('🚀${e.device.advName}');
        },
      );
    }, onError: (e) {
      print(e);
    });
  }

  void connectToDeviceFunction(BluetoothDevice device) {
    device.connect(autoConnect: true, mtu: null);
    device.connectionState.listen((BluetoothConnectionState state) async {
      if (state != BluetoothConnectionState.disconnected) {
        await device.requestMtu(512);
        print('🚀${device.advName} connected');
      } else {
        print('❌${device.advName} disconnected, reason: $reason, code: $code');
      }
      setState(() => {});
    });
  }
```

**Details:**
Device: Galaxy S22 Ultra
Version: Android 14
Flutter version: 3.22.1
Package version: flutter_reactive_ble: ^5.3.1

chipweinberger, 2024-07-17:
auto### Save Device

To save a device between app restarts, just write the `remoteId` to a file.

Now you can connect without needing to scan again, like so:

```dart
final String remoteId = await File('/remoteId.txt').readAsString();
var device = BluetoothDevice.fromId(remoteId);
// AutoConnect is convenient because it does not "time out"
// even if the device is not available / turned off.
await device.connect(autoConnect: true);

```

xOldeVx, 2024-07-22:
I noticed in this way not all of the `BluetoothDevice` parameters is available, indeed only the `remoteId` (see screenshot).
Is there a way to get all the `BluetoothDevice` parameters? (like in regular scan)
[image]

chipweinberger, 2024-07-22:
not possible. save them also.