chipweinberger/flutter_blue_plus #923: [Help]: FlutterBluePlusException | connect | android-code: 133 | ANDROID_SPECIFIC_ERROR

changning420, 2024-07-03:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.4

### Flutter Version

3.16.5

### What OS?

Android

### OS Version

android 11

### Bluetooth Module

BluetoothGatt

### What is your problem?

connect Device___FlutterBluePlusException | connect | android-code: 133 | ANDROID_SPECIFIC_ERROR

### Logs

```shell
D/[FBP-Android](28012): [FBP] onMethodCall: connect
D/BluetoothGatt(28012): connect() - device: 48:AA:CE:83:E6:5A, auto: false
D/BluetoothGatt(28012): registerApp()
D/BluetoothGatt(28012): registerApp() - UUID=86d04fdb-1afa-4b09-8251-271648dc2eae
D/BluetoothGatt(28012): onClientRegistered() - status=0 clientIf=6
D/BluetoothGatt(28012): onClientConnectionState() - status=133 clientIf=6 device=48:AA:CE:83:E6:5A
D/[FBP-Android](28012): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android](28012): [FBP]   status: ANDROID_SPECIFIC_ERROR
D/BluetoothGatt(28012): close()
D/BluetoothGatt(28012): unregisterApp() - mClientIf=6
```

chipweinberger, 2024-07-03:
this issue is not related to FBP

device or phone issue

closed 2024-07-03 by chipweinberger

changning420, 2024-07-03:
Thank you for your reply. How can I solve this kind of problem? @chipweinberger

chipweinberger, 2024-07-03:
it's in the readme.

santa010, 2024-07-12:
@changning420 Did you find a solution for this problem? I'm having the same issue.

changning420, 2024-07-12:
I've never solved it. @santa010

jasaw, 2024-07-17:
Android error 133 may be caused by many factors. As far as I understand, it's a catch all for Bluetooth spec violation errors.
For what it's worth, I have been seeing this error when connecting to one of my BLE devices that I control (I have the device firmware source code). My scenario is caused by Bluetooth timing violation and I resolved it by making the device advertise faster, run the CPU at full throttle, and make sure the Bluetooth stack doesn't miss its scheduled radio reception (I am stealing the radio time every now and then to do other stuff).