chipweinberger/flutter_blue_plus #925: When my phone turns off the screen and triggers a startScan(), it will report an error(UNKNOWN_SCAN_ERROR). If I search again, it will report an error(SCAN_FAILED_ALREADY_STARTED)

summeryi, 2024-07-04:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.8

### Flutter Version

3.19.5

### What OS?

Android

### OS Version

Android 13

### Bluetooth Module

null

### What is your problem?

When my phone turns off the screen and triggers a startScan(), it will report an error(UNKNOWN_SCAN_ERROR). If I search again, it will report an error(SCAN_FAILED_ALREADY_STARTED)

### Logs

```shell
error1:
I/flutter ( 9962): [FBP] <startScan> result: true
E/[FBP-Android]( 9962): [FBP] onScanFailed: UNKNOWN_SCAN_ERROR (100)
I/flutter ( 9962): [FBP] [[ OnScanResponse ]] result: {error_string: UNKNOWN_SCAN_ERROR (100), advertisements: [], success: 0, error_code: 100}

error2:
I/flutter ( 9962): [FBP] <startScan> args: {with_services: [], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: false, continuous_divisor: 1, android_scan_mode: 2, android_uses_fine_location: false}
D/[FBP-Android]( 9962): [FBP] onMethodCall: startScan
D/BluetoothAdapter( 9962): isLeEnabled(): ON
E/[FBP-Android]( 9962): [FBP] onScanFailed: SCAN_FAILED_ALREADY_STARTED
I/flutter ( 9962): [FBP] <startScan> result: true
I/flutter ( 9962): [FBP] [[ OnScanResponse ]] result: {error_string: SCAN_FAILED_ALREADY_STARTED, advertisements: [], success: 0, error_code: 1}
```

chipweinberger, 2024-07-04:
how are you triggering the scan when the screen is off?

summeryi, 2024-07-05:
After turning off the screen on my end, I wrote a timer that triggered it in 1 second. The problem occurred 100% of the time, and after the problem occurred, the search continued to fail

summeryi, 2024-07-05:
Future onScanPressed() async {
   Future.delayed(Duration(seconds: 1),() async {
     try {
       _systemDevices = await FlutterBluePlus.systemDevices;
     } catch (e) {
       Snackbar.show(ABC.b, prettyException("System Devices Error:", e), success: false);
     }
     try {
       await FlutterBluePlus.startScan(timeout: const Duration(seconds: 15));
     } catch (e) {
       Snackbar.show(ABC.b, prettyException("Start Scan Error:", e), success: false);
     }
     if (mounted) {
       setState(() {});
     }
   });
  }

chipweinberger, 2024-07-08:
closing in favor of new issue 

https://github.com/boskokg/flutter_blue_plus/issues/930

closed 2024-07-08 by chipweinberger