chipweinberger/flutter_blue_plus #930: [Feature] Android background support

chipweinberger, 2024-07-08:
### FlutterBluePlus Version

1.32.8

### Flutter Version

3.22

### What OS?

Android

### OS Version

all

### Bluetooth Module

all

### What is your feature request?

support BLE when app is in the background

### Logs

```shell
none
```

TheValkDokk, 2024-07-15:
don't know if this helps but i just having this issue on SM-991N, the scan results doesn't return anything, but on xiaomi 12T, it's fine, i can stream data from the flutter_foreground_task to the main ui thread

explodus, 2024-07-26:
as a specialization, not only in the background, in the background and switched off screen

TheValkDokk, 2024-07-29:
This is how i do it:

- Request ignore battery optimization
- in android manifest, request the type of service of your case
- use the [flutter_background_service](https://pub.dev/packages/flutter_background_service), have the FlutterBluePlus instance in the TaskHanler itself, not from the main UI thread

My project is private so i can't provide code but i will make an example if i have time

GreedyGrinner, 2024-08-29:
On android scanning only works in doze mode if the service is scanning for a specific mac-address or ServiceUUID, otherwise it returns empty. On iOS you need to scan for specific ServiceUUIDs as well. In terms of scanning for all advertising devices i have no found a solution.

explodus, 2024-10-02:
A lot of time has passed and I have now solved it without scanning. We changed the hardware and now have a connection and notifications from the device.

The important point to note with iOS is that you set the UIBackgroundModes correctly.

```
	<key>UIBackgroundModes</key>
	<array>
		<string>bluetooth-central</string>
		<string>bluetooth-peripheral</string>
		<string>fetch</string>
	</array>
```

The notification from the device comes when the device sends and only when the app is hard killed will no more notifications come.

Maqcel, 2024-11-25:
@TheValkDokk can you share an example?

cos-overclock, 2024-11-30:
One way to do this would be to add an option to register the target BLE device as a Companion Device.
I think the problems with using Companion Device are
- The required callbacks are not supported until Android 12 or later.
- Pairing and bonding required to register as a Companion Device

https://developer.android.com/develop/connectivity/bluetooth/companion-device-pairing?#keep-awake

3ph, 2024-12-16:
> On android scanning only works in doze mode if the service is scanning for a specific mac-address or ServiceUUID, otherwise it returns empty. On iOS you need to scan for specific ServiceUUIDs as well. In terms of scanning for all advertising devices i have no found a solution.

You can't even start scanning when the screen is off - `startScan` reports `UNKNOWN_SCAN_ERROR`

@chipweinberger It's not possible to recover from the `UNKNOWN_SCAN_ERROR` even when the app is foregrounded, subsequent `startScan` invocations would return `SCAN_FAILED_ALREADY_STARTED` even though the scan was finished or `stopScan` called... Is this also Android issue rather than FBP issue?

tayz98, 2025-02-16:
According to this android documentation (https://developer.android.com/develop/connectivity/bluetooth/ble/background#connect-background) there are four ways to do this:

- workmanager
- foreground service
- pendingIntent
- CompanionDeviceService

For the first two ways libraries already exist. Cos-Overclock already recommended the Companion Device Functionality. So what's left is, is using a pendingIntent. But I have never used, hence why I can't give say if it can be integrated into FBP or not.

cos-overclock, 2025-08-27:
Regarding #930, I believe the advantage of supporting Companion Devices lies in enabling background operation without requiring external packages like Foreground Services or workmanager.