chipweinberger/flutter_blue_plus #932: [Help]: Sometime i can not submit notify

kle-wang, 2024-07-15:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.7

### Flutter Version

3.22.1

### What OS?

Android

### OS Version

android 14

### Bluetooth Module

earphone

### What is your problem?

When I try to connect my device, I may not be able to subscribe to the notification service's uuid and encounter an error. This error is usually resolved by turning on the Bluetooth switch again or waiting a few minutes to try connecting again. Is there any way to solve this matter. This incident occurred on my Redmi K60 phone, but I did not find this issu
e on the Samsung S20 model. At the same time, I discovered a connection issue in the Android BLE GitHub for your reference.
github page : https://github.com/NordicSemiconductor/Android-BLE-Library/issues/507

### Logs

```shell
D/[FBP-Android]( 6592): [FBP] onMethodCall: setNotifyValue
D/BluetoothGatt( 6592): setCharacteristicNotification() - uuid: 00008888-0000-1000-8000-00805f9b34fb enable: true
E/flutter ( 6592): [ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PlatformException(androidException, java.lang.SecurityException: Need BLUETOOTH PRIVILEGED permission: Neither user 10302 nor current process has android.permission.BLUETOOTH_PRIVILEGED., java.lang.SecurityException: Need BLUETOOTH PRIVILEGED permission: Neither user 10302 nor current process has android.permission.BLUETOOTH_PRIVILEGED.
E/flutter ( 6592): 	at android.app.ContextImpl.enforce(ContextImpl.java:2343)
E/flutter ( 6592): 	at android.app.ContextImpl.enforceCallingOrSelfPermission(ContextImpl.java:2371)
E/flutter ( 6592): 	at android.content.ContextWrapper.enforceCallingOrSelfPermission(ContextWrapper.java:993)
E/flutter ( 6592): 	at com.android.bluetooth.Utils.enforceBluetoothPrivilegedPermission(Utils.java:474)
E/flutter ( 6592): 	at com.android.bluetooth.gatt.GattService.permissionCheck(GattService.java:751)
E/flutter ( 6592): 	at com.android.bluetooth.gatt.GattService.registerForNotification(GattService.java:4971)
E/flutter ( 6592): 	at com.android.bluetooth.gatt.GattService$BluetoothGattBinder.registerForNotification(GattService.java:1398)
E/flutter ( 6592): 	at com.android.bluetooth.gatt.GattService$BluetoothGattBinder.registerForNotification(GattService.java:1386)
E/flutter ( 6592): 	at android.bluetooth.IBluetoothGatt$Stub.onTransact(IBluetoothGatt.java:1110)
E/flutter ( 6592): 	at android.os.Binder.execTransactInternal(Binder.java:1351)
E/flutter ( 6592): 	at android.os.Binder.execTransact(Binder.java:1282)
E/flutter ( 6592): , null)
E/flutter ( 6592): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:648:7)
E/flutter ( 6592): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:334:18)
E/flutter ( 6592): <asynchronous suspension>
E/flutter ( 6592): #2      FlutterBluePlus._invokeMethod (package:flutter_blue_plus/src/flutter_blue_plus.dart:603:13)
E/flutter ( 6592): <asynchronous suspension>
E/flutter ( 6592): #3      BluetoothCharacteristic.setNotifyValue (package:flutter_blue_plus/src/bluetooth_characteristic.dart:270:22)
E/flutter ( 6592): <asynchronous suspension>
E/flutter ( 6592): #4      Ble._subscriptionCharacteristics (package:roselink/lib/ble.dart:312:5)
E/flutter ( 6592): <asynchronous suspension>
E/flutter ( 6592):
```

chipweinberger, 2024-07-16:
https://issuetracker.google.com/issues/304893289?pli=1

> This issue is observed in the field a lot with our end users on phones running Android 13 and higher.
>
> Looking at the source where it happens, it's easily explained that the issue starts whenever the `onGetGattDb` method is invoked by thread A while thread B is still busy performing a GATT operation (read, write, registering notifications, etc.), filling the mRestrictedHandles map with handles pointing to characteristics that are not 'restricted' at all.
>
> Source: >https://cs.android.com/android/platform/superproject/+/master:packages/modules/Bluetooth/android/app/src/com/android/bluetooth/gatt/GattService.java:278

chipweinberger, 2024-07-16:
> This is fixed in [Android 15 Beta 2](https://developer.android.com/about/versions/15/release-notes#beta-2).
>
> Please update to Android 15 Beta and verify.
>
> Thank you!

kle-wang, 2024-07-16:
In our country.Most people's phones are based on manufacturer modified ROM systems, which may not be able to update Android versions in a timely manner. Is there a way to solve this problem? Do you have any suggestions, such as adding message queues or reading whether the system is currently busy?

chipweinberger, 2024-07-16:
we already use message queues.

I don't have a better solution.

you can try to read the android source code to understand the issue better.

chipweinberger, 2024-07-17:
this is an android issue. no workaround that i know of.

closed 2024-07-17 by chipweinberger

rongshuizhou, 2024-07-30:
> In our country.Most people's phones are based on manufacturer modified ROM systems, which may not be able to update Android versions in a timely manner. Is there a way to solve this problem? Do you have any suggestions, such as adding message queues or reading whether the system is currently busy?

Have you solved this problem?

kle-wang, 2024-07-30:
No...May be in ios i use ble and android use spp...

rongshuizhou, 2024-07-30:
Can you tell me how you solved this problem? I have this problem in
Bluetooth subscription

kle-wang, 2024-07-30:
ahhh
我用经典蓝牙的spp了
老铁

rongshuizhou, 2024-07-31:
谢谢，我直接用flutter_reactive_ble解决了这个问题