chipweinberger/flutter_blue_plus #934: [Help]: Android: Race condition between gatt.disconnect and Bluetooth stack establishing connection

jasaw, 2024-07-16:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.8

### Flutter Version

3.22.2

### What OS?

Android

### OS Version

14

### Bluetooth Module

Nordic nRF

### What is your problem?

In there rare case of FBP calling `gatt.disconnect()` while the Android Bluetooth stack is in the middle of establishing Bluetooth connection, the connection won't be cancelled even though `gatt.disconnect()` and `gatt.close()` have been called, but FBP responds to the app with `connection canceled`. After that, the `onConnectionStateChange` callback gets called and FBP happily stores the `gatt` against the `remoteId`, maintaining a Bluetooth connection that the app is not aware of.

This is not a serious issue as the app can periodically poll FBP to disconnect all Bluetooth connections, but it would be nicer if the app doesn't need to poll FBP.

### Logs

```shell
D/[FBP-Android](30007): [FBP] onMethodCall: connect
D/BluetoothAdapter(30007): getBleEnabledArray(): ON
D/BluetoothGatt(30007): connect() - device: XX:XX:XX:XX:8A:77, auto: false
D/BluetoothGatt(30007): registerApp()
D/BluetoothGatt(30007): registerApp() - UUID=f06d1d8e-06a2-4466-9beb-a9dbe69aca7f
D/BluetoothGatt(30007): onClientRegistered() - status=0 clientIf=8
D/BluetoothAdapter(30007): getBleEnabledArray(): ON
I/flutter (30007): 0:00:50.491900	I	Device:	 Disconnecting Bluetooth connection
D/[FBP-Android](30007): [FBP] onMethodCall: disconnect
D/[FBP-Android](30007): [FBP] disconnect: cancelling connection in progress
D/BluetoothGatt(30007): cancelOpen() - device: XX:XX:XX:XX:8A:77
D/BluetoothGatt(30007): close()
D/BluetoothGatt(30007): onClientConnectionState() - status=0 clientIf=8 device=XX:XX:XX:XX:8A:77
D/[FBP-Android](30007): [FBP] onConnectionStateChange:connected
D/[FBP-Android](30007): [FBP]   status: SUCCESS
D/BluetoothGatt(30007): unregisterApp() - mClientIf=8
D/[FBP-Android](30007): [FBP] onMethodCall: requestMtu
D/BluetoothGatt(30007): configureMTU() - device: XX:XX:XX:XX:8A:77 mtu: 512
```

chipweinberger, 2024-07-17:
Why would the 2 second delay not solve this?

chipweinberger, 2024-07-17:
we already have this

```
    private final Map<String, BluetoothGatt> mConnectedDevices = new ConcurrentHashMap<>();
    private final Map<String, BluetoothGatt> mCurrentlyConnectingDevices = new ConcurrentHashMap<>();
```

I dont think we need

`    private final Map<String, BluetoothGatt> mStayDisconnectedDevices = new ConcurrentHashMap<>();`

also I'm not confident this code will work correctly with autoconnect

jasaw, 2024-08-01:
Fixed by #935

closed 2024-08-01 by jasaw

samnbd, 2024-09-10:
The issue is still happening on version **_1.32.12_**

chipweinberger, 2024-09-10:
share logs ^