chipweinberger/flutter_blue_plus #938: [Feature]: Allow legacy only scan

jasaw, 2024-07-17:
### FlutterBluePlus Version

1.32.8

### Flutter Version

3.22.2

### What OS?

Android

### OS Version

14

### Bluetooth Module

Nordic nRF

### What is your feature request?

Add `legacyOnly` option to `FlutterBluePlus.startScan` function. This should effectively set `setLegacy(true)` in Android. I haven't looked into iOS.

Currently FBP defaults to scanning all supported LE PHY, which is 1M (legacy, BLE 4) and 2M phy. A device scanning for BLE advertisements would typically time slice the one Bluetooth receiver on the phone so it can listen on the 2 different phy modes. Most phones take the sensible approach of listening on 1M for a few hundred milliseconds, then 2M for a few hundred milliseconds, then cycle back to 1M. However, there are some Samsung Android phones that run each time slice as a 4 second block. The consequence is, if the app is only receiving advertisement packets from 1M BLE peripherals, then the app would get no advertisement packets for the whole 4 seconds when the Bluetooth receiver is time-sliced to listen to 2M. This makes the app user experience unpleasant and in some cases, unacceptable. When the Bluetooth receiver is configured to legacy mode, it does not do time slicing, so no gaps.

### Logs

```shell
I/flutter (23037): 0:00:22.721585	W	BLE adv:	 RSSI: -56
I/flutter (23037): 0:00:22.819617	W	BLE adv:	 RSSI: -56
I/flutter (23037): 0:00:22.907776	W	BLE adv:	 RSSI: -57
I/flutter (23037): 0:00:26.946804	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:27.011124	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:27.134106	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:27.193982	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:27.265313	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:27.334196	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:27.438117	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:27.470906	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:27.540466	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:27.657761	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:27.659668	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:27.727361	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:27.805404	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:27.858611	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:28.081101	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:28.082593	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:28.131663	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:28.206736	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:28.271321	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:28.328822	W	BLE adv:	 RSSI: -56
I/flutter (23037): 0:00:28.407752	W	BLE adv:	 RSSI: -56
I/flutter (23037): 0:00:28.464146	W	BLE adv:	 RSSI: -56
I/flutter (23037): 0:00:28.530425	W	BLE adv:	 RSSI: -56
I/flutter (23037): 0:00:28.678693	W	BLE adv:	 RSSI: -56
I/flutter (23037): 0:00:28.742831	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:28.841678	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:28.862937	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:28.932725	W	BLE adv:	 RSSI: -56
I/flutter (23037): 0:00:29.022595	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:29.074796	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:29.155209	W	BLE adv:	 RSSI: -55
I/flutter (23037): 0:00:29.247606	W	BLE adv:	 RSSI: -56
I/flutter (23037): 0:00:29.286405	W	BLE adv:	 RSSI: -56
I/flutter (23037): 0:00:29.356074	W	BLE adv:	 RSSI: -56
I/flutter (23037): 0:00:29.467808	W	BLE adv:	 RSSI: -56
I/flutter (23037): 0:00:29.488967	W	BLE adv:	 RSSI: -56
I/flutter (23037): 0:00:29.550167	W	BLE adv:	 RSSI: -56
I/flutter (23037): 0:00:29.638587	W	BLE adv:	 RSSI: -56
I/flutter (23037): 0:00:29.678411	W	BLE adv:	 RSSI: -56
I/flutter (23037): 0:00:29.814928	W	BLE adv:	 RSSI: -56
I/flutter (23037): 0:00:29.817269	W	BLE adv:	 RSSI: -56
```

PrimaelQuemerais, 2024-07-17:
Hello,
I'm using FBP to connect a bluetooth thermometer to Samsung tablets and my app stopped finding any device since a few weeks without having made any changes to my code. Could this issue be related to it?

chipweinberger, 2024-07-17:
I will not be implementing this. So please open a PR if you want it.

closed 2024-07-17 by chipweinberger

jasaw, 2024-07-18:
@PrimaelQuemerais Your Bluetooth scan problem sounds a little different from this issue that I described here. Here's what I would do to investigate the problem.
1. Check that the Bluetooth thermometer is actually sending advertisement packets. Use a different Bluetooth app like "nRF Connect" on 2 different devices (your Samsung tablet and main mobile phone) to confirm your thermometer is transmitting and your Samsung tablet is not broken.
2. If your thermometer is not transmitting, try power cycling it, rebooting it.
3. If everything is working with nRF Connect app, then it's something in your app or FBP. There are a few Android Bluetooth settings that are not set by FBP, so they have default values that are determined by Samsung and Samsung has definitely messed around with their Bluetooth stack in their recent OS upgrades (at least on my phone).