chipweinberger/flutter_blue_plus #941: [Help]: Background Bluetooth search, no response when the app is opened

gaogaogoo, 2024-07-19:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.8

### Flutter Version

3.19.6

### What OS?

Android

### OS Version

14

### Bluetooth Module

none

### What is your problem?

Here, my test phone is an OPPO K12 5G. The steps are as follows:
1. Open the app and start Bluetooth scanning.
2. Send the app to the background and lock the screen.
3. Wait for 1-2 hours and then reopen the app from the background.
4. The app becomes unresponsive, and after a few seconds, the system kills the app automatically.

After encountering this issue, I exported two ANR logs from the phone. Could you please help me check the problem?

### Logs

```shell
The first log is as follows:
Build fingerprint: 'OPPO/PJR110/OP5A15L1:14/TP1A.220905.001/U.1863da6_1-45fc:user/release-keys'
ABI: 'arm64'
Build type: optimized
suspend all histogram:	Sum: 6.967ms 99% C.I. 0.426us-2207.999us Avg: 102.455us Max: 2705us
DALVIK THREADS (32):
"main" prio=5 tid=1 Runnable
  | group="main" sCount=0 ucsCount=0 flags=0 obj=0x724dff88 self=0xb400007150610800
  | sysTid=2735 nice=-10 cgrp=top-app sched=0/0 handle=0x72036ce500
  | state=R schedstat=( 15358415690 412825335 7450 ) utm=1444 stm=91 core=6 HZ=100
  | stack=0x7fdf9dd000-0x7fdf9df000 stackSize=8188KB
  | held mutexes= "mutator lock"(shared held)
  at android.os.MessageQueue.enqueueMessage(MessageQueue.java:629)
  - locked <0x057d113b> (a android.os.MessageQueue)
  at android.os.Handler.enqueueMessage(Handler.java:786)
  at android.os.Handler.sendMessageAtTime(Handler.java:735)
  at android.os.Handler.sendMessageDelayed(Handler.java:705)
  at android.os.Handler.post(Handler.java:435)
  at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.invokeMethodUIThread(FlutterBluePlusPlugin.java:2684)
  at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.access$500(FlutterBluePlusPlugin.java:78)
  at com.lib.flutter_blue_plus.FlutterBluePlusPlugin$3.onScanResult(FlutterBluePlusPlugin.java:2045)
  at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:577)
  at android.os.Handler.handleCallback(Handler.java:958)
  at android.os.Handler.dispatchMessage(Handler.java:99)
  at android.os.Looper.loopOnce(Looper.java:257)
  at android.os.Looper.loop(Looper.java:368)
  at android.app.ActivityThread.main(ActivityThread.java:8848)
  at java.lang.reflect.Method.invoke(Native method)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:572)
  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1049)

The second log is as follows:
Build fingerprint: 'OPPO/PJR110/OP5A15L1:14/TP1A.220905.001/U.1863da6_1-45fc:user/release-keys'
ABI: 'arm64'
Build type: optimized
suspend all histogram:	Sum: 5.140ms 99% C.I. 0.222us-1245.759us Avg: 65.897us Max: 1406us
DALVIK THREADS (33):
"main" prio=5 tid=1 Runnable
  | group="main" sCount=0 ucsCount=0 flags=0 obj=0x724dff88 self=0xb400007150610800
  | sysTid=19079 nice=-10 cgrp=top-app sched=0/0 handle=0x72036ce500
  | state=R schedstat=( 17744349814 241346868 9378 ) utm=1661 stm=112 core=7 HZ=100
  | stack=0x7fdf9dd000-0x7fdf9df000 stackSize=8188KB
  | held mutexes= "mutator lock"(shared held)
  at java.lang.Integer.parseInt(Integer.java:757)
  at java.lang.Integer.parseInt(Integer.java:915)
  at java.util.Formatter$FormatSpecifier.width(Formatter.java:2910)
  at java.util.Formatter$FormatSpecifier.<init>(Formatter.java:2962)
  at java.util.Formatter$FormatSpecifierParser.<init>(Formatter.java:2778)
  at java.util.Formatter.parse(Formatter.java:2711)
  at java.util.Formatter.format(Formatter.java:2660)
  at java.util.Formatter.format(Formatter.java:2614)
  at java.lang.String.format(String.java:4017)
  at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.bytesToHex(FlutterBluePlusPlugin.java:2725)
  at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.access$900(FlutterBluePlusPlugin.java:78)
  at com.lib.flutter_blue_plus.FlutterBluePlusPlugin$3.onScanResult(FlutterBluePlusPlugin.java:2014)
  at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:577)
  at android.os.Handler.handleCallback(Handler.java:958)
  at android.os.Handler.dispatchMessage(Handler.java:99)
  at android.os.Looper.loopOnce(Looper.java:257)
  at android.os.Looper.loop(Looper.java:368)
  at android.app.ActivityThread.main(ActivityThread.java:8848)
  at java.lang.reflect.Method.invoke(Native method)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:572)
  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1049)
```

chipweinberger, 2024-07-21:
maybe there are many thousands of scan results taking too long to run bytesToHex?

try again, but:
1. log the byte length in bytesToHex
2. if its above some big big limit, maybe dont skip the results?

chipweinberger, 2024-07-21:
use this code. it's faster.

```
private static String bytesToHex(byte[] bytes) {
    if (bytes == null) {
        return "";
    }
    StringBuilder sb = new StringBuilder(bytes.length * 2);
    for (byte b : bytes) {
        sb.append(Character.forDigit((b >> 4) & 0xF, 16));
        sb.append(Character.forDigit(b & 0xF, 16));
    }
    return sb.toString();
}
```

gaogaogoo, 2024-07-22:
> use this code. it's faster.
>
> ```
> private static String bytesToHex(byte[] bytes) {
>     if (bytes == null) {
>         return "";
>     }
>     StringBuilder sb = new StringBuilder(bytes.length * 2); [...]

Thank you for your reply. After using this code, I still encountered ANR errors. The error is as follows:

Build fingerprint: 'OPPO/PJR110/OP5A15L1:14/TP1A.220905.001/U.1863da6_1-45fc:user/release-keys'
ABI: 'arm64'
Build type: optimized
suspend all histogram:	Sum: 5.862ms 99% C.I. 0.426us-2247.680us Avg: 112.730us Max: 2865us
DALVIK THREADS (37):
"main" prio=5 tid=1 TimedWaiting
  | group="main" sCount=1 ucsCount=0 flags=1 obj=0x724dff88 self=0xb400007150610800
  | sysTid=17911 nice=-10 cgrp=top-app sched=0/0 handle=0x72036ce500
  | state=S schedstat=( 16504636957 964614829 13725 ) utm=1453 stm=197 core=4 HZ=100
  | stack=0x7fdf9dd000-0x7fdf9df000 stackSize=8188KB
  | held mutexes=
  at jdk.internal.misc.Unsafe.park(Native method)
  - waiting on an unknown object
  at java.util.concurrent.locks.LockSupport.parkNanos(LockSupport.java:252)
  at java.util.concurrent.CompletableFuture$Signaller.block(CompletableFuture.java:1842)
  at java.util.concurrent.ForkJoinPool.unmanagedBlock(ForkJoinPool.java:3466)
  at java.util.concurrent.ForkJoinPool.managedBlock(ForkJoinPool.java:3437)
  at java.util.concurrent.CompletableFuture.timedGet(CompletableFuture.java:1915)
  at java.util.concurrent.CompletableFuture.get(CompletableFuture.java:2071)
  at com.android.modules.utils.SynchronousResultReceiver.awaitResultNoInterrupt(SynchronousResultReceiver.java:210)
  at android.bluetooth.BluetoothDevice.getName(BluetoothDevice.java:1743)
  at com.lib.flutter_blue_plus.FlutterBluePlusPlugin.bmScanAdvertisement(FlutterBluePlusPlugin.java:2490)
  at com.lib.flutter_blue_plus.FlutterBluePlusPlugin$3.onScanResult(FlutterBluePlusPlugin.java:2044)
  at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:577)
  at android.os.Handler.handleCallback(Handler.java:958)
  at android.os.Handler.dispatchMessage(Handler.java:99)
  at android.os.Looper.loopOnce(Looper.java:257)
  at android.os.Looper.loop(Looper.java:368)
  at android.app.ActivityThread.main(ActivityThread.java:8848)
  at java.lang.reflect.Method.invoke(Native method)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:572)
  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1049)

chipweinberger, 2024-07-23:
this seems like a bug in android.

closed 2024-07-23 by chipweinberger