chipweinberger/flutter_blue_plus #942: When I use my iPhone 15 to scan Bluetooth, I cannot detect the device's Bluetooth

jyljames, 2024-07-19:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.6

### Flutter Version

3.17

### What OS?

iOS

### OS Version

17.5.1

### Bluetooth Module

aic8800m40

### What is your problem?

When I use my iPhone 15 to scan Bluetooth, I cannot detect the device's Bluetooth, which is an inevitable phenomenon. After checking the logs, I found that the information scanned seems to be somewhat incorrect. This is a screenshot of the log
[image]
Afterwards, I checked the logs using a device that could scan them normally, and I found that the logs of the two were different. Below are the logs that can scan the device normally

[image]

### Logs

```shell
log pasted on problem
```

chipweinberger, 2024-07-21:
this is expected.

closed 2024-07-21 by chipweinberger

jyljames, 2024-07-21:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

jyljames, 2024-07-22:
What kind of situation can cause the scanned result to be the same as the first image?