chipweinberger/flutter_blue_plus #943: [Help]: FlutterBluePlus.systemDevices return empty on IOS

cheasonxie, 2024-07-20:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.8

### Flutter Version

3.16.0

### What OS?

iOS

### OS Version

17.5.1

### Bluetooth Module

iphone 12 mini (I don't know what ble module this phone uses)

### What is your problem?

FlutterBluePlus.systemDevices cannot found paired devices.
if disconnect paired device, this device can be found in scanresult.

my device name is ZY-M630, cannot be found in the following log

### Logs

```shell
Launching lib/main.dart on 李金峰的iPhone in debug mode...
Warning: Missing build name (CFBundleShortVersionString).
Warning: Missing build number (CFBundleVersion).
Action Required: You must set a build name and number in the pubspec.yaml file version field before submitting to the App Store.
Automatically signing iOS for device deployment using specified development team in Xcode project: 598ZRQGDNB
Running Xcode build...                                          
 └─Compiling, linking and signing...                      2,069ms
Xcode build done.                                           16.3s
Installing and launching...                                     
You may be prompted to give access to control Xcode. Flutter uses Xcode to run your app. If access is not allowed, you can change this through your Settings > Privacy & Security > Automation.
[ERROR:flutter/shell/platform/darwin/graphics/FlutterDarwinContextMetalImpeller.mm(42)] Using the Impeller rendering backend.
flutter: [FBP] \^[[1;30m<setLogLevel>\^[[0m args: \^[[1;35<…>
flutter: [FBP] \^[[1;30m<setLogLevel>\^[[0m result: \^[[1;33mtr<…>
flutter: [FBP] \^[[1;30m<getAdapterState>\^[[0m args: \^[[1;35mnu<…>
flutter: [FBP] \^[[1;30m[[ OnAdapterStateChanged ]]\^[[0m result: \^[[1;33m{adapter_state: <…>
flutter: [FBP] \^[[1;30m<getAdapterState>\^[[0m result: \^[[1;33m{adapter_state: <…>
Installing and launching...                                        32.2s
Syncing files to device 李金峰的iPhone...                               60ms

Flutter run key commands.
r Hot reload. 🔥🔥🔥
R Hot restart.
h List all available interactive commands.
d Detach (terminate "flutter run" but leave application running).
c Clear the screen
q Quit (terminate the application on the device).

A Dart VM Service on 李金峰的iPhone is available at: http://127.0.0.1:59453/8-seri1O9V0=/
The Flutter DevTools debugger and profiler on 李金峰的iPhone is available at: http://127.0.0.1:9100?uri=http://127.0.0.1:59453/8-seri1O9V0=/
Unexpectedly found more than one Dart VM Service report for com.charlesweinberger.flutterBluePlusExample4._dartVmService._tcp.local - using first one (51555).
Unexpectedly found more than one Dart VM Service report for com.charlesweinberger.flutterBluePlusExample4 (2)._dartVmService._tcp.local - using first one (59453).
flutter: [FBP] \^[[1;30m<getSystemDevices>\^[[0m args: \^[[1;35mnu<…>
[... 15 lines cut ...]
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{manufacturer_data: {31342: 88a000048811577e}, service_uuids: [ffe0, fee7], rssi: -71, remote_id: 591B5405-A0BA-6A5D-4EA2-25AE3E12C2D3, platform_name: ZY-E500, connectable: 1, adv_name: ZY-E500}<…>
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{remote_id: 5197ADFA-5B14-9657-33D6-FE3EDF314F6E, platform_name: JIIIA1230329109C, tx_power_level: 0, connectable: 1, rssi: -94, adv_name: JIIIA1230329109C}<…>
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{remote_id: 46356F55-82FE-1F9D-C9D6-802F0DDE07F6, platform_name: Sinilink-APP, service_uuids: [1812], connectable: 1, rssi: -71}<…>
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{rssi: -42, remote_id: FC522AB1-8E20-E0B2-C9F0-55FDEA62555D, manufacturer_data: {6: 010f20225a1e8829af62c0dc92f2f0ca9c5081c06e504f5af10717}}<…>
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{remote_id: 1A56B271-B483-1FFB-2430-06FD03F65722, platform_name: NEEWER-DL300, manufacturer_data: {7183: 88a0000488113d1d}, connectable: 1, service_uuids: [ffe0, fee7], rssi: -89}<…>
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{rssi: -91, remote_id: 684564E8-0A9B-50F5-D3F9-0F1BFDF35FA3, manufacturer_data: {6: 01092002fbb77fcc760f7483f07ed7bd6170d130ae8d258c3fe41c}}<…>
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{remote_id: 869E6716-78C5-9ADE-C604-F408519034B3, platform_name: U-AC2520, rssi: -86, connectable: 1, service_data: {ff02: 0539315236810668e478e72520, ff01: 552b0515000511011a0510a30541}, adv_name: U-AC2520}<…>
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{rssi: -80, remote_id: 5EBEC51F-99AC-B652-5C36-91A73821F1B6, manufacturer_data: {6: 010f2002fef3499bee92ec66f4d1b7ed688aee8b36c3ded2563d96}}<…>
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{remote_id: 0868B9C7-5B5D-20FE-B031-B5BE3B7F2516, platform_name: 34J192200001, service_uuids: [fe79], connectable: 1, rssi: -88, adv_name: 34J192200001}<…>
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{remote_id: 880F9C4A-0645-D405-CF4D-15F189AB6F5F, platform_name: U-AC045E, rssi: -87, connectable: 1, service_data: {ff02: 0539316139800668e478e6045e, ff01: 552b0515000511011a5410a30541}, adv_name: U-AC045E}<…>
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{manufacturer_data: {7217: 88a0000488113d1f}, service_uuids: [ffe0, fee7], rssi: -98, remote_id: BF751C63-7632-6D0B-7262-303CB0BDA826, platform_name: NEEWER-DL300, connectable: 1, adv_name: NEEWER-DL300}<…>
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{remote_id: 024F7CC3-EB4B-3610-2107-FDFC65E802C5, platform_name: U-AC758D, rssi: -84, connectable: 1, service_data: {ff02: 0539316139800668e478b9758d, ff01: 552b0515000511011a5410a30541}, adv_name: U-AC758D}<…>
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{rssi: -73, remote_id: 03F4887E-325D-33CE-014B-B0A7FBC9E936}<…>
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{rssi: -61, remote_id: 742C8265-B5A0-AFBE-1CBB-7DBCBC990B21, manufacturer_data: {6: 010f202292e93f15b851513e8fcc2b4a4f552bebeee87928c3294e}}<…>
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{rssi: -90, remote_id: 7BCA3067-12F5-164D-7211-6E145AAFDCE5}<…>
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{manufacturer_data: {18522: 4a49454c49}, service_uuids: [1812], rssi: -74, remote_id: 46356F55-82FE-1F9D-C9D6-802F0DDE07F6, platform_name: Sinilink-APP, connectable: 1, adv_name: Sinilink-APP}<…>
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{remote_id: 26BAD521-0021-5AA1-49CE-B984DB87C175, rssi: -90, connectable: 1, tx_power_level: 7}<…>
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{manufacturer_data: {7183: 88a0000488113d1d}, service_uuids: [ffe0, fee7], rssi: -88, remote_id: 1A56B271-B483-1FFB-2430-06FD03F65722, platform_name: NEEWER-DL300, connectable: 1, adv_name: NEEWER-DL300}<…>
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{remote_id: 08FCA8F7-5F46-E31E-9AE0-6F81CFA7DBD7, platform_name: U-ACCFBE, rssi: -95, connectable: 1, service_data: {ff02: 0539315236810668e478e5cfbe, ff01: 552b0515000511011a0510a30541}, adv_name: U-ACCFBE}<…>
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{service_uuids: [fe79], rssi: -86, remote_id: 0868B9C7-5B5D-20FE-B031-B5BE3B7F2516, platform_name: 34J192200001, tx_power_level: 0, connectable: 1, adv_name: 34J192200001}<…>
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{rssi: -92, remote_id: 52D4D00B-78BC-19C1-FF34-3BABCC62B80F}<…>
flutter: [FBP] \^[[1;30m[[ OnScanResponse ]]\^[[0m result: \^[[1;33m{advertisements: [{rssi: -99, remote_id: 649983C9-BE4F-34CC-B340-566F03928B36, manufacturer_data: {6: 010f20228c711515bd9f8db4001643e1888c03e6399b29aa0d20a2}}<…>
flutter: [FBP] \^[[1;30m<stopScan>\^[[0m args: \^[[1;35mnu<…>
flutter: [FBP] \^[[1;30m<stopScan>\^[[0m result: \^[[1;33mtr<…>
```

cheasonxie, 2024-07-20:
lightblue can found this paired device, Its source code can be refered to.

cheasonxie, 2024-07-20:
It's ok after I add my device's UUID to uuid array in file FlutterBluePlusPlugin.m
under "getSystemDevices" method, like this:
```
            // Cannot pass blank UUID list for security reasons.
            // Assume all devices have the Generic Access service 0x1800
            CBUUID* gasUuid = [CBUUID UUIDWithString:@"1800"];
             CBUUID* gasUuid1 = [CBUUID UUIDWithString:@"ffe0"];
             CBUUID* gasUuid2 = [CBUUID UUIDWithString:@"69400001-b5a3-f393-e0a9-e50e24dcca99"];

             // this returns devices connected by *any* app
             NSArray *periphs = [self.centralManager retrieveConnectedPeripheralsWithServices:@[gasUuid, gasUuid1, gasUuid2]];
```

closed 2024-07-20 by cheasonxie

burekas7, 2024-08-28:
@cheasonxie 
I have the same issue.
This is the only solution?
How do I know what is the UUID?

cheasonxie, 2024-08-30:
@burekas7 
hi, I don't know whether is there another solution or not. I am not familiar with IOS development. 
you can use bluetooth debug app such as nRF Connect or LightBlue to find your device's UUID