chipweinberger/flutter_blue_plus #945: [Help]: The second time cannot be written

zhaoxwgit, 2024-07-23:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

No

### FlutterBluePlus Version

1.32.8

### Flutter Version

3.19.6

### What OS?

iOS

### OS Version

17.5.1

### Bluetooth Module

n32 mcc

### What is your problem?

The second time cannot be written

### Logs

```shell
flutter: [FBP] <setNotifyValue> args: {remote_id: 19CC7449-916B-6B4C-6AF5-0156D81FA962, service_uuid: ffe0, secondary_service_uuid: null, characteristic_uuid: ffe1, force_indications: false, enable: true}
flutter: [FBP] <setNotifyValue> result: true
flutter: [FBP] [[ OnDescriptorWritten ]] result: {error_string: success, remote_id: 19CC7449-916B-6B4C-6AF5-0156D81FA962, secondary_service_uuid: null, characteristic_uuid: ffe1, success: 1, value: 01000000, error_code: 0, service_uuid: ffe0, descriptor_uuid: 2902}
flutter: write: AA 70 01 00 2B 55
flutter: Is subscribed: true
flutter: [FBP] <writeCharacteristic> args: {remote_id: 19CC7449-916B-6B4C-6AF5-0156D81FA962, service_uuid: ffe0, secondary_service_uuid: null, characteristic_uuid: ffe1, write_type: 1, allow_long_write: 0, value: aa7001002b55}
flutter: [FBP] <writeCharacteristic> result: true
flutter: [FBP] [[ OnCharacteristicWritten ]] result: {remote_id: 19CC7449-916B-6B4C-6AF5-0156D81FA962, secondary_service_uuid: null, characteristic_uuid: ffe1, success: 1, value: aa7001002b55, error_code: 0, service_uuid: ffe0, error_string: success}
flutter: Write: Success
flutter: [FBP] [[ OnCharacteristicReceived ]] result: {remote_id: 19CC7449-916B-6B4C-6AF5-0156D81FA962, secondary_service_uuid: null, characteristic_uuid: ffe1, success: 1, value: aaf00f4d34310a252f01110101340c000000e755, error_code: 0, service_uuid: ffe0, error_string: success}
flutter: [170, 240, 15, 77, 52, 49, 10, 37, 47, 1, 17, 1, 1, 52, 12, 0, 0, 0, 231, 85]
flutter: get info:0
flutter: write: AA 70 01 01 75 55
flutter: Is subscribed: true
flutter: [FBP] <writeCharacteristic> args: {remote_id: 19CC7449-916B-6B4C-6AF5-0156D81FA962, service_uuid: ffe0, secondary_service_uuid: null, characteristic_uuid: ffe1, write_type: 1, allow_long_write: 0, value: }
flutter: [FBP] <writeCharacteristic> result: true
flutter: [FBP] [[ OnCharacteristicWritten ]] result: {remote_id: 19CC7449-916B-6B4C-6AF5-0156D81FA962, secondary_service_uuid: null, characteristic_uuid: ffe1, success: 1, value: , error_code: 0, service_uuid: ffe0, error_string: success}
flutter: Write: Success
```

zhaoxwgit, 2024-07-23:
first: [FBP] <writeCharacteristic> args: {remote_id: 19CC7449-916B-6B4C-6AF5-0156D81FA962, service_uuid: ffe0, secondary_service_uuid: null, characteristic_uuid: ffe1, write_type: 1, allow_long_write: 0, value: aa7001002b55}
second:[FBP] <writeCharacteristic> args: {remote_id: 19CC7449-916B-6B4C-6AF5-0156D81FA962, service_uuid: ffe0, secondary_service_uuid: null, characteristic_uuid: ffe1, write_type: 1, allow_long_write: 0, value:}

chipweinberger, 2024-07-23:
not enough info. you'll need to debug this yourself. probably an issue in your code / ble device.

closed 2024-07-23 by chipweinberger

zhaoxwgit, 2024-07-23:
**await**  Characteristic.write()   is work 
@chipweinberger tank you