chipweinberger/flutter_blue_plus #948: [Bug]: Failed to control characteristics while service is not primary

xavier-wong-hk, 2024-07-25:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.11

### Flutter Version

3.22.3

### What OS?

Android

### OS Version

13

### Bluetooth Module

Build-in BLE module on Samsung Galaxy Tab A9+

### What is your problem?

I am using FBP to connect a weight scale using BLE. The scale has 2 service (`0x181D` for weight and `0x181B` for body composition). Each of them should contain 2 characteristics for `READ` and `INDICATE` with a Descriptor for CCC.

Characteristics `0x2A9E`, `0x2A9D` under service `0x181D` are working fine
However, Characteristics `0x2A9B`, `0x2A9C` under service `0x181B` (which `isPrimary: false`) are not able to `READ` or `INDICATE`

A `PlatformException` return while tried to `setNotifyValue(true)` on `0x2A9C`
```
I/flutter (12268): [FBP] <setNotifyValue> args: {remote_id: 00:09:1F:82:D1:50, 
service_uuid: 181d, secondary_service_uuid: null, characteristic_uuid: 2a9c, 
force_indications: false, enable: true}
D/[FBP-Android](12268): [FBP] onMethodCall: setNotifyValue
I/flutter (12268): PlatformException(setNotifyValue, characteristic not found in 
service (chr: '2a9c' svc: '181d'), null, null)
```
> Please note that the error mentions `characteristic not found in 
service (chr: '2a9c' svc: '181d')`, `0x2A9C` should 'belong' to service `0x181B` but not `0x181D`

Seems while I am trying to do anything on the characteristic under service which is not primary service, it tried to sending command to it's serviceUuid but not the secondaryServiceUuid. To temporary fix this issue, I change the code in 
[`flutter_blue_plus-1.32.11\lib\src\bluetooth_characteristic.dart` line `248`](https://github.com/boskokg/flutter_blue_plus/blob/54e81550204156127f20c719feae5b98d435bb5f/lib/src/bluetooth_characteristic.dart#L248)

from 
```
serviceUuid: serviceUuid,
```
to
```
serviceUuid: secondaryServiceUuid ?? serviceUuid,
```
So that I can temporary `READ` or `setNotifyValue` to the characteristic

Here are some details of the `isPrimary: false` service and characteristic under the service.

`0x181B` BluetoothService:
```
I/flutter (16523): BluetoothService{remoteId: 00:09:1F:82:D1:50, serviceUuid: 181b, isPrimary: false, 
characteristics: [BluetoothCharacteristic{remoteId: 00:09:1F:82:D1:50, serviceUuid: 181d, 
secondaryServiceUuid: 181b, characteristicUuid: 2a9b, descriptors: [], properties: CharacteristicProperties
{broadcast: false, read: false, writeWithoutResponse: false, write: false, notify: false, indicate: false, 
authenticatedSignedWrites: false, extendedProperties: false, notifyEncryptionRequired: false, 
indicateEncryptionRequired: false}, value: []}, BluetoothCharacteristic{remoteId: 00:09:1F:82:D1:50, 
serviceUuid: 181d, secondaryServiceUuid: 181b, characteristicUuid: 2a9c, descriptors: [], properties: 
CharacteristicProperties{broadcast: false, read: false, writeWithoutResponse: false, write: false, notify: 
false, indicate: false, authenticatedSignedWrites: false, extendedProperties: false, 
notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}], includedServices: []}
```

`0x2A9C` BluetoothCharacteristic:
```
I/flutter (16523): BluetoothCharacteristic{remoteId: 00:09:1F:82:D1:50, serviceUuid: 
181d, secondaryServiceUuid: 181b, characteristicUuid: 2a9c, descriptors: [], 
properties: CharacteristicProperties{broadcast: false, read: false, 
writeWithoutResponse: false, write: false, notify: false, indicate: false, 
authenticatedSignedWrites: false, extendedProperties: false, 
notifyEncryptionRequired: false, indicateEncryptionRequired: false}, value: []}
````
This is what FBP example app discover:
[image]

This is a simple demo flutter app using FBP to list all services and characteristics and properties:
[image]

This is the services discover result using nRF Connect app on the same device connecting the same ble device:
[image]
[image]

### Logs

```shell
D/[FBP-Android](12268): [FBP] onMethodCall: flutterRestart
D/[FBP-Android](12268): [FBP] initializing BluetoothAdapter
I/flutter (12268): [FBP] <setLogLevel> args: 5
D/[FBP-Android](12268): [FBP] disconnectAllDevices(flutterRestart)
D/[FBP-Android](12268): [FBP] connectedPeripherals: 0
D/[FBP-Android](12268): [FBP] onMethodCall: setLogLevel
I/flutter (12268): [FBP] <setLogLevel> result: true
I/flutter (12268): [FBP] <getBondedDevices> args: null
D/[FBP-Android](12268): [FBP] onMethodCall: getBondedDevices
I/flutter (12268): [FBP] <getBondedDevices> result: {devices: [{remote_id: 00:09:1F:82:D1:50, platform_name: UC-421BLE_82D150}, {remote_id: C4:AC:59:7A:09:F0, platform_name: UA-1200BLE_7A09F0}, {remote_id: FF:8D:C5:03:3C:00, platform_name: UP-200BLE_000025}]}
I/flutter (12268): [FBP] <connect> args: {remote_id: 00:09:1F:82:D1:50, auto_connect: 0}
D/[FBP-Android](12268): [FBP] onMethodCall: connect
I/flutter (12268): [FBP] <connect> result: true
I/flutter (12268): [FBP] <getAdapterState> args: null
D/[FBP-Android](12268): [FBP] onMethodCall: getAdapterState
I/flutter (12268): [FBP] <getAdapterState> result: {adapter_state: 4}
D/[FBP-Android](12268): [FBP] onConnectionStateChange:connected
D/[FBP-Android](12268): [FBP]   status: SUCCESS
I/flutter (12268): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 0, disconnect_reason_string: SUCCESS, remote_id: 00:09:1F:82:D1:50, connection_state: 1}
I/flutter (12268): [FBP] <requestMtu> args: {remote_id: 00:09:1F:82:D1:50, mtu: 512}
D/[FBP-Android](12268): [FBP] onMethodCall: requestMtu
I/flutter (12268): [FBP] <requestMtu> result: true
D/[FBP-Android](12268): [FBP] onMtuChanged:
D/[FBP-Android](12268): [FBP]   mtu: 512
D/[FBP-Android](12268): [FBP]   status: GATT_SUCCESS (0)
I/flutter (12268): [FBP] [[ OnMtuChanged ]] result: {error_string: GATT_SUCCESS, success: 1, remote_id: 00:09:1F:82:D1:50, error_code: 0, mtu: 512}
I/flutter (12268): [FBP] <discoverServices> args: 00:09:1F:82:D1:50
D/[FBP-Android](12268): [FBP] onMethodCall: discoverServices
I/flutter (12268): [FBP] <discoverServices> result: true
D/[FBP-Android](12268): [FBP] onServicesDiscovered:
D/[FBP-Android](12268): [FBP]   count: 11
D/[FBP-Android](12268): [FBP]   status: 0GATT_SUCCESS
I/flutter (12268): [FBP] [[ OnDiscoveredServices ]] result: {error_string: GATT_SUCCESS, success: 1, remote_id: 00:09:1F:82:D1:50, error_code: 0, services: [{included_services: [], characteristics: [{descriptors: [], service_uuid: 1800, remote_id: 00:09:1F:82:D1:50, characteristic_uuid: 2a00, properties: {broadcast: 0, write_without_response: 0, notify_encryption_required: 0, read: 1, authenticated_signed_writes: 0, extended_properties: 0, indicate: 0, indicate_encryption_required: 0, write: 0, notify: 0}}, {descriptors: [], service_uuid: 1800, remote_id: 00:09:1F:82:D1:50, characteristic_uuid: 2a01, properties: {broadcast: 0, write_without_response: 0, notify_encryption_required: 0, read: 1, authenticated_signed_writes: 0, extended_properties: 0, indicate: 0, indicate_encryption_required: 0, write: 0, notify: 0}}, {descriptors: [], service_uuid: 1800, remote_id: 00:09:1F:82:D1:50, characteristic_uuid: 2a04, properties: {broadcast: 0, write_without_response: 0, notify_encryption_required: 0, read: 1, authenticated_signed_write
I/flutter (12268): [FBP] <setNotifyValue> args: {remote_id: 00:09:1F:82:D1:50, service_uuid: 1801, secondary_service_uuid: null, characteristic_uuid: 2a05, force_indications: false, enable: true}
D/[FBP-Android](12268): [FBP] onMethodCall: setNotifyValue
I/flutter (12268): [FBP] <setNotifyValue> result: true
D/[FBP-Android](12268): [FBP] onDescriptorWrite:
D/[FBP-Android](12268): [FBP]   chr: 2a05
D/[FBP-Android](12268): [FBP]   desc: 2902
D/[FBP-Android](12268): [FBP]   status: GATT_SUCCESS (0)
I/flutter (12268): [FBP] [[ OnDescriptorWritten ]] result: {error_string: GATT_SUCCESS, service_uuid: 1801, success: 1, remote_id: 00:09:1F:82:D1:50, descriptor_uuid: 2902, error_code: 0, characteristic_uuid: 2a05, value: 0200}
D/[FBP-Android](12268): [FBP] onConnectionStateChange:disconnected
D/[FBP-Android](12268): [FBP]   status: SUCCESS
I/flutter (12268): [FBP] [[ OnConnectionStateChanged ]] result: {disconnect_reason_code: 0, disconnect_reason_string: SUCCESS, remote_id: 00:09:1F:82:D1:50, connection_state: 0}
```

chipweinberger, 2024-07-27:
please open a PR

thanks

chipweinberger, 2024-07-27:
but i don't think your fix is correct

chipweinberger, 2024-07-27:
try

secondaryServiceUuid: secondaryServiceUuid

chipweinberger, 2024-07-27:
to be clear, fbp has never been tested with secondaryServiceUuid

so you should test everything and open a pr to fix everything

DHT-Xavier, 2024-07-29:
> to be clear, fbp has never been tested with secondaryServiceUuid
> 
> so you should test everything and open a pr to fix everything

I am not familiar with BLE and android native. I did mention that it was a temporary code modification in order to subscribe to the characteristic. The characteristic properties still shows notify: false, indicate: false, etc. I was able to tell what I think is a problem base on the hint in the PlatformException.

DHT-Xavier, 2024-07-29:
> try
> 
> secondaryServiceUuid: secondaryServiceUuid

A new PlatformException returned: 
```
PlatformException(setNotifyValue, secondaryService not found '181b', null, null)
```

Seems this is returning from [`flutter_blue_plus-1.32.11\android\src\main\java\com\lib\flutter_blue_plus\FlutterBluePlusPlugin.java` line `1614`](https://github.com/boskokg/flutter_blue_plus/blob/54e81550204156127f20c719feae5b98d435bb5f/android/src/main/java/com/lib/flutter_blue_plus/FlutterBluePlusPlugin.java#L1614)

```
1609: // secondary
1610: BluetoothGattService secondaryService = null;
1611: if(secondaryServiceId != null && secondaryServiceId.length() > 0) {
1612:   secondaryService = getServiceFromArray(serviceId, primaryService.getIncludedServices());
1613:   if(secondaryService == null) {
1614:     return new ChrFound(null, "secondaryService not found '" + secondaryServiceId + "'");
1615:   }
1616: }
```
On the same file, line 1612, should `serviceId` change to `secondaryServiceId` ?

DHT-Xavier, 2024-10-23:
I forked this project and tried to apply the fix. Those changes seems work as excepted in my use case.

Please feel free to take a look and verify the change => [Comparing changes](https://github.com/chipweinberger/flutter_blue_plus/compare/master...DHT-Xavier:flutter_blue_plus:Fix-android-secondary-services#diff-310978df92b0f776b922ac4d06bd42d89a0b406f89721cfc814eccefbf3d3546L153-R153).

DHT-Xavier, 2024-10-23:
Sorry for closing. pressed the wrong button by accident.

chipweinberger, 2024-11-06:
hi, thanks for following up! sorry for the slow response.

I'll merge them!

chipweinberger, 2024-11-06:
this change does not look correct:

`if (s.serviceUuid == (secondaryServiceUuid ?? serviceUuid)) `

I think we would check both serviceUuid && secondaryServiceUuid match.

chipweinberger, 2024-11-07:
I think it should be this

```
  /// look through known services
  BmBluetoothCharacteristic? get _bmchr {
    if (FlutterBluePlus._knownServices[remoteId] != null) {
      for (var s in FlutterBluePlus._knownServices[remoteId]!.services) {
        if (s.serviceUuid == serviceUuid) {
          if (secondaryServiceUuid != null) {
            // search included services (i.e. secondary services)
            for (var s2 in s.includedServices) {
              if (s2.serviceUuid == secondaryServiceUuid) {
                for (var c in s2.characteristics) {
                  if (c.characteristicUuid == uuid) {
                    return c;
                  }
                }
              }
            }
          } else {
            // search primary service
            for (var c in s.characteristics) {
              if (c.characteristicUuid == uuid) {
                return c;
              }
            }
          }
        }
      }
    }
    return null;
  }
  ```

chipweinberger, 2024-11-07:
for reference, a discovery services result with secondary services would look like this:

```json
[
  {
    "uuid": "180D",
    "type": "primary",
    "description": "Heart Rate Service",
    "characteristics": [
      {
        "uuid": "2A37",
        "description": "Heart Rate Measurement"
      },
      {
        "uuid": "2A38",
        "description": "Body Sensor Location"
      }
    ]
  },
  {
    "uuid": "180F",
    "type": "primary",
    "description": "Battery Service",
    "characteristics": [
      {
        "uuid": "2A19",
        "description": "Battery Level"
      }
    ]
  },
  {
    "uuid": "1823",
    "type": "secondary",
    "description": "Heart Rate Sensor Configuration",
    "characteristics": [
      {
        "uuid": "2A52",
        "description": "Heart Rate Control Point"
      }
    ],
    "associatedPrimaryService": "180D"
  }
]
```

you can see, the **secondaryService** has an **associatedPrimaryService**.

This seems like a good easy way to represent them. 

But annoyingly it looks like iOS, Android, and FlutterBlue instead sometimes use **includedServices** to represent them, and other times use a **primary** + **secondary** uuid pair. 

the code is pretty annoying.

chipweinberger, 2024-11-07:
^ I implemented this change, here: https://github.com/chipweinberger/flutter_blue_plus/commit/2064c18e40d1a01e2bf1e382c746acc588cc029f#diff-acb51591d3f05a2f2fb0b18fb30b61f3ced2c41c24f8e4707a25f81ca2704f92R1073

chipweinberger, 2024-11-08:
added in 1.34.0

please open new issues if you find bugs.

closed 2024-11-08 by chipweinberger