chipweinberger/flutter_blue_plus #949: [Help]: Empty scan result on HiOS v14.0.0

ogak-github, 2024-07-30:
### Requirements

- [X] I've looked at the README 'Common Problems' section

### Have you checked this problem on the example app?

Yes

### FlutterBluePlus Version

1.32.11

### Flutter Version

3.22.2

### What OS?

Android

### OS Version

17

### Bluetooth Module

Bluetooth

### What is your problem?

Empty scan results
[image]

### Logs

```shell
D/[FBP-Android](18557): [FBP] onMethodCall: getSystemDevices
I/flutter (18557): [FBP] <getSystemDevices> result: {devices: []}
I/flutter (18557): [FBP] <startScan> args: {with_services: [], with_remote_ids: [], with_names: [], with_keywords: [], with_msd: [], with_service_data: [], continuous_updates: false, continuous_divisor: 1, android_legacy: false, android_scan_mode: 2, android_uses_fine_location: false}
D/[FBP-Android](18557): [FBP] onMethodCall: startScan
D/BluetoothAdapter(18557): isLeEnabled(): ON
D/BluetoothLeScanner(18557): onScannerRegistered() - status=0 scannerId=2 mScannerId=0
I/flutter (18557): [FBP] <startScan> result: true
I/flutter (18557): [FBP] <stopScan> args: null
D/[FBP-Android](18557): [FBP] onMethodCall: stopScan
D/BluetoothAdapter(18557): isLeEnabled(): ON
I/flutter (18557): [FBP] <stopScan> result: true
```

duncanwinner, 2024-07-30:
I have exactly the same issue on a Galaxy A55 running Android version 14.  

The same code works perfectly on a Galaxy A21s running Android 12.

Flutter version 3.22.3.

chipweinberger, 2024-08-01:
seems like OS problem

closed 2024-08-01 by chipweinberger

spehj, 2024-08-02:
@ogak-github it's probably OS-related. Check your AndroidManifest.xml if you granted all permssions needed (location for example).

ogak-github, 2024-08-07:
@spehj these are permission I used on AndroidManifest.xml. 
```
<uses-feature
        android:name="android.hardware.bluetooth_le"
        android:required="true" />
     <!-- required for API 18 - 30 -->
    <uses-permission
            android:name="android.permission.BLUETOOTH"/>
    <uses-permission
            android:name="android.permission.BLUETOOTH_ADMIN"/>
            
 <!-- API 31+ -->
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <uses-permission
            android:name="android.permission.BLUETOOTH_SCAN"
            android:usesPermissionFlags="neverForLocation" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />

    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    
```